introlab/rtabmap #374: Cannot get Octomap in Rivz

ken012git, 2019-03-18:
Hi,

I got some problems to visualize the octomap in Rviz.

As I ran `roslaunch turtlebot_octomap turtlebot_build_map.launch`  and `roslaunch kobuki_keyop keyop.launch`, I could control the turtlebot and the RTAB worked fine.

[image]

I could also see some pointcloud as I subscribe /rtabmap/cloud_map in RVIZ
[image]

I also can connect `/rtabmap/cloud_map` topic to an octomap_server. However, as I tried to visualize the octomap, I saw nothing in my RVIZ.
[image]

- `turtlebot_build_map.launch`
```
<?xml version="1.0"?>

<launch>
        <include file="$(find turtlebot_gazebo)/launch/turtlebot_world.launch" />

        <include file="$(find turtlebot_octomap)/launch/turtlebot_rtabmap.launch">
            <arg name="simulation" value="true"/>
        </include>

        <node pkg="tf" type="static_transform_publisher" name="map_world_broadcaster" args="0 0 0 0 0 0 world /map 100" />

        <node name="octomap_server" pkg="octomap_server" type="octomap_server_node">
                <remap from="cloud_in" to="/rtabmap/cloud_map" />
                <param name="frame_id" type="string" value="map" />
                <param name="latch" value="false" />
                <param name="resolution" value="0.05" />
        </node>

        <node name="rviz" pkg="rviz" type="rviz" args="-d $(find rtabmap_ros)/launch/config/turtlebot_navigation.rviz"/>

</launch>
```

- `turtlebot_rtabmap.launch` (is the same with `demo_turtlebot_mapping.launch` under `/opt/ros/kinetic/share/rtabmap_ros/launch/demo`)
```
<launch>
  <arg name="database_path"     default="rtabmap.db"/>
  <arg name="rgbd_odometry"     default="false"/>
  <arg name="rtabmapviz"        default="false"/>
  <arg name="localization"      default="false"/>
  <arg name="simulation"        default="false"/>
  <arg name="sw_registered"     default="false"/>
  <arg     if="$(arg localization)" name="args"  default=""/>
  <arg unless="$(arg localization)" name="args"  default="--delete_db_on_start"/>

  <arg     if="$(arg simulation)" name="rgb_topic"   default="/camera/rgb/image_raw"/>
  <arg unless="$(arg simulation)" name="rgb_topic"   default="/camera/rgb/image_rect_color"/>
  <arg     if="$(arg simulation)" name="depth_topic" default="/camera/depth/image_raw"/>
  <arg unless="$(arg simulation)" name="depth_topic" default="/camera/depth_registered/image_raw"/>
  <arg name="camera_info_topic" default="/camera/rgb/camera_info"/>
  
  <arg name="wait_for_transform"  default="0.2"/> 
  <!-- 
      robot_state_publisher's publishing frequency in "turtlebot_bringup/launch/includes/robot.launch.xml" 
      can be increase from 5 to 10 Hz to avoid some TF warnings.
  -->
  
  <!-- Navigation stuff (move_base) -->
  <include unless="$(arg simulation)" file="$(find turtlebot_bringup)/launch/3dsensor.launch">
     <arg     if="$(arg sw_registered)" name="depth_registration" value="false"/>
     <arg unless="$(arg sw_registered)" name="depth_registration" value="true"/>
  </include>
  <include file="$(find turtlebot_navigation)/launch/includes/move_base.launch.xml"/>
  
  <!-- Mapping -->
  <group ns="rtabmap">

    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)">
	  <param name="database_path"       type="string" value="$(arg database_path)"/>
	  <param name="frame_id"            type="string" value="base_footprint"/>
	  <param name="wait_for_transform_duration"  type="double"   value="$(arg wait_for_transform)"/>
	  <param name="subscribe_depth"     type="bool"   value="true"/>
	  <param name="subscribe_scan"      type="bool"   value="true"/>
	  <param name="map_negative_poses_ignored" type="bool" value="true"/>
	
	  <!-- inputs -->
	  <remap from="scan"            to="/scan"/>
	  <remap from="rgb/image"       to="$(arg rgb_topic)"/>
  	  <remap from="depth/image"     to="$(arg depth_topic)"/>
  	  <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
	  <remap unless="$(arg rgbd_odometry)" from="odom" to="/odom"/>
  	  
  	  <!-- output -->
  	  <remap from="grid_map" to="/map"/>
	
	  <!-- RTAB-Map's parameters: do "rosrun rtabmap rtabmap (double-dash)params" to see the list of available parameters. -->
	  <param name="RGBD/ProximityBySpace"        type="string" value="true"/>   <!-- Local loop closure detection (using estimated position) with locations in WM -->
	  <param name="RGBD/OptimizeFromGraphEnd"    type="string" value="false"/>  <!-- Set to false to generate map correction between /map and /odom -->
	  <param name="Kp/MaxDepth"                  type="string" value="4.0"/>
	  <param name="Reg/Strategy"                 type="string" value="0"/>      <!-- Loop closure transformation: 0=Visual, 1=ICP, 2=Visual+ICP -->
	  <param name="Icp/CorrespondenceRatio"      type="string" value="0.3"/>
	  <param name="Vis/MinInliers"               type="string" value="15"/>      <!-- 3D visual words minimum inliers to accept loop closure -->
	  <param name="Vis/InlierDistance"           type="string" value="0.1"/>    <!-- 3D visual words correspondence distance -->
	  <param name="RGBD/AngularUpdate"           type="string" value="0.1"/>    <!-- Update map only if the robot is moving -->
	  <param name="RGBD/LinearUpdate"            type="string" value="0.1"/>    <!-- Update map only if the robot is moving -->
	  <param name="RGBD/ProximityPathMaxNeighbors" type="string" value="0"/> 
	  <param name="Rtabmap/TimeThr"              type="string" value="700"/>
	  <param name="Mem/RehearsalSimilarity"      type="string" value="0.30"/>
	  <param name="Optimizer/Slam2D"             type="string" value="true"/>
	  <param name="Reg/Force3DoF"                type="string" value="true"/>
	  <param name="GridGlobal/MinSize"           type="string" value="20"/>
          <param name="RGBD/OptimizeMaxError"        type="string" value="0.1"/>

	  
	  <!-- localization mode -->
	  <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
	  <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
	  <param name="Mem/InitWMWithAllNodes" type="string" value="$(arg localization)"/> 
    </node>
   
    <!-- Odometry : ONLY for testing without the actual robot! /odom TF should not be already published. -->
    <node if="$(arg rgbd_odometry)" pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry" output="screen">
      <param name="frame_id"                    type="string" value="base_footprint"/>
      <param name="wait_for_transform_duration" type="double" value="$(arg wait_for_transform)"/>
      <param name="Reg/Force3DoF"               type="string" value="true"/>
      <param name="Vis/InlierDistance"          type="string" value="0.05"/>
      
      <remap from="rgb/image"       to="$(arg rgb_topic)"/>
      <remap from="depth/image"     to="$(arg depth_topic)"/>
      <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
    </node>
    
    <!-- visualization with rtabmapviz -->
    <node if="$(arg rtabmapviz)" pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
  	  <param name="subscribe_depth"             type="bool" value="true"/>
      <param name="subscribe_scan"              type="bool" value="true"/>
      <param name="frame_id"                    type="string" value="base_footprint"/>
      <param name="wait_for_transform_duration" type="double" value="$(arg wait_for_transform)"/>
    
      <remap from="rgb/image"       to="$(arg rgb_topic)"/>
      <remap from="depth/image"     to="$(arg depth_topic)"/>
      <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
      <remap from="scan"            to="/scan"/>
    </node>
    
  </group>
</launch>
```

I also provide the rqt_graph here in case.
[image]

Does anyone have the experience of visualizing octomap in rviz and could help me solve this?

Thanks a lot!

ken012git, 2019-03-18:
I found that I should add MarkerArray in my Rviz and choose occupied_cells_vis_array under Marker topic. I could see some voxels now, but It only appears the ground plane occupied voxels, as shown in the image below.

How to visualize 3D octomap in Rviz?
[image]

ken012git, 2019-03-18:
I tried to remap `cloud_in` to `/camera/depth/points` and got a reasonable result. 

Here is the snapshot and modified launch file.

[image]

```
<?xml version="1.0"?>

<launch>
        <include file="$(find turtlebot_gazebo)/launch/turtlebot_world.launch" />

        <include file="$(find turtlebot_octomap)/launch/turtlebot_rtabmap.launch">
            <arg name="simulation" value="true"/>
        </include>

        <node pkg="tf" type="static_transform_publisher" name="map_world_broadcaster" args="0 0 0 0 0 0 world /map 100" />

        <node name="octomap_server" pkg="octomap_server" type="octomap_server_node">
                <!--<remap from="cloud_in" to="/rtabmap/cloud_map" />-->
                <remap from="cloud_in" to="/camera/depth/points" />
                <param name="frame_id" type="string" value="map" />
                <param name="latch" value="false" />
                <param name="resolution" value="0.05" />
        </node>

        <node name="rviz" pkg="rviz" type="rviz" args="-d $(find rtabmap_ros)/launch/config/turtlebot_navigation.rviz"/>

</launch>

```

However, does it make sense to remap `cloud_in` to `/camera/depth/points`?

matlabbe, 2019-03-23:
For the last question about `octomap_server`, yes it makes sense, but you won't get any map corrections (after a loop closure). The best is to use OctoMap published by RTAB-Map instead of using octomap_server. In your configuration of rtabmap, it is subscribing to a laser scan topic, creating a 2D map by default. To get the OctoMap, you should force rtabmap to create the map from the RGB-D camera instead of the scan. We can do that by explcitily set the following under rtabmap node (`Grid/FromDepth` parameter should be true):
```xml
<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)">
   ...
   <param name="Grid/FromDepth"  type="string" value="true"/>
</node>
```
then `/rtabmap/octomap_XXXX` topics should be published and not empty.

cheers,
Mathieu

ken012git, 2019-03-23:
Thank you so much!   

After setting the `Grid/FromDepth` to `true` and `suscribe_scan` to `false`, the `octomap_ground`, `octomap_obstacles` and `octomap_occupied_space` seems to work fine in Rviz right now. 

However, I still get 3D map both from `/rtabmap/octomap_grid`  and `/map` (remapped from `grid_map` in the launch file).

`/rtabmap/octomap_grid`
[image]

`/map`
[image]

Here is my revised rtabmap launch file

```
<launch>
  <arg name="database_path"     default="rtabmap.db"/>
  <arg name="rgbd_odometry"     default="false"/>
  <arg name="rtabmapviz"        default="false"/>
  <arg name="localization"      default="false"/>
  <arg name="simulation"        default="false"/>
  <arg name="sw_registered"     default="false"/>
  <arg     if="$(arg localization)" name="args"  default=""/>
  <arg unless="$(arg localization)" name="args"  default="--delete_db_on_start"/>

  <arg     if="$(arg simulation)" name="rgb_topic"   default="/camera/rgb/image_raw"/>
  <arg unless="$(arg simulation)" name="rgb_topic"   default="/camera/rgb/image_rect_color"/>
  <arg     if="$(arg simulation)" name="depth_topic" default="/camera/depth/image_raw"/>
  <arg unless="$(arg simulation)" name="depth_topic" default="/camera/depth_registered/image_raw"/>
  <arg name="camera_info_topic" default="/camera/rgb/camera_info"/>
  
  <arg name="wait_for_transform"  default="0.2"/> 
  <!-- 
      robot_state_publisher's publishing frequency in "turtlebot_bringup/launch/includes/robot.launch.xml" 
      can be increase from 5 to 10 Hz to avoid some TF warnings.
  -->
  
  <!-- Navigation stuff (move_base) -->
  <include unless="$(arg simulation)" file="$(find turtlebot_bringup)/launch/3dsensor.launch">
     <arg     if="$(arg sw_registered)" name="depth_registration" value="false"/>
     <arg unless="$(arg sw_registered)" name="depth_registration" value="true"/>
  </include>
  <include file="$(find turtlebot_navigation)/launch/includes/move_base.launch.xml"/>
  
  <!-- Mapping -->
  <group ns="rtabmap">

    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)">
	  <param name="database_path"       type="string" value="$(arg database_path)"/>
	  <param name="frame_id"            type="string" value="base_footprint"/>
	  <param name="wait_for_transform_duration"  type="double"   value="$(arg wait_for_transform)"/>
	  <param name="subscribe_depth"     type="bool"   value="true"/>
	  <param name="subscribe_scan"      type="bool"   value="false"/>
	  <param name="map_negative_poses_ignored" type="bool" value="true"/>
	
	  <!-- inputs -->
	  <remap from="scan"            to="/scan"/>
	  <remap from="rgb/image"       to="$(arg rgb_topic)"/>
  	  <remap from="depth/image"     to="$(arg depth_topic)"/>
  	  <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
	  <remap unless="$(arg rgbd_odometry)" from="odom" to="/odom"/>
  	  
  	  <!-- output -->
  	  <remap from="grid_map" to="/map"/>
	
	  <!-- RTAB-Map's parameters: do "rosrun rtabmap rtabmap (double-dash)params" to see the list of available parameters. -->
	  <param name="RGBD/ProximityBySpace"        type="string" value="true"/>   <!-- Local loop closure detection (using estimated position) with locations in WM -->
	  <param name="RGBD/OptimizeFromGraphEnd"    type="string" value="false"/>  <!-- Set to false to generate map correction between /map and /odom -->
	  <param name="Kp/MaxDepth"                  type="string" value="4.0"/>
	  <param name="Reg/Strategy"                 type="string" value="0"/>      <!-- Loop closure transformation: 0=Visual, 1=ICP, 2=Visual+ICP -->
[... 35 lines cut ...]
  	  <param name="subscribe_depth"             type="bool" value="true"/>
      <param name="subscribe_scan"              type="bool" value="false"/>
      <param name="frame_id"                    type="string" value="base_footprint"/>
      <param name="wait_for_transform_duration" type="double" value="$(arg wait_for_transform)"/>
    
      <remap from="rgb/image"       to="$(arg rgb_topic)"/>
      <remap from="depth/image"     to="$(arg depth_topic)"/>
      <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
      <remap from="scan"            to="/scan"/>
    </node>
    
  </group>
</launch>
```

I paste the snapshots of  `octomap_ground`, `octomap_obstacles` and `octomap_occupied_space`  here and I adjust the voxel size for better visualization.

`octomap_ground`
[image]

`octomap_obstacles`
[image]

`octomap_occupied_space`
[image]

matlabbe, 2019-03-23:
Hi,

`/rtabmap/octomap_grid` and `/rtabmap/grid_map` are 2D occupancy grid maps ([nav_msgs/OcccupancyGrid](http://docs.ros.org/melodic/api/nav_msgs/html/msg/OccupancyGrid.html)). If you want to see the acutal OctoMap in RVIZ, you should add the [octomap_rviz_plugins/ColorOccupancyGrid](http://wiki.ros.org/octomap_rviz_plugins) display, and select `/rtabmap/octomap_binary` topic (which is an [octomap_msgs/Octomap](http://docs.ros.org/api/octomap_msgs/html/msg/Octomap.html) msg).

Note that `/rtabmap/octomap_grid` is the projection of `/rtabmap/octomap_binary` in 2D.

cheers,
Mathieu

ken012git, 2019-03-28:
Thank you for your help. It does work well after changing the settings to what you suggested.

[image]

Thanks again!

closed 2019-03-28 by ken012git