introlab/rtabmap #376: Setting Odommetry Strategy to Okvis requires a ConfigPath [Standalone version]

ahmadhasan2k8, 2019-03-18:
I set the Odometry strategy to Okvis and try running the euroc_dataset. I get the following problem. Could you please tell me the proper way to run this?
```
./rtabmap-euroc_dataset .
Paths:
   Sequence number:  .
   Sequence path:    .
   Output:           .
   Output name:      rtabmap
   left images:      ./mav0/cam0/data
   right images:     ./mav0/cam1/data
   left calib:       ./mav0/cam0/sensor.yaml
   right calib:      ./mav0/cam1/sensor.yaml
   Ground truth:     ./mav0/state_groundtruth_estimate0/data.csv
   IMU:              ./mav0/imu0/data.csv
   Exposure Compensation: false
   Disparity:        false
   Raw images:       false
Parameters:
   Rtabmap/PublishRAMUsage=true
   Rtabmap/WorkingDirectory=.
RTAB-Map version: 0.19.0
Saved calibration "rtabmap_calib" to "."
baseToImu=xyz=0.000000,0.000000,0.000000 rpy=0.000000,-1.570796,0.000000
baseToCam0:
-0.0257745   0.00375622   0.999661     0.00981073   
-0.999557    -0.0149673   -0.0257155   0.064677     
0.0148655    -0.999881    0.0041402    -0.0216401   

baseToCam0=xyz=0.009811,0.064677,-0.021640 rpy=-1.566656,-0.014866,-1.596576
imuToCam0=xyz=-0.021640,-0.064677,0.009811 rpy=0.003757,0.025777,1.555925
imuToCam1=xyz=-0.019844,0.045369,0.007862 rpy=0.017907,0.025393,1.558237
No. IMU measurements: 36820
[ WARN] (2019-03-18 17:51:08.840) StereoCameraModel.cpp:257::load() Missing "essential_matrix" field in "./rtabmap_calib_pose.yaml"
[ WARN] (2019-03-18 17:51:08.840) StereoCameraModel.cpp:273::load() Missing "fundamental_matrix" field in "./rtabmap_calib_pose.yaml"
[ WARN] (2019-03-18 17:51:09.342) CameraImages.cpp:418::readPoses() 728 valid poses of 3682 stamps
Processing 3682 images...
[ERROR] (2019-03-18 17:51:09.407) OdometryOkvis.cpp:145::OdometryOkvis() OKVIS config file is empty (OdomOKVIS/ConfigPath)!
[ WARN] (2019-03-18 17:51:09.579) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ WARN] (2019-03-18 17:51:09.579) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ WARN] (2019-03-18 17:51:09.579) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ERROR] (2019-03-18 17:51:09.579) OdometryOkvis.cpp:288::computeTransform() OKVIS config file is empty (OdomOKVIS/ConfigPath)!
[ WARN] (2019-03-18 17:51:09.586) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ WARN] (2019-03-18 17:51:09.586) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ WARN] (2019-03-18 17:51:09.586) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ WARN] (2019-03-18 17:51:09.586) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ WARN] (2019-03-18 17:51:09.586) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ WARN] (2019-03-18 17:51:09.586) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ WARN] (2019-03-18 17:51:09.586) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ WARN] (2019-03-18 17:51:09.586) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ WARN] (2019-03-18 17:51:09.586) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ WARN] (2019-03-18 17:51:09.586) OdometryOkvis.cpp:212::computeTransform() Ignoring IMU, waiting for an image to initialize...
[ERROR] (2019-03-18 17:51:09.586) OdometryOkvis.cpp:288::computeTransform() OKVIS config file is empty (OdomOKVIS/ConfigPath)!

```

ahmadhasan2k8, 2019-03-22:
@matlabbe so, I tried changing Parameters.h s:
```
 RTABMAP_PARAM_STR(OdomOKVIS, ConfigPath,     "/home/ahasan/okvis/config/config_fpga_p2_euroc.yaml",  "Path of OKVIS config file.");
```
But still get the same error. 

Then I changed line 145 OdometryOkvis.cpp as:
```
#configFilename_ = uReplaceChar(configFilename_, '~', UDirectory::homeDir());
configFilename_ = /home/ahasan/okvis/config/config_fpga_p2_euroc.yaml;
```

And it works. Shouldn't the first change already take care of that? Is there any better way to pass the location of the yaml file?

matlabbe, 2019-03-23:
Yes there is a better way. All parameters in Parameters.h can be set dynamically on command line:
```bash
./rtabmap-euroc_dataset --OdomOKVIS/ConfigPath "/home/ahasan/okvis/config/config_fpga_p2_euroc.yaml" --Odom/Strategy 6
```

You can also look at more examples in this script: [run_euroc_datasets.sh](https://github.com/introlab/rtabmap/blob/master/docker/jfr2018/run_euroc_datasets.sh).

ahmadhasan2k8, 2019-03-25:
I see. Thanks. It works

closed 2019-03-25 by ahmadhasan2k8

jacklu333333, 2021-07-14:
> Yes there is a better way. All parameters in Parameters.h can be set dynamically on command line:
> 
> ```shell
> ./rtabmap-euroc_dataset --OdomOKVIS/ConfigPath "/home/ahasan/okvis/config/config_fpga_p2_euroc.yaml" --Odom/Strategy 6
> ```
> 
> You can also look at more examples in this script: [run_euroc_datasets.sh](https://github.com/introlab/rtabmap/blob/master/docker/jfr2018/run_euroc_datasets.sh).

Hi, I have same problem as the above. 
But right after, I command 
`rtabmap-euroc_dataset --OdomOKVIS/ConfigPath "/home/ahasan/okvis/config/config_fpga_p2_euroc.yaml" --Odom/Strategy 6
`
It outputs 
```
[ WARN] (2021-07-14 06:10:58.833) main.cpp:172::main() Ground truth file path doesn't exist: "6/mav0/state_groundtruth_estimate0/data.csv", benchmark values won't be computed.
Paths:
   Sequence number:  6
   Sequence path:    6
   Output:           6
   Output name:      rtabmap
   left images:      6/mav0/cam0/data
   right images:     6/mav0/cam1/data
   left calib:       6/mav0/cam0/sensor.yaml
   right calib:      6/mav0/cam1/sensor.yaml
   IMU:              6/mav0/imu0/data.csv
   IMU Filter:       1
   Exposure Compensation: false
   Disparity:        false
   Raw images:       false
Parameters:
   Odom/Strategy=6
   OdomOKVIS/ConfigPath=/root/okvis/config/config_fpga_p2_euroc.yaml
   Rtabmap/PublishRAMUsage=true
   Rtabmap/WorkingDirectory=6
RTAB-Map version: 0.20.7
terminate called after throwing an instance of 'YAML::BadFile'
  what():  yaml-cpp: error at line 0, column 0: bad file

Signal 6 caught...
Aborted (core dumped)
```
How can I resolve this issue? Also, what is the purpose of this config file?

Thanks in advance.

matlabbe, 2021-07-29:
The yaml file is used for okvis vio related parameters (tuned for the dataset or sensors used), loaded here 
https://github.com/introlab/rtabmap/blob/a5685c3e3104cb7f620fd5ad37e073521962b41e/corelib/src/odometry/OdometryOkvis.cpp#L287-L296

If the file really exists, it may be a permission issue causing this error:
```
terminate called after throwing an instance of 'YAML::BadFile'
  what():  yaml-cpp: error at line 0, column 0: bad file
```

OKVIS seems using opencv's yaml reader to read the file, the exception seems coming from there:
https://github.com/ethz-asl/okvis/blob/f0c9ba472b89f85030cd58dfbae03ceec7817679/okvis_common/src/VioParametersReader.cpp#L69-L81