introlab/rtabmap #377: Question: how are Aruco markers used?

lexvandersluijs, 2019-03-29:
Hi,

In the code I see some references to Aruco markers and April tags, but I can't seem to find information in the Wiki and website about how they are used. I can imagine a number of scenarios:
- as an aid in loop closure, as they can be identified more robustly than natural features
- as an aid to increase precision, as they can be localized in the image more precisely than natural features
- as an aid with relocalization, in case the system is used for tracking (e.g. for AR)
- or maybe they are 'just' reconstructed in 3D, to increase the ease of working with the scan-result

I'm planning to build the software from source myself, so that I can more easily follow the code-paths in Visual Studio. But if someone could answer this question in the meantime that would be very helpful.

matlabbe, 2019-03-29:
It is quite a new feature, it is why there is no official documentation yet. 

### Install:
To have the built-in option in RTAB-Map standalone, OpenCV should be built with [aruco](https://github.com/opencv/opencv_contrib/tree/master/modules/aruco) module from [opencv_contrib](https://github.com/opencv/opencv_contrib). To support apriltags2 dictionary, at least OpenCV 3.4.2 should be built.

### Usage:
In RTAB-Map, go to Preferences (advanced) -> Marker Detection panel, then check the checkbox to enable marker detection. Assumptions:

 * Markers have all the same size
 * Markers are static
 * Markers have all unique ids
 * Markers should have non-zero / non-negative IDs
 * If marker size parameter is not set, we should place the camera perpendicular (in front) of a marker to initialize the size. Once initialization is done, the markers can be detected in any orientations.

### Effects:

 * Markers are used like visual loop closure detection. When a previous marker is detected, a new constraint is added to graph and the map is optimized. 
 * Markers will be shown in 3D Map view as coordinate frames
 * Markers can be also used as "[anchors](https://github.com/introlab/rtabmap/wiki/Multi-Session-Mapping-with-RTAB-Map-Tango#1-scanning-tips)" to merge multiple sessions together.
 * They **are not** _yet_ used for visual odometry, only for loop closure detection and localization.

### ROS:
This built-in approach can be used with rtabmap_ros. There is also an [example](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_apriltags2.launch) of integration with [apriltags2_ros](http://wiki.ros.org/apriltags2_ros) if we want to use the external package instead of the built-in option, which may be more convenient on some systems if we don't want to rebuild another OpenCV with minimum version above. With the external approach, we can also define mosaics of tags that can be interesting for some applications.

cheers,
Mathieu

lexvandersluijs, 2019-03-30:
Hi Mathieu,

That sounds great, thanks!
I'm quite impressed by the amount of technology that has been integrated into RTAB-Map, nice work! 

I've got the Windows version compiling, building and running now. I had to make a few small changes in a few CMakeLists, what would be the best way to communicate them? In a thread like this or as a pull request?

By the way, I ran into a runtime issue, will start a new Issue for that one.

Cheers,
Lex

matlabbe, 2019-04-03:
A pull request is better to compare changes. For the windows build, you may also check this file to see if the configuration used could have fix your build errors:
https://github.com/introlab/rtabmap/blob/master/.appveyor.yml

If your changes are related to newer versions of the dependencies used in that build file, I'll be happy to integrate them.

cheers,
Mathieu

molinadavid, 2019-04-09:
Hi Mathieu,

Great work integrating the apriltags to rtabmap, I have been taking a look at it but I don't know how to confirm if the tags are actually helping for loop closures or not, if you could give me some pointers on how to actually confirm that the tags are working I would be very grateful.

I am attaching 3 databases for your convenience, 2 of those have the apriltags, and for the last one I covered the tags in order to test the differences.

[apriltags](https://www.dropbox.com/s/jd250d2kw9jmxf1/rtabmap_april.db?dl=0)
[apriltagsv2](https://www.dropbox.com/s/ovhsl3sjs085uam/rtabmap_april_v2.db?dl=0)
[notags](https://www.dropbox.com/s/gqy2u469hsv4c6r/rtabmap_april_notags.db?dl=0)

Best regards,
David

matlabbe, 2019-04-13:
If tags are detected, you would see coordinate frames on them in 3D Map view. In graph view, you would see dark green links and nodes with negative id (note that it is not yet implemented in database viewer). Also in loop closure detection view during mapping, we can see a green background on loop closure, but you should see an orange background when closing a loop on a landmark.

In your databases, I don't see any tags added to the graph. With some testings, it is because you are using TORO as optimizer and only GTSAM/G2O are supported with landmarks. After optimization, the landmarks disappeared from the graph view. Set Optimizer/Strategy to 1 (g2o) or 2 (gtsam). Also, don't set at the same time `Optimizer/Robust` and `RGBD/OptimizeMaxError`, they are not compatible. You may also set `Aruco/VarianceAngular` to 9999 if you find that orientation of the landmarks are poorly estimated.

Here is an example of what you should see when landmarks are detected and added to graph:
[image]
Note the orange background and the coordinate frames of landmark printed in images under loop closure detection view and the dark green links on the graph view.

cheers,
Mathieu

molinadavid, 2019-04-17:
Hi Mathieu,

Thank you very much for your response, I was able to confirm the presence of the tags as you suggested.

Best regards,
David

1fabrism, 2019-04-22:
Hey guys, I have a question about using this feature. I was wondering what happens if the markers are present in the environment but the MarkerDetection parameter is set to false (I'm talking specifically about rtabmap_ros). I imagine the markers would still be used by the software as useful features, especially if they are placed on an otherwise plain wall? If so, what exactly is the difference between turning the feature on or off? 
I'm planning to do tests to show how aruco markers can improve localization, so I was wondering if I can leave the markers hanging on the wall all the time and just turn the feature on and off, or if I should still make sure rtabmap is not using them "passively" by removing them from the environment altogether.

matlabbe, 2019-04-23:
rtabmap's visual loop closure detection extracts visual features everywhere on the image. As the markers have some contrast, visual features may be extracted indirectly from the markers and used for visual loop closure detection.

When we enable `RGBD/MarkerDetection` parameter, another process (i.e., [ArUco marker detection from OpenCV](https://github.com/opencv/opencv_contrib/tree/master/modules/aruco)) is executed to find markers in the image. When detected, they add a new type of constraints in the graph. Like visual loop closure constraints, those "landmark" constraints are used for loop closure detection and graph optimization.

If you disable `RGBD/MarkerDetection`, rtabmap may still extract visual features on them, so indirectly affect visual loop closure detection performance. It is like adding artificial visual features to environment. You may cover them with a blank paper if you really don't want visual loop closure detection affected by them. 

Finally, if you have want a scenario where we disable visual loop closure detection and use only marker detection, `Kp/MaxFeatures` can be set to `-1` to disable visual loop closure detection.

cheers,
Mathieu

molinadavid, 2019-05-15:
Hi Mathieu,

As always thank you very much for all your work. We are currently trying to test using the apriltags2 as an external package as your [example](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_apriltags2.launch), the issue we have at the moment is that when I visualize the mapping process the tf of the tag appears on the map but the TF is wrong making the tag appear on the Z axis of the robot rather than the on the front of it.

Please see this 2 screenshots for an example:

[image]
[image]

the database created is here
[rtabmap_externalTags.db](https://www.dropbox.com/s/oxlg1pn29j1wvnh/rtabmap_externalTags.db?dl=0)

Best regards,
David

matlabbe, 2019-05-15:
Is `Tag_3` the one we see in the camera feed on left? If so, the TF may appear correct from the apriltag output, referred in camera optical frame. In TF, it seems the camera is looking upward. You may change the `camera_frame` parameter in the appriltags2_ros example:
```xml
<arg name="camera_frame_id" default="camera_color_optical_frame"/>
```
to maybe `camera_link`.

I looked in the database and there are no landmark links. Is rtabmap connected to apriltags output like in the following screenshot?
[image]

cheers,
Mathieu

szx0112, 2021-09-08:
Hi @matlabbe , it seems the link of the example file is gone. Could you update the link? I'd like to try the marker detection function in my project. Thanks.

matlabbe, 2021-09-09:
See https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_apriltag_ros.launch

szx0112, 2021-09-27:
Thanks for the link. Just a follow up question. Is it possible to pre-upload the marker positions? For example, we already have the ground truth poses of the markers or the marker poses are computed from a previous mapping using RTABMAP. We can extract and save it for future usage.

matlabbe, 2021-09-28:
See http://official-rtab-map-forum.206.s1.nabble.com/How-to-improve-mapping-accuracy-based-on-ArUco-identification-code-tp8393p8497.html, it is not actually implemented in rtabmap, but that post shows a way to implement it.

ZitongLan, 2024-01-16:
Hello, I am new to rtabmap. and I have a .db file containing tag detection results. And I could use it to visualize the tag in the 3d graph view. Where is the tags' information stored? How can I get the tags location from this .db file (with their ID)
I attached a .db file for here.
https://drive.google.com/file/d/1bjzE68mvXHUCNhVxd3hRkiu6eC-2NTSz/view?usp=drive_link

matlabbe, 2024-01-20:
See https://github.com/introlab/rtabmap/issues/1199

takahiko-hasegawa-hdjp, 2024-02-16:
Does this mean that AR maker can only be used in creating a map? and not used for map matching?

matlabbe, 2024-02-17:
Well, it depends how you want to do map matching. We can combine two maps together with same tags detected in each other like this:
```
rtabmap-reprocess "map1.db;map2.db" output.db
```
The two maps will be merged together using tag constraints.

takahiko-hasegawa-hdjp, 2024-02-18:
Sorry when I said "map matching", i mean visual localization. During the localzation phase, even If the tags are stored in the map (.db) and all the tags are still located in the same place in the environment, it would not help localization with AR marker capabilities ON?

matlabbe, 2024-02-19:
The tags are also used in localization mode to relocalize the robot.

shubham-shahh, 2025-06-19:
> It is quite a new feature, it is why there is no official documentation yet.
>
> ### Install:
> To have the built-in option in RTAB-Map standalone, OpenCV should be built with [...]

Hi,
Thanks for the explaination, is there an example for RTAB + Aruco in ROS2?
Thanks!

matlabbe, 2025-06-20:
If rtabmap library has been built with opencv+arcuo module, then you could enable it internally with `RGBD/MarkerDetection` and setting the right dictionary `Marker/Dictionary`:
```
rtabmap --params | grep Marker
Param: Marker/CornerRefinementMethod = "0"                 [Corner refinement method (0: None, 1: Subpixel, 2:contour, 3: AprilTag2). For OpenCV <3.3.0, this is "doCornerRefinement" parameter: set 0 for false and 1 for true.]
Param: Marker/Dictionary = "0"                             [Dictionary to use: DICT_ARUCO_4X4_50=0, DICT_ARUCO_4X4_100=1, DICT_ARUCO_4X4_250=2, DICT_ARUCO_4X4_1000=3, DICT_ARUCO_5X5_50=4, DICT_ARUCO_5X5_100=5, DICT_ARUCO_5X5_250=6, DICT_ARUCO_5X5_1000=7, DICT_ARUCO_6X6_50=8, DICT_ARUCO_6X6_100=9, DICT_ARUCO_6X6_250=10, DICT_ARUCO_6X6_1000=11, DICT_ARUCO_7X7_50=12, DICT_ARUCO_7X7_100=13, DICT_ARUCO_7X7_250=14, DICT_ARUCO_7X7_1000=15, DICT_ARUCO_ORIGINAL = 16, DICT_APRILTAG_16h5=17, DICT_APRILTAG_25h9=18, DICT_APRILTAG_36h10=19, DICT_APRILTAG_36h11=20]
Param: Marker/Length = "0"                                 [The length (m) of the markers' side. 0 means automatic marker length estimation using the depth image (the camera should look at the marker perpendicularly for initialization).]
Param: Marker/MaxDepthError = "0.01"                       [Maximum depth error between all corners of a marker when estimating the marker length (when Marker/Length is 0). The smaller it is, the more perpendicular the camera should be toward the marker to initialize the length.]
Param: Marker/MaxRange = "0.0"                             [Maximum range in which markers will be detected. <=0 for unlimited range.]
Param: Marker/MinRange = "0.0"                             [Miniminum range in which markers will be detected. <=0 for unlimited range.]
Param: Marker/Priors = ""                                  [World prior locations of the markers. The map will be transformed in marker's world frame when a tag is detected. Format is the marker's ID followed by its position (angles in rad), markers are separated by vertical line ("id1 x y z roll pitch yaw|id2 x y z roll pitch yaw"). Example:  "1 0 0 1 0 0 0|2 1 0 1 0 0 1.57" (marker 2 is 1 meter forward than marker 1 with 90 deg yaw rotation).]
Param: Marker/PriorsVarianceAngular = "0.001"              [Angular variance to set on marker priors.]
Param: Marker/PriorsVarianceLinear = "0.001"               [Linear variance to set on marker priors.]
Param: Marker/VarianceAngular = "0.01"                     [Angular variance to set on marker detections. If Marker/VarianceOrientationIgnored is enabled, it is ignored with Optimizer/Strategy=1 (g2o) and it corresponds to bearing variance with Optimizer/Strategy=2 (GTSAM).]
Param: Marker/VarianceLinear = "0.001"                     [Linear variance to set on marker detections. If Marker/VarianceOrientationIgnored is enabled and Optimizer/Strategy=2 (GTSAM): it is the variance of the range factor, with 9999 to disable range factor and to do only bearing.]
Param: Marker/VarianceOrientationIgnored = "false"         [When this setting is false, the landmark's orientation is optimized during graph optimization. When this setting is true, only the position of the landmark is optimized. This can be useful when the landmark's orientation estimation is not reliable. Note that for Optimizer/Strategy=1 (g2o), only Marker/VarianceLinear needs be set if we ignore orientation. For Optimizer/Strategy=2 (GTSAM), instead of optimizing the landmark's position directly, a bearing/range factor is used, with Marker/VarianceLinear as the variance of the range factor (with 9999 to optimize the position with only a bearing factor) and Marker/VarianceAngular as the variance of the bearing factor (pitch/yaw).]
Param: RGBD/MarkerDetection = "false"                      [Detect static markers to be added as landmarks for graph optimization. If input data have already landmarks, this will be ignored. See "Marker" group for parameters.]
```

Otherwise, if you are using `apriltag_msgs`, the rtabmap_slam node subscribe to a `tag_detections` topic, of type `apriltag_msgs/msg/AprilTagDetectionArray`.

Which package do you use for Aruco? We could probably at the compatibility directly in rtabmap_slam. It seems there are [aruco_msgs](https://github.com/pal-robotics/aruco_ros) and [aruco_opencv_msgs](https://github.com/fictionlab/ros_aruco_opencv) packages on all ros2 distros, and [aruco_marker_msgs](https://github.com/namo-robotics/aruco_markers) on humble only.

matlabbe, 2025-06-29:
I added support for many aruco pkgs in https://github.com/introlab/rtabmap_ros/pull/1334

For your questions:
1. I don't have strong opinions on this, I used one or the other in the past. Orientation accuracy is always the pain point for both of them, but generally is more a resolution and/or tag size issue.
2. If you are in localization mode, you should be able to reset odometry whenever you want and rtabmap won't change the map used. In SLAM mode, rtabmap will indeed start a new session if it detects odometry has been reset. In pure visual odometry, it is generally a good thing because continuing the same map can add very large errors in it. If you are combing vo with another kind of odometry (e.g., using `guess_frame_id` like wheel odometry) that could be relatively accurate while vo is lost, you can set `'publish_null_when_lost': true` with `'OdomResetCountdown': '5'` and it won't start a new session.
3. Yes, localization mode, the initial pose topic is in map frame of the map inside the database.
4. Yes, you can set `'subscribe_scan': true` with `'Reg/Strategy': '1'` and `'RGBD/NeighborLinkRefining': 'true'` for rtabmap node. [Here](https://github.com/introlab/rtabmap_ros/tree/ros2/rtabmap_demos#turtlebot3-nav2-2d-lidar-and-rgb-d-slam) an example of that on turtlebot3.