introlab/rtabmap #378: GTSAM 3.2.0 with rtabmap compiled succesfully and installed.However the rtabmap --version with no GTSAM.

Arunsid, 2019-03-29:
[image]

Compiled succesfully RTAB with GTSAM,g20 after the boost::timer error with Cmakelist of rtab.
I followed the https://github.com/introlab/rtabmap_ros#installation.  Installed the build from source.However after the after the succesfull catkin_make -j1 from the workspace,when checked the rtab map version,could find with no GTSAM.

Following output results of ---> catkin_make -j1
Base path: /home/user1/catkin_ws
Source space: /home/user1/catkin_ws/src
Build space: /home/user1/catkin_ws/build
Devel space: /home/user1/catkin_ws/devel
Install space: /home/user1/catkin_ws/install
####
#### Running command: "cmake /home/user1/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/user1/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/user1/catkin_ws/install -G Unix Makefiles" in "/home/user1/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/user1/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/user1/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/user1/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/user1/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - beginner_tutorials
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'beginner_tutorials'
-- ==> add_subdirectory(beginner_tutorials)
-- +++ processing catkin package: 'rtabmap_ros'
-- ==> add_subdirectory(rtabmap_ros)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at rtabmap_ros/CMakeLists.txt:28 (find_package):
  By not providing "Findapriltags2_ros.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "apriltags2_ros", but CMake did not find one.

  Could not find a package configuration file provided by "apriltags2_ros"
  with any of the following names:

    apriltags2_rosConfig.cmake
    apriltags2_ros-config.cmake

  Add the installation prefix of "apriltags2_ros" to CMAKE_PREFIX_PATH or set
  "apriltags2_ros_DIR" to a directory containing one of the above files.  If
  "apriltags2_ros" provides a separate development package or SDK, be sure it
  has been installed.

-- Found RTABMap: /usr/local/lib/librtabmap_core.so;/usr/local/lib/librtabmap_utilite.so;/usr/local/lib/librtabmap_gui.so;/usr/lib/x86_64-linux-gnu/libz.so;/opt/ros/kinetic/lib/libg2o_core.so;/opt/ros/kinetic/lib/libg2o_types_slam2d.so;/opt/ros/kinetic/lib/libg2o_types_slam3d.so;/opt/ros/kinetic/lib/libg2o_types_sba.so;/opt/ros/kinetic/lib/libg2o_stuff.so;/opt/ros/kinetic/lib/libg2o_solver_csparse.so;/opt/ros/kinetic/lib/libg2o_csparse_extension.so;/usr/lib/x86_64-linux-gnu/libcxsparse.so;/opt/ros/kinetic/lib/libg2o_solver_cholmod.so;/usr/lib/x86_64-linux-gnu/libcholmod.so;/usr/lib/x86_64-linux-gnu/libfreenect.so;/usr/lib/x86_64-linux-gnu/libfreenect_sync.so;/opt/ros/kinetic/lib/liboctomap.so;/opt/ros/kinetic/lib/liboctomath.so (Required is at least version "0.19.1") 
-- Found OpenCV: /opt/ros/kinetic (found version "3.3.1") 
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found eigen: /usr/include/eigen3  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
[... 400 lines cut ...]
[ 93%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_wifi_signal_pub.dir/src/WifiSignalPubNode.cpp.o
[ 94%] Linking CXX executable /home/user1/catkin_ws/devel/lib/rtabmap_ros/wifi_signal_pub
[ 94%] Built target rtabmap_wifi_signal_pub
Scanning dependencies of target rtabmapviz
[ 94%] Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/GuiNode.cpp.o
[ 95%] Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/GuiWrapper.cpp.o
[ 95%] Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/PreferencesDialogROS.cpp.o
[ 96%] Linking CXX executable /home/user1/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz
[ 96%] Built target rtabmapviz
Scanning dependencies of target rtabmap_stereo_camera
[ 97%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_stereo_camera.dir/src/StereoCameraNode.cpp.o
[ 97%] Linking CXX executable /home/user1/catkin_ws/devel/lib/rtabmap_ros/stereo_camera
[ 97%] Built target rtabmap_stereo_camera
Scanning dependencies of target rtabmap_rgbd_relay
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_rgbd_relay.dir/src/RGBDRelayNode.cpp.o
[ 98%] Linking CXX executable /home/user1/catkin_ws/devel/lib/rtabmap_ros/rgbd_relay
[ 98%] Built target rtabmap_rgbd_relay
Scanning dependencies of target rtabmap_icp_odometry
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_icp_odometry.dir/src/ICPOdometryNode.cpp.o
[100%] Linking CXX executable /home/user1/catkin_ws/devel/lib/rtabmap_ros/icp_odometry
[100%] Built target rtabmap_icp_odometry
Scanning dependencies of target rtabmap_map_assembler
[100%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_map_assembler.dir/src/MapAssemblerNode.cpp.o
[100%] Linking CXX executable /home/user1/catkin_ws/devel/lib/rtabmap_ros/map_assembler
[100%] Built target rtabmap_map_assembler

matlabbe, 2019-03-29:
The version is 0.17.6, which is the binary version of kinetic. Make sure to uninstall rtabmap and rtabmap_ros binaries so that the version built from source (with GTSAM) is used instead:
```bash
$ sudo apt-get remove ros-kinetic-rtabmap ros-kinetic-rtabmap-ros
```

Arunsid, 2019-03-30:
Thanks for the info.I tried to compile succesfully the rtabmap and include rtabmap_ros as depency in my new catkin_packages .However
For example when i created a ros package,    catkin_create_pkg new_pkg roscpp rtabmap_ros std_msgs  (rtabmap_ros as dependency,to use rtabmap_ros related datas) it was succesful.And indeed i was able to work with header files from eg: ( rtabmap_ros/MapGraph.h).

But as i include zed 2.7 with cuda 9.0 ,and i try to recompile the catkin_package ,i get the error that the rtabmap_ros  is not compiled properly.

[image]

Arunsid, 2019-03-30:
i get these error only when i compile with cuda 9.0 and zed sdk.if not catkin_packages with rtabmap_ros depency works fine.Hopes the problem with the integration of cuda and zed sdk with rtabmap.

matlabbe, 2019-04-03:
This is a problem coming from the linker imports from zed/cuda, which they define `-lpthread` to be linked. It seems it has been resolved in recent cmake versions: https://answers.ros.org/question/311685/cmake-compile-error-pthread-not-found-how-to-tell-the-compiler-to-look-for-pthread-lib/

Without changing cmake version, you may also try editing the CMakeCache.txt of catkin_ws/build directory to replace all `-lpthread` by `pthread`.

cheers,
Mathieu