introlab/rtabmap #379: Windows build runtime errors "Cannot find actor named"

lexvandersluijs, 2019-03-30:
Hi,

I built RTabMap for Windows, using latest source from git and the prebuilt dependencies listed on the wiki, except for OpenCV 3.3, boost 1.60, ZED SDK and librealsense which I already had. 
Also installed the Kinect 2 SDK downloaded from Microsoft and the OpenNI library from Occipital.

I had to hunt for the PSAPI lib a bit, found it in the Windows SDK, C:\Program Files (x86)\Windows Kits\10\Lib\10.0.17134.0\um\x64\psapi.lib

Compiler is Visual Studio 2017, target is x64. This is the log from CMake

```
Info :
  Version : 0.19.1
  CMAKE_INSTALL_PREFIX = C:/Program Files (x86)/RTABMap
  CMAKE_BUILD_TYPE =     
  CMAKE_INSTALL_LIBDIR = lib
  BUILD_APP =            ON
  BUILD_TOOLS =          ON
  BUILD_EXAMPLES =       ON
  BUILD_AS_BUNDLE =      OFF
  CMAKE_CXX_FLAGS = /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MP8  -openmp
  FLANN_KDTREE_MEM_OPT = OFF
  PCL_DEFINITIONS =   -D__SSSE3__ -D__SSE2__ -D__SSE__;-DBOOST_ALL_NO_LIB;-DDISABLE_OPENNI2;-DDISABLE_PCAP;-DDISABLE_PNG
Optional dependencies ('*' affects some default parameters) :
 *With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
  With external SQLite3     = NO (sqlite3 not found, internal version is used for convenience)
  With Freenect             = NO (libfreenect not found)
  With OpenNI2              = YES (License: Apache v2)
  With Freenect2            = NO (libfreenect2 not found)
  With Kinect for Windows 2 = YES (License: Apache v2 and/or GPLv2)
  With dc1394               = NO (WITH_DC1394=OFF)
  With FlyCapture2/Triclops = NO (WITH_FLYCAPTURE2=OFF)
  With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
 *With g2o                  = YES (License: BSD)
 *With GTSAM                = YES (License: BSD)
  With VERTIGO              = YES (License: GPLv3)
  With cvsba                = NO (cvsba not found)
 *With libpointmatcher      = NO (libpointmatcher not found)
  With loam_velodyne        = NO (loam_velodyne not found)
  With ZED                  = YES (With CUDA)
  With RealSense            = YES (License: Apache-2)
  With RealSenseSlam        = NO (WITH_REALSENSE_SLAM=OFF)
  With RealSense2           = NO (librealsense2 not found)
  With OCTOMAP              = NO (octomap not found)
  With CPUTSDF              = NO (CPUTSDF not found)
  With OpenChisel           = NO (open_chisel not found)
  With libfovis             = NO (libfovis not found)
  With libviso2             = NO (libviso2 not found)
  With dvo_core             = NO (dvo_core not found)
  With okvis                = NO (okvis not found)
  With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
  With VINS-Fusion          = NO (VINS-Fusion not found)
  With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
  With Qt5                  = YES (License: Open Source or Commercial)
```
After copying some DLL's into the bin folder that has the RTAB-Map executable, and selecting a sensor like RealSense R200 or ZED Mini, the software appears to start OK: an image and something like a pointcloud appears. Then the processing seems to halt, and these messages appear in the log:

```
[ERROR] (2019-03-30 19:59:55) CloudViewer.cpp:2558::rtabmap::CloudViewer::setCloudVisibility() Cannot find actor named "featuresOdom".
[ERROR] (2019-03-30 19:59:56) CloudViewer.cpp:2558::rtabmap::CloudViewer::setCloudVisibility() Cannot find actor named "cloud1".
[ERROR] (2019-03-30 19:59:56) CloudViewer.cpp:2558::rtabmap::CloudViewer::setCloudVisibility() Cannot find actor named "featuresOdom".
```

Any idea what could be the solution for this?

Cheers,
Lex

lexvandersluijs, 2019-04-01:
Update: I built a Debug version (w g2o and gtsam disabled as these were giving trouble) and noticed that there was an error originating from Qt in the construction of CloudViewer (or something to that effect, don't have it in front of me right now), so that's a clue.

Which version of Qt are you using right now?

Note: when I tried to rebuild with Qt 5.12, the latest right now, I got a dynamic link error for a function in QTreeView, so I suppose it would be a little older than that.

lexvandersluijs, 2019-04-01:
Another update..

When the exception occurs in the Debug version I was suddenly struck by the fact that references to Qt5Widgets.dll appear. That's the non-debug version of Qt. We are also linking to the debug-version of Qt, as evidenced by the list of loaded modules

```
Qt5Widgetsd.dll	Qt5Widgetsd.dll	C:\SDKs\Qt\5.10.0\5.10.0\msvc2017_64\bin\Qt5Widgetsd.dll	N/A	Yes	Cannot find or open the PDB file.		12	5.10.0.0	11/30/2017 7:02 PM	0000000070DD0000-0000000071900000	[70852] RTABMap.exe		
Qt5Cored.dll	Qt5Cored.dll	C:\SDKs\Qt\5.10.0\5.10.0\msvc2017_64\bin\Qt5Cored.dll	N/A	Yes	Cannot find or open the PDB file.		15	5.10.0.0	11/30/2017 6:45 PM	00000000700E0000-0000000070DC6000	[70852] RTABMap.exe		
Qt5Svgd.dll	Qt5Svgd.dll	C:\SDKs\Qt\5.10.0\5.10.0\msvc2017_64\bin\Qt5Svgd.dll	N/A	Yes	Cannot find or open the PDB file.		53	5.10.0.0	11/30/2017 7:18 PM	0000000070020000-00000000700D9000	[70852] RTABMap.exe		
Qt5PrintSupportd.dll	Qt5PrintSupportd.dll	C:\SDKs\Qt\5.10.0\5.10.0\msvc2017_64\bin\Qt5PrintSupportd.dll	N/A	Yes	Cannot find or open the PDB file.		54	5.10.0.0	11/30/2017 7:03 PM	00007FFD33220000-00007FFD332C9000	[70852] RTABMap.exe		
vtkGUISupportQt-6.3.dll	vtkGUISupportQt-6.3.dll	C:\Projects\libraries\rtabmap\bin\vtkGUISupportQt-6.3.dll	N/A	Yes	Binary was not built with debug information.		65		7/9/2018 8:33 PM	00007FFD498E0000-00007FFD4991F000	[70852] RTABMap.exe		
Qt5Guid.dll	Qt5Guid.dll	C:\SDKs\Qt\5.10.0\5.10.0\msvc2017_64\bin\Qt5Guid.dll	N/A	Yes	Cannot find or open the PDB file.		89	5.10.0.0	11/30/2017 6:54 PM	00007FFCE4630000-00007FFCE5415000	[70852] RTABMap.exe		
Qt5Widgets.dll	Qt5Widgets.dll	C:\SDKs\Qt\5.10.0\5.10.0\msvc2017_64\bin\Qt5Widgets.dll	N/A	Yes	Cannot find or open the PDB file.		133	5.10.0.0	11/30/2017 7:02 PM	000000006FAD0000-0000000070019000	[70852] RTABMap.exe		
Qt5Core.dll	Qt5Core.dll	C:\SDKs\Qt\5.10.0\5.10.0\msvc2017_64\bin\Qt5Core.dll	N/A	Yes	Cannot find or open the PDB file.		134	5.10.0.0	11/30/2017 6:46 PM	000000006F520000-000000006FAC5000	[70852] RTABMap.exe		
Qt5Gui.dll	Qt5Gui.dll	C:\SDKs\Qt\5.10.0\5.10.0\msvc2017_64\bin\Qt5Gui.dll	N/A	Yes	Cannot find or open the PDB file.		135	5.10.0.0	11/30/2017 6:54 PM	00007FFCFE7C0000-00007FFCFEDCC000	[70852] RTABMap.exe		
```
As I understand it, this is not a scenario supported by Qt, right? 

To check: is it currently possible to build and run the Debug version of RTAB-Map on Windows, or would some refactoring of the CMakeLists be required?

lexvandersluijs, 2019-04-02:
Another update: I thought I'd built VTK 6.3 in Debug mode so that I can troubleshoot what's going on in more detail, but when I configure it in CMake and direct it to built with Qt 5.9.10 this message appears:

```
CMake Error at C:/Program Files/CMake/share/cmake-3.14/Modules/FindQt4.cmake:1314 (message):
  Found unsuitable Qt version "5.10.0" from
  C:/SDKs/Qt/5.10.0/5.10.0/msvc2017_64/bin/qmake.exe, this code requires Qt
  4.x
Call Stack (most recent call first):
  GUISupport/Qt/CMakeLists.txt:71 (find_package)
```
Clearly, mixing versions of Qt in a single executable is not possible, so now I'm puzzled as to how this was done in the ready-made binaries.. Any hints?

lexvandersluijs, 2019-04-03:
Ok, so I built a Debug version of VTK which properly links to the Debug version of Qt 5.10.0. Then RTAB-Map.exe starts OK. There is no g2o or GTSAM and it runs slowly, but _it runs_.  

In other words: no conflict between the debug and release versions of the Qt dlls anymore, and basic setup of the development environment looks OK.

Then, using the same set of dependencies in Release mode (and manually copying the VTK .lib and .dll files from the lib and bin Release folders into their respective parent folder, AND copying the .dlls into the bin folder of RTAB-Map, just to be sure.. fixing CMakeLists upstream would come later..), I build and run the Release version and the "Cannot find actor named..." error messages appear again :(.  

There are no obvious '#if' or '#ifdef' statements to be found that say 'if Release build then...' or 'if dependency x is not present then...'

So I'm at a bit of a loss right now.. I really want to avoid also building PCL from scratch.. So if you have any idea, it would be greatly appreciated!

matlabbe, 2019-04-04:
I cannot really tell about issues when using latest versions of the dependencies. We tried many configurations and the one working at the moment is using some fixed versions of the dependencies. I would refer you to this file: https://github.com/introlab/rtabmap/blob/master/.appveyor.yml
In this file, RTAB-Map is built against most dependencies in Release mode. We don't have debug binaries for all dependencies, so I cannot really help on this either. 

cheers,
Mathieu

lexvandersluijs, 2019-04-05:
Hi Mathieu,

Thanks for your comment, it set me on the right track. I noticed in the mentioned yml file that all the dependencies are built for VS 2015, and I was using 2017. Now, if all the DLL's had a C-style interface this would not have mattered, but in the case of Boost and others they expose C++ classes and probably they are passing objects of these types among each other as well. This can of course cause trouble if they use different versions of the STL / are built with different compiler versions.

So when I installed VS2015 and made sure all the dependencies were using this too, it started working for me :-) 
I had some hangs with the ZED and ZED Mini camera, but after upgrading this SDK to the latest version (and CUDA, and their firmware) this now also works flawlessly.

The Tara camera does not work for me right now: when entering the calibration dialog I see a garbled image, my guess is that something has changed in their firmware in the meantime. Maybe I'll look at this later, but considering that the FOV of the lenses on our camera is a bit narrow, this does not have a large priority.

In any event, this issue can be closed. Thanks for your feedback again.
Lex

matlabbe, 2019-04-10:
PCL provides some binaries for msvc2017: https://github.com/PointCloudLibrary/pcl/releases
Most rtabmap dependencies are built with that package, at least with the basic options you may get rtabmap working. I did not try it though.

For the Tara driver, you may ask @e-consystemgit, who [added](https://github.com/introlab/rtabmap/pull/331) the tara support in rtabmap.

cheers,
Mathieu

closed 2019-04-10 by matlabbe