introlab/rtabmap #380: Make errors

Williamc17, 2019-04-09:
I get this error after installing **successfully** VINS-Mono and VINS-Fusion on Ubuntu 16.04, ROS Kinetic.

~~~
~/rtabmap/build$ make 
[  4%] Built target rtabmap_utilite
[  5%] Built target res_tool
[  5%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o
[  5%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o
[  5%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBDriverSqlite3.cpp.o
[  5%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/DBReader.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Recovery.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Camera.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraModel.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraFreenect.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraFreenect2.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraImages.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraK4W2.cpp.o
[  8%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenni.cpp.o
[  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenNI2.cpp.o
[  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraOpenNICV.cpp.o
[  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRealSense.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRealSense2.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRGBDImages.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraStereoDC1394.cpp.o
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraStereoFlyCapture2.cpp.o
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraStereoImages.cpp.o
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraStereoVideo.cpp.o
[ 12%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraStereoZed.cpp.o
In file included from /usr/local/zed/include/sl/Camera.hpp:1:0,
                 from /home/william/rtabmap/corelib/src/camera/CameraStereoZed.cpp:34:
/usr/local/zed/include/sl_zed/Camera.hpp: In constructor ‘sl::InitParameters::InitParameters(sl::RESOLUTION, int, int, sl::String, bool, sl::DEPTH_MODE, sl::UNIT, sl::COORDINATE_SYSTEM, bool, int, float, bool, bool, bool, int, sl::String, int, CUcontext, sl::InputType, sl::String)’:
/usr/local/zed/include/sl_zed/Camera.hpp:250:14: warning: ‘sl::InitParameters::sdk_verbose’ will be initialized after [-Wreorder]
         bool sdk_verbose;
              ^
/usr/local/zed/include/sl_zed/Camera.hpp:241:18: warning:   ‘CUdevice sl::InitParameters::sdk_gpu_id’ [-Wreorder]
         CUdevice sdk_gpu_id;
                  ^
/usr/local/zed/include/sl_zed/Camera.hpp:328:9: warning:   when initialized here [-Wreorder]
         InitParameters(RESOLUTION camera_resolution_ = RESOLUTION_HD720,
         ^
/usr/local/zed/include/sl_zed/Camera.hpp:241:18: warning: ‘sl::InitParameters::sdk_gpu_id’ will be initialized after [-Wreorder]
         CUdevice sdk_gpu_id;
                  ^
/usr/local/zed/include/sl_zed/Camera.hpp:218:15: warning:   ‘float sl::InitParameters::depth_minimum_distance’ [-Wreorder]
         float depth_minimum_distance;
               ^
/usr/local/zed/include/sl_zed/Camera.hpp:328:9: warning:   when initialized here [-Wreorder]
         InitParameters(RESOLUTION camera_resolution_ = RESOLUTION_HD720,
         ^
/usr/local/zed/include/sl_zed/Camera.hpp:218:15: warning: ‘sl::InitParameters::depth_minimum_distance’ will be initialized after [-Wreorder]
         float depth_minimum_distance;
               ^
In file included from /usr/local/zed/include/sl/Camera.hpp:1:0,
                 from /home/william/rtabmap/corelib/src/camera/CameraStereoZed.cpp:34:
/usr/local/zed/include/sl_zed/Camera.hpp:146:14: warning:   ‘bool sl::InitParameters::camera_disable_self_calib’ [-Wreorder]
         bool camera_disable_self_calib;
              ^
In file included from /usr/local/zed/include/sl/Camera.hpp:1:0,
                 from /home/william/rtabmap/corelib/src/camera/CameraStereoZed.cpp:34:
/usr/local/zed/include/sl_zed/Camera.hpp:328:9: warning:   when initialized here [-Wreorder]
         InitParameters(RESOLUTION camera_resolution_ = RESOLUTION_HD720,
         ^
In file included from /usr/local/zed/include/sl/Camera.hpp:1:0,
                 from /home/william/rtabmap/corelib/src/camera/CameraStereoZed.cpp:34:
/usr/local/zed/include/sl_zed/Camera.hpp:146:14: warning: ‘sl::InitParameters::camera_disable_self_calib’ will be initialized after [-Wreorder]
         bool camera_disable_self_calib;
              ^
/usr/local/zed/include/sl_zed/Camera.hpp:136:13: warning:   ‘int sl::InitParameters::camera_image_flip’ [-Wreorder]
         int camera_image_flip;
[... 231 lines cut ...]
		${vins_LIBRARIES}
#/home/william/catkin_ws/src/VINS-Fusion
		${LIBRARIES}
	)
#    MESSAGE(STATUS "INCLUDE_DIRS: ${INCLUDE_DIRS}")
#    MESSAGE(STATUS "vins_LIBRARIES: ${LIBRARIES}")
ENDIF(vins_FOUND)
~~~

The third question is,  **JUST ONE** version of PCL in my machine 'cos ROS Kinetic was fully desktop installed. But rtabmap still cannot find library of pcl.

~~~
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libpng.so', needed by '../bin/librtabmap_core.so.0.19.1'.  Stop.
CMakeFiles/Makefile2:1586: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
tools/ExtractObject/CMakeFiles/extractObject.dir/build.make:62: recipe for target 'tools/ExtractObject/CMakeFiles/extractObject.dir/main.cpp.o' failed
make[2]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/main.cpp.o] Interrupt
CMakeFiles/Makefile2:1970: recipe for target 'tools/ExtractObject/CMakeFiles/extractObject.dir/all' failed
make[1]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/all] Interrupt
Makefile:160: recipe for target 'all' failed
make: *** [all] Interrupt
~~~

Thank you for answering my questions.

matlabbe, 2019-04-10:
Hi,

For the PCL errors, it seems you did previously build rtabmap with PCL installed in /usr/local. You may delete the CMakeCache.txt in rtabmap build directory and catkin build directory to refresh the paths. Then rebuild.

For VINS-Fusion, it is not required for rtabmap. you can do `cmake -DWITH_VINS=OFF ..` in rtabmap/build directory to disable it. If you want to use it in rtabmap, there is a patch to apply to VINS-Fusion repo so that the header files can be found by cmake. See this patch: https://gist.github.com/matlabbe/795ab37067367dca58bbadd8201d986c

cheers,
Mathieu

Williamc17, 2019-04-10:
Hi, Mathieu

* For VINS, I would like do 3D mapping (6DOF) on rtabmap using VIO approach (VINS-Mono or VINS-Fusion) with zed-mini or Realsense D435i. May I get some tutorials about how to do this?

* Can rtabmap_ros do dense mapping in this case because VINS produces sparse map rather than a dense map? For now, I have not installed rtabmap_ros yet. 

* There is a question, It seems 3D dense mapping of 6DOF is bad in my test environment (a biped climbing robot on the truss with a sensor suite). Every step motion of the robot is on a plane, but the whole trajectory is still 6DOF. **How can I set the parameters to 6DOF? (For now, I use default settings in rtabmap)**. Moreover, I'll be appreciated If you have any ideas about how to modify the source code to constrain every movement on a plane.

Just some help and it will make a big difference to me.

Welcome to help me by emailing vlinwilliam@gmail.com.

Yours,
William

matlabbe, 2019-04-11:
Hi William,

You can use external odometry to rtabmap node. For D435i, it seems to have an [example](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/tree/master/config/realsense_d435i) with VINS-Fusion. You may try if you can make it work alone before integrating with RTAB-Map. If rtabmap is built with VINS-Fusion, there is a VIO example [here](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/euroc_datasets.launch) using EuRoC rosbags. I've never tested it live with a real camera though.

By default, rtabmap is 6DOF. To constraint to 3DOF, we can set parameter `Reg/Force3DoF` to true. However, I am not used to biped robots, but as long you can have TF representing the whole robot at any time accordingly to a `base_footprint` frame on ground, odometry can be transformed in that frame and thus estimation in 3DoF will be done accordingly to `base_footprint`. 

cheers,
Mathieu

closed 2019-07-31 by Williamc17