introlab/rtabmap #385: Localizing when grid map is empty

nathangeorge1, 2019-04-22:
Hello,

I'm using rtabmap to localize and plan paths for a robot. Localization is fine in rooms or areas with many features, but once it hits a hallways and only sees a blank image, it gives:

```
[ WARN] (2019-04-22 13:57:33.915) util3d.cpp:604::cloudFromDepthRGB() Cloud with only NaN values created!
[N=/rtabmap/rtabmap,F=publishMaps-1334]: Grid map is empty! (local maps=1)
```

Basically, whenever I turn a robot in a hallway, the robot changes it's pose guess to be inside of a wall, unless I turn fast enough, to where the wall frames are skipped. Is there a parameter or way to improve this so I can turn slowly in hallways?

matlabbe, 2019-04-24:
That warning is just telling that no cloud can be created from the current data. This indeed will appear in front of a white wall with stereo cameras (no disparity can be computed). It just says that the occupancy grid cannot be updated.

If the robot will see often white walls, consider adding wheel odometry to your robot. Stereo odometry from rtabmap won't be better than ZED's odometry on featureless areas.

> the robot changes it's pose guess to be inside of a wall due to lack of features

Do you have a picture of before and after turning? Is ZED's odometry reset when it sees nothing?

cheers,
Mathieu

nathangeorge1, 2019-04-25:
I'm probably going to fuse with wheel encoders or an IMU.

I can create pictures of before and after turning towards a blank wall if it'd be helpful, but beforehand in RViz, the pose is correct, and afterwards, it jumps to random places. The odometry doesn't reset back to (0,0,0). They said on their roadmap is to fix this issue.

It seems if there's no built-in parameter to deal with a lack of features, I can add more sensors. Thanks.

closed 2019-04-25 by nathangeorge1