introlab/rtabmap #387: Loading map and continuing updates

nathangeorge1, 2019-04-25:
Hello,

Although we can enter mapping mode from an empty database (Mem/IncrementalMemory=true) or localization mode from an existing database (Mem/IncrementalMemory=false, database_path=path, rtabmap_args=""), is there a way to enter mapping mode from an existing database (Mem/IncrementalMemory=true, database_path=path, rtabmap_args="")?

Thanks,
Nathan

matlabbe, 2019-04-27:
Yes, we can. Just don't set "--delete_db_son_start" and the previous session will be loaded. A new map is created in the same database. When the robot will find a loop closure with a previous session in the database, the maps should be joined. You can also force rtabmap to wait for a loop closure before starting a new map with `Rtabmap/StartNewMapOnLoopClosure=true`.

cheers,
Mathieu

nathangeorge1, 2019-04-29:
Although I set `rtabmap_args=""` and whether `Rtabmap/StartNewMapOnLoopClosure` is true or false, I get...

```
[ WARN] (2019-04-29 18:37:51.683) Memory.cpp:739::update() The working memory is empty and the memory is not incremental (Mem/IncrementalMemory=False), no loop closure can be detected! Please set Mem/IncrementalMemory=true to increase the memory with new images or decrease the STM size (which is 1 including the new one added).
```

matlabbe, 2019-04-30:
This message means that the database loaded is empty. Make sure --delete_db_on_start is not used. Check if the database is empty:
```
$ ls -l ~/.ros/rtabmap.db
```
or by opening it in rtabmap databaseViewer:
```
$ rtabmap-databaseViewer ~/.ros/rtabmap.db
```

nathangeorge1, 2019-04-30:
I must've been using an empty database because it's sort of working now. 

Although I have `Rtabmap/StartNewMapOnLoopClosure=true` and physically put the robot in an unseen part of the database's map, it stopped loop closing at a wrong spot on the map:

```
[ WARN] (2019-04-30 18:13:24.037) Rtabmap.cpp:994::process() Update map correction based on last localization saved in database! correction = xyz=0.262052,0.090729,0.134713 rpy=0.278033,0.213126,-1.112052, nearest id = 110 of last pose = xyz=0.262187,0.090770,0.133929 rpy=0.278582,0.211479,-1.111456, odom = xyz=0.000189,-0.000075,-0.000770 rpy=0.000422,-0.001423,0.001012
[ WARN] (2019-04-30 18:13:24.402) Rtabmap.cpp:1928::process() Rejected loop closure 28 -> 118: Not enough inliers 0/20 (matches=8) between 28 and 118
[ WARN] (2019-04-30 18:13:25.457) Rtabmap.cpp:1928::process() Rejected loop closure 20 -> 119: Not enough inliers 0/20 (matches=3) between 20 and 119
[N=/rtabmap/rtabmap,F=map<int, rtabmap::Transform> MapsManager::updateMapCaches-487]: Many occupancy grids should be loaded (~53), this may take a while to update the map(s)...
[N=/rtabmap/rtabmap,F=map<int, rtabmap::Transform> MapsManager::updateMapCaches-735]: Map(s) updated! (0.221005 s)
[ WARN] (2019-04-30 18:13:26.465) Rtabmap.cpp:1928::process() Rejected loop closure 28 -> 120: Not enough inliers 0/20 (matches=8) between 28 and 120
[N=/move_base,F=DWAPlanner::DWAPlanner-147]: Sim period is set to 0.20
[ WARN] (2019-04-30 18:13:27.474) Rtabmap.cpp:1928::process() Rejected loop closure 28 -> 121: Not enough inliers 0/20 (matches=4) between 28 and 121
[ WARN] (2019-04-30 18:13:28.479) Rtabmap.cpp:1928::process() Rejected loop closure 84 -> 122: Not enough inliers 0/20 (matches=7) between 84 and 122
[N=/move_base,F=ClearCostmapRecovery::initialize-68]: Recovery behavior will clear layer obstacles
[N=/move_base,F=ClearCostmapRecovery::initialize-68]: Recovery behavior will clear layer obstacles
[N=/move_base,F=OdometryHelperRos::odomCallback-46]: odom received!
[ WARN] (2019-04-30 18:13:29.503) Rtabmap.cpp:1928::process() Rejected loop closure 28 -> 123: Not enough inliers 0/20 (matches=7) between 28 and 123
[N=/zed/zed_wrapper_node,F=ZEDWrapperNodelet::device_poll_thread_func-2068]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.
[N=/zed/zed_wrapper_node,F=ZEDWrapperNodelet::device_poll_thread_func-2068]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.
[N=/zed/zed_wrapper_node,F=ZEDWrapperNodelet::device_poll_thread_func-2068]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.
[N=/zed/zed_wrapper_node,F=ZEDWrapperNodelet::device_poll_thread_func-2068]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.
[N=/zed/zed_wrapper_node,F=ZEDWrapperNodelet::device_poll_thread_func-2068]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.
```

At the time of `[ WARN] (2019-04-30 18:13:29.503) Rtabmap.cpp:1928::process() Rejected loop closure 28 -> 123: Not enough inliers 0/20 (matches=7) between 28 and 123` I think the loop closure stopped, but I want it to keep loop closing until it is more confident. Do I need to increase the number of features loop closure requires in a parameter?

matlabbe, 2019-05-01:
The loop closure detection doesn't stop, the loop closure hypothesis is just not high enough to be considered for loop closure transform estimation. It is still running. What kind of environment do you have? Can you show a screenshot of what the robot is seeing when you try to localize? It is possible to decrease `VIs/MinInliers` but it is preferred to force it localize on locations having a lot of discriminative visual features to get better transformations.

nathangeorge1, 2019-05-01:
My environment is my apartment. Below is a screenshot of my app and also my rtabmap params, that is outputting the following warning continuously. Can you also help me understand what the warning means?

```
[ WARN] (2019-05-01 13:38:39.233) Rtabmap.cpp:2442::process() Loop closure 296->89 rejected!
[ WARN] (2019-05-01 13:38:39.233) Rtabmap.cpp:2442::process() Loop closure 296->22 rejected!
[ WARN] (2019-05-01 13:38:40.187) Rtabmap.cpp:2410::process() Rejecting all added loop closures (2) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 1.975410 (edge 89->297, type=1, abs error=0.057381 m, stddev=0.029048). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 1.000000 of std deviation.
```
[image]

[image]

matlabbe, 2019-05-01:
In this case the loop closure is detected, the transform is computed, but it is rejected after graph optimization accordingly to `RGBD/OptimizeMaxError` parameter. I recommend to set `RGBD/OptimizeMaxError` to 3 instead of 1 (to accept loop closures causing graph deformation at most 3 times the standard deviation of the covariance set in previous links in the graph). In that case with `RGBD/OptimizeMaxError=3`, the loop closure would have been accepted.

nathangeorge1, 2019-05-03:
If `RGBD/OptimizeMaxError`=3 and `Mem/IncrementalMemory`=false, the loop closure was accepted and the robot found its localization value.

If `RGBD/OptimizeMaxError`=3 and `Mem/IncrementalMemory`=true, the loop closure wasn't found and outputted:

```
[ WARN] (2019-05-03 14:29:50.308) Rtabmap.cpp:1042::process() Odometry is reset (large odometry change detected > 0.500000). A new map (1) is created! Last pose = xyz=0.848381,1.705683,-0.159361 rpy=-0.046875,0.094995,1.704353, new pose = xyz=-0.003553,0.002580,-0.001492 rpy=0.001709,-0.000488,-0.000291
[ WARN] (2019-05-03 14:29:51.022) Rtabmap.cpp:2410::process() Rejecting all added loop closures (1) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 7.264494 (edge 64->68, type=0, abs error=0.007264 m, stddev=0.001000). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2019-05-03 14:29:51.023) Rtabmap.cpp:2432::process() Rejecting all added loop closures (1) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 7.344191 (edge 64->68, type=0, abs error=0.420791 deg, stddev=0.001000). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2019-05-03 14:29:51.023) Rtabmap.cpp:2442::process() Loop closure 596->179 rejected!
[ WARN] (2019-05-03 14:29:51.031) Rtabmap.cpp:2647::process() Ignoring location 596 because a global loop closure is required before starting a new map!
```
I tested changing `Mem/IncrementalMemory` back and forth from the exact same robot position, and it only loop closes when `Mem/IncrementalMemory`=false

I also noticed `RGBD/NewMapOdomChangeDistance`=.5 and x, y, and yaw changed more than that because the robot might have tried to change it's position w.r.t to the map.

matlabbe, 2019-05-03:
The odometry covariance seems underestimated. The std deviation of is "`stddev=0.001000`", which is quite small. It is why loop closures are easily discarded in mapping mode. In localization mode, that check is not done by default, so all loop closures are accepted. Can you output the odometry topic values?
```bash
$ rostopic echo /my_odom_topic
```

nathangeorge1, 2019-05-03:
Sure; I manually set the covariance diagonals to 1e-6

Wheel Encoder Odom
```
header: 
  seq: 1543551
  stamp: 
    secs: 1556902116
    nsecs: 909785244
  frame_id: "odom"
child_frame_id: "base_link"
pose: 
  pose: 
    position: 
      x: -0.0532943237467
      y: -0.000870532358399
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.00959346164786
      w: 0.999953981688
  covariance: [1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: [1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06]
---
```

ZED Visual Odom (no inertial)
```
header: 
  seq: 1549002
  stamp: 
    secs: 1556902117
    nsecs: 475513120
  frame_id: "odom"
child_frame_id: "zed_camera_center"
pose: 
  pose: 
    position: 
      x: 0.0708055714066
      y: 0.011920033469
      z: -0.0034424580854
    orientation: 
      x: 0.0028143079935
      y: -0.00184218360472
      z: 0.00215634726475
      w: 0.999992018066
  covariance: [1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: [1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06]
---

```
```

matlabbe, 2019-05-03:
Covariance in position could be 1e-4 (to give 1 cm error). You can also disable this loop closure verification by setting RGBD/OptimizeMaxError=0 or increase the value.

nathangeorge1, 2019-05-06:
Thanks that did it!

My optimal covariance values turned out to be CAMERA_COVARIANCE = .00002 and ENCODER_COVARIANCE = .00001. If it was larger at 1e-4 the actual odometry would be off.

Disabling with `RGBD/OptimizeMaxError=0` made the tf correction coming out of rtabmap too off; but I did `RGBD/OptimizeMaxError=10` instead and that seemed to be a good balance.

closed 2019-05-06 by nathangeorge1

ShubhamBhatt, 2021-07-22:
> Thanks that did it!
>
> My optimal covariance values turned out to be CAMERA_COVARIANCE = .00002 and ENCODER_COVARIANCE = .00001. If it was larger at 1e-4 the actual odometry would be off.
>
> Disabling [...]

Hi I am also have similar issue. I am new to do this I want to know how did you change the parameters??

szx0112, 2021-09-02:
Hi @matlabbe , When I continuous mapping from previous maps, even when loop closure is detected, I cannot see the old map merges to the new map. However, if I re-open rtabmap after the mapping, and download all clouds, I can see the maps merged together. I am wondering if it is possible to show the old map in real-time once loop closure is detected. It just help me figure out if the new map is correctly registered to the old ones, and if the merged map is distorted, when the distortion starts.

matlabbe, 2021-09-05:
It is because the old map is not yet in the cache of rtabmapviz. You would see two trajectories with different color when the maps are merged, then you know that the old map has just been merged. Without closing rtabmap, just do Download all clouds, it will get data of the old map.

In rviz, if you watch `/rtabmap/cloud_map` topic, it should be reconstructed automatically with all data of the old map.