introlab/rtabmap #391: improve free space creation

nathangeorge1, 2019-05-03:
Hello,

Sometimes if the camera angle is not tilted down enough or lighting is too dark, free or white space is not generated enough. Too much is generated as obstacles. As such, I had a few questions:

1. After looking at `rosrun rtabmap_ros rtabmap --params` I still don't understand `Odom/Strategy`, `OdomF2M/MaxSize`, and `Vis/CorType`. What do they mean?
2. Are there other params in the `Vis/` namespace that will help?
3. Do you have any general suggestions, besides physically changing the camera or lighting, to make more ground points free instead of obstacles?
4. Is there a way to check while rtabmap is running how many features are being created per image?

Thanks,
Nathan

matlabbe, 2019-05-03:
Hi Nathan,

1. `Odom/Strategy` is the odometry strategy used. F2M (Frame-to-Map) and F2F (Frame-to-KeyFrame) are the built-in odometry approaches in rtabmap (see this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf), figure 2). The other ones are community (visual or lidar) odometry approaches integrated in rtabmap for convenience (when rtabmap is built with these dependencies). `OdomF2M/MaxSize` is used with Odometry F2M strategy, it is the maximum size of the local feature map (see the same section in the paper). `Vis/CorType` is how visual features are matched, by descriptor matching or by optical flow. Optical flow can be used with F2F odometry.

2. `Vis/` parameters don't affect how obstacles are detected in the occupancy grid map. Look at the `Grid/` parameters. To generate more features, you can check `Vis/MaxFeatures`. The feature detector chosen `Vis/FeatureType` and its related parameters can also affect the number of features extracted.

3. If points that should be ground are obstacles, not sure I remember in what kind of environment you are, but it could be possible to change the segmentation strategy if the sensor is too noisy. For example, setting `Grid/NormalsSegmentation=false` and adjusting `Grid/MaxGroundHeight` threshold so that all points under 10 cm are ground, all other are obstacles for example.

4. Yes, open Statistics tab (Window->Show view), then check "Keypoint / Keypoints count in the last signature" value

cheers,
Mathieu

nathangeorge1, 2019-05-07:
[image]

For 4., where is it?

matlabbe, 2019-05-07:
Window -> Show View -> Statistics

nathangeorge1, 2019-05-08:
Didn't see the word "Statistics" earlier; thanks that's what I was looking for.

For improving free space in general, turns out I needed to optimize the stereo cameras parameters, such as gain, exposure, resolution, and frame rate; that did the trick.

closed 2019-05-08 by nathangeorge1