introlab/rtabmap #392: exported point clouds equality is poor

MattixLee, 2019-05-07:
Hello,

thank you very much for your works , I found the exported point cloud quality is worse than the point cloud displayed in the 3d map window. In export 3d clouds window, I just change voxel size to 0.

I want to know how to export the point clouds are shown in 3D map window.

Thank you!

matlabbe, 2019-05-07:
[image]
With those settings (with `Regenerate clouds` to false), the generated cloud should be exactly the same than in the 3D Map, as all clouds are copied and assembled directly.

MattixLee, 2019-05-08:
Hi， matlabbe 

Thank you very much for your reply. According to the settings you provide， there are still some differences between the derived point cloud and the display point cloud. As shown in first image, it is point clouds shown in 3D map window; As shown in second image, it  is point clouds expoerted according to your setting; As shown in third image,it  is point cloud exported accoring to setting of fourth image.

[image]
[image]
[image]
[image]

Thank you !

matlabbe, 2019-05-15:
I think the difference between RTAB-Map and MeshLab rendering is that in MeshLab, by default shading is enabled using the normals of the point cloud, as in RTAB-Map there is no shading. Here is an example with shading:
[image]

then disabling shading in Meshlab (clouds look identical):
[image]

The shading helps to see the geometry of the cloud. With your parameters, you are smoothing the surfaces, which then appears smoother with shading on.

cheers,
Mathieu

MattixLee, 2019-05-16:
@matlabbe 
hello,
Thank you very much ! I've got it.

closed 2019-05-16 by MattixLee