introlab/rtabmap #393: Difference in RViz vs Rtabmapviz

nathangeorge1, 2019-05-07:
Hello,

At the same time, rendering /map in RViz and "Graph View" in rtabmapviz gives

[image]

In rtabmapviz it doesn't put obstacles in the entrance way like in RViz. Why is that?

matlabbe, 2019-05-07:
If you just opened rtabmapviz, make sure you downloaded the whole map (Edit->Download all clouds). It seems that only latest data have been used for the grid map.

cheers,
Mathieu

nathangeorge1, 2019-05-08:
This time I kept rtabmapviz and rviz open the whole time and it seems better but the door is still blocked, only for RViz.

[image]

I forgot that rtabmap only updates the map when the odometry detects motion; so after moving the robot back and forth, it was able to see through the hallway!

[image]

Do you know why before moving the robot back and forth, there's a difference in rviz and rtabmapviz?

matlabbe, 2019-05-10:
which topic is visualized in rviz? normally `/rtabmap/grid_map` would match the map created in rtabmapviz.

nathangeorge1, 2019-05-12:
RViz is using `/map` but I use

`<remap from="/rtabmap/grid_map" to="/map" />`

matlabbe, 2019-05-15:
I reproduced the behavior. The difference is that in rtabmapviz, the grid is updated whenever the robot is moving or not, as with `/map` topic, it is updated only if the robot moves.

If you are looking to the same behavior than rtabmapviz with `/map` topic, set `map_negative_scan_empty_ray_tracing` parameter to true for rtabmap node.

cheers,
Mathieu

nathangeorge1, 2019-05-15:
That param looks perfect. I can't test it for another month; but will update then.

nathangeorge1, 2019-06-11:
Does `map_negative_scan_empty_ray_tracing` have a namespace, like the `Mem` in `Mem/IncrementalMemory`? 

I also don't see it in `rosrun rtabmap_ros rtabmap --params`. Is there a way to tell if the parameter is set correctly?

matlabbe, 2019-06-12:
This one is a ros-related parameter, it won't appear in `--params`. There is indeed missing documentation on those parameters on [ros wiki](http://wiki.ros.org/rtabmap_ros). In recent version you would have a warning that this parameter changed name to [map_empty_ray_tracing](https://github.com/introlab/rtabmap_ros/blob/48bcda84abd06562baba542a777790e0f60e4673/src/MapsManager.cpp#L108), but it should be still be used:
https://github.com/introlab/rtabmap_ros/blob/48bcda84abd06562baba542a777790e0f60e4673/src/MapsManager.cpp#L99-L107

The state of the parameter is logged in terminal on initialization here:
https://github.com/introlab/rtabmap_ros/blob/48bcda84abd06562baba542a777790e0f60e4673/src/MapsManager.cpp#L127

You should also be able to get it from rosparam server:
```bash
$ rosparam get /rtabmap/map_empty_ray_tracing
```

nathangeorge1, 2019-06-12:
Although I set both `map_negative_scan_empty_ray_tracing=true` and `map_empty_ray_tracing=true`, rtabmapviz is showing updates, while rviz is still not.

I don't see the deprecation warning, leading me to believe I'm not setting the parameter; but I tried it with the `/rtabmap` namespace as well as the namespace I'm currently using for all the other rtabmap params, but it's still not working.

My rtabmap_ros version is 0.17.6.

My rtabmap printout for maps is this, only have the deprecated variable:

```
[N=/rtabmap/rtabmap,F=init-98]: /rtabmap/rtabmap(maps): map_filter_radius          = 0.000000
[N=/rtabmap/rtabmap,F=init-99]: /rtabmap/rtabmap(maps): map_filter_angle           = 30.000000
[N=/rtabmap/rtabmap,F=init-100]: /rtabmap/rtabmap(maps): map_cleanup                = true
[N=/rtabmap/rtabmap,F=init-101]: /rtabmap/rtabmap(maps): map_negative_poses_ignored = true
[N=/rtabmap/rtabmap,F=init-102]: /rtabmap/rtabmap(maps): map_negative_scan_ray_tracing = true
[N=/rtabmap/rtabmap,F=init-103]: /rtabmap/rtabmap(maps): cloud_output_voxelized     = true
[N=/rtabmap/rtabmap,F=init-104]: /rtabmap/rtabmap(maps): cloud_subtract_filtering   = false
[N=/rtabmap/rtabmap,F=init-105]: /rtabmap/rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[N=/rtabmap/rtabmap,F=init-121]: /rtabmap/rtabmap(maps): octomap_tree_depth         = 16
```

nathangeorge1, 2020-04-10:
Hi @matlabbe, is there any update on this one?

Unless I'm missing something, doing `rosparam get /rtabmap/map_empty_ray_tracing` returns nothing.

matlabbe, 2020-04-17:
I reproduced the problem. Setting the parameter works, but it doesn't show up in rosparam server unless we explicit set it.

Example 1:
```
$ roscore
$ rosrun rtabmap_ros rtabmap -d
...
[ INFO] [1587155120.744167196]: /rtabmap(maps): map_empty_ray_tracing      = true
...
$ rosparam get /rtabmap/map_empty_ray_tracing
ERROR: Parameter [/rtabmap/map_empty_ray_tracing] is not set
```

Example 2:
```
$ roscore
$ rosrun rtabmap_ros rtabmap -d _map_empty_ray_tracing
...
[ INFO] [1587155120.744167196]: /rtabmap(maps): map_empty_ray_tracing      = true
...
$ rosparam get /rtabmap/map_empty_ray_tracing
ERROR: Parameter [/rtabmap/map_empty_ray_tracing] is not set
```

nathangeorge1, 2020-04-21:
I was able to change it by setting the arg using rosrun:

```
nathan@nathan-vm3:~/newmind$ rosrun rtabmap_ros rtabmap -d _map_empty_ray_tracing:=false
...
[N=/rtabmap,F=init-127]: /rtabmap(maps): map_empty_ray_tracing      = false
```

but in the launch file, setting rtabmap_args to it doesn't work
```
<arg name="rtabmap_args"		default="-d _map_empty_ray_tracing:=false" />
...
	<include if="$(eval sensor_brand == 'astra' or sensor_brand == 'zed')" file="$(find rtabmap_ros)/launch/rtabmap.launch">
		<arg name="rtabmap_args"                value="$(arg rtabmap_args)" />
...
```
```
[N=/rtabmap/rtabmap,F=init-127]: /rtabmap/rtabmap(maps): map_empty_ray_tracing      = true
```

I thought it might be an issue with [private namespacing](https://answers.ros.org/question/217030/private-parameters-in-a-sub-namespace/), but trying various combinations of the "_rtabmap" prefix didn't work.

I tried editing rtabmap_ros's rtabmap.launch, but it also didn't work:
```
   <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="$(arg output)" args="$(arg rtabmap_args)" launch-prefix="$(arg launch_prefix)">
      <param name="rtabmap_ros/rtabmap/map_negative_scan_empty_ray_tracing" value="false" />
...
```

I took another image as I thought it might be helpful:

[image]

I believe the issue has something to do with how [CoreNode.cpp](https://github.com/introlab/rtabmap_ros/blob/71294263b8b01662cd42ec339ea0c9ee50802951/src/CoreNode.cpp) passes data to [MapsManager](https://github.com/introlab/rtabmap_ros/blob/71294263b8b01662cd42ec339ea0c9ee50802951/src/MapsManager.cpp#L108). Maybe the launch uses a nodelet and rosrun uses a node, which is causing some private namespace issues, or maybe I'm not understanding how private namespace prefixes work.

Is there a workaround to getting `map_empty_ray_tracing:=false` without needing to use `rosrun`?

nathangeorge1, 2020-04-23:
It's working! Turns out the parameter we needed was `map_always_update=true`; now, a static robot in both RViz and Rtabmapviz will degenerate obstacles if the obstacle is removed.

In rtabmap_ros's rtabmap.launch, scroll down to the `<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="$(arg output)" args="$(arg rtabmap_args)" launch-prefix="$(arg launch_prefix)">` part and add

`      <param name="map_always_update" value="true" />`

Thanks for your help on this. Would I also be able to know more about the differences between `map_always_update`, `map_empty_ray_tracing`, and all the other private variables accessed with the private node handle?

matlabbe, 2020-04-27:
For all those [parameters](https://github.com/introlab/rtabmap_ros/blob/71294263b8b01662cd42ec339ea0c9ee50802951/src/MapsManager.cpp#L123-L130):
```
ROS_INFO("%s(maps): map_filter_radius          = %f", name.c_str(), mapFilterRadius_);
ROS_INFO("%s(maps): map_filter_angle           = %f", name.c_str(), mapFilterAngle_);
ROS_INFO("%s(maps): map_cleanup                = %s", name.c_str(), mapCacheCleanup_?"true":"false");
ROS_INFO("%s(maps): map_always_update          = %s", name.c_str(), alwaysUpdateMap_?"true":"false");
ROS_INFO("%s(maps): map_empty_ray_tracing      = %s", name.c_str(), scanEmptyRayTracing_?"true":"false");
ROS_INFO("%s(maps): cloud_output_voxelized     = %s", name.c_str(), cloudOutputVoxelized_?"true":"false");
ROS_INFO("%s(maps): cloud_subtract_filtering   = %s", name.c_str(), cloudSubtractFiltering_?"true":"false");
ROS_INFO("%s(maps): cloud_subtract_filtering_min_neighbors = %d", name.c_str(), cloudSubtractFilteringMinNeighbors_);
```
I updated the ros wiki with the parameters missing in the documentation. See parameters section under [rtabmap](http://wiki.ros.org/rtabmap_ros#rtabmap) node.

cheers,
Mathieu

nathangeorge1, 2020-04-27:
Looks perfect! Thanks again.

closed 2020-04-27 by nathangeorge1