introlab/rtabmap #394: obstacles should be free space

nathangeorge1, 2019-05-07:
Hello,

When going down a hallway, the image input to rtabmap cannot see an open doorway; it looks like a wall, which is the same for a person who doesn't turn his head. Rtabmap makes this area obstacles.

When I get to the doorway, I turn 90 degrees to face the room, and it can see inside the room, but at the doorway it still sees obstacles. This is a similar problem to https://github.com/introlab/rtabmap/issues/393.

Is there a preferred way to replace the obstacles at the doorway with free space?

matlabbe, 2019-05-07:
Try adding `Grid/RayTracing=true` and `Grid/3D=false` parameters. Obstacles would be cleared if the sensor sees a wall behind them.

nathangeorge1, 2019-05-08:
That helped a lot! In addition to moving back and forth, the robot can now see through doorways or tight angles. 

Thanks for all the hard work on this too, as I've been asking a lot of questions. I've tried ORB_SLAM, gmapping, hector, and seen cartographer, but rtabmap is the best overall for robot motion in my opinion. I'm trying to create a product out of rtabmap; so if I ever start to generate revenue, we can consider a partnership later.

closed 2019-05-08 by nathangeorge1