introlab/rtabmap #396: RGBD-Mapping without inheriting QWidget

ahmadhasan2k8, 2019-05-11:
Is there a way to get the point cloud and pose out during running RGBD-Mapping (example) without inheriting QWidget. I do not need the visualization but would like to get point cloud and pose out based on a queries. Is there any location where the updated point cloud/ map is stored?

ahmadhasan2k8, 2019-05-14:
It was some work but done. Closing.

closed 2019-05-14 by ahmadhasan2k8

DormouseLi, 2021-12-21:
@ahmadhasan2k8 Hello！I have the same problem too. Is there any suggestion?

matlabbe, 2021-12-21:
Just don't inherit QWidget and remove all related QWidget stuff, but keep UEventsHandler if you are using the events based [example](https://github.com/introlab/rtabmap/tree/master/examples/RGBDMapping). Without events, look at this [example](https://github.com/introlab/rtabmap/tree/master/examples/NoEventsExample) instead.

fjlxb, 2022-12-07:
> Just don't inherit QWidget and remove all related QWidget stuff, but keep UEventsHandler if you are using the events based [...]

Did all change in MapBuilder.h file?

matlabbe, 2022-12-11:
You would have to remove Qt stuff in main.cpp too. You would also need to do your own "waiting loop" here: 
https://github.com/introlab/rtabmap/blob/a4761ad397ce195c254f65be10f7e8e2810facd0/examples/RGBDMapping/main.cpp#L227

fjlxb, 2022-12-15:
> You would have to remove Qt stuff in main.cpp too. You would also need to do your own "waiting loop" here:
> 
> https://github.com/introlab/rtabmap/blob/a4761ad397ce195c254f65be10f7e8e2810facd0/examples/RGBDMapping/main.cpp#L227

Oh, thx, it works!