introlab/rtabmap #397: PMD Pico Flexx 3D mapping without odometry

adaltun, 2019-05-15:
Hello, I have a pico flexx time of flight camera from PMD and I want to create an indoor 3D mapping over ROS without any odometry information. I have tried various packages however I wasn't successful so far. I have checked there are many packages to create map with RGB-D cameras like RTAB-Map however my camera is not a RGB camera. It is only a depth camera. I wonder if I can use rtabmap for my purpose. If not, I will appreciate if you could suggest me some packages or way. 
Regards,

adaltun, 2019-05-15:
I have Ubuntu 16.04 and using ROS Kinect version.

matlabbe, 2019-05-15:
You may buy a RGB camera and register it to PMD in order to create a RGB-D camera. See http://wiki.ros.org/openni_launch/Tutorials/ExtrinsicCalibrationExternal, you will then access to all RGB-D SLAM approaches.

To use PMD alone, it could be possible to do with RTAB-Map (latest version from source can subscribe only to a PointCloud2 topic) with [icp_odometry](http://wiki.ros.org/rtabmap_ros#icp_odometry). The default parameters may not work though, so some tuning would be required with changing `$ rtabmap --params | grep Icp/` parameters. You could try `icp_odometry` alone before integrating rtabmap to see if the tracking can be good enough:
```xml
<node pkg="rtabmap_ros" type="icp_odometry" name="icp_odometry" output="screen">
   <remap from="scan_cloud"         to="/pmd/points"/>
   <param name="frame_id"           type="string" value="pmd_base_link"/>
   <param name="Icp/VoxelSize"      type="string" value="0.05"/>
   <param name="Icp/PointToPlane"   type="string" value="true"/>
   ...
</node>
```

The limitations I see using the PMD alone is that a lot of geometry should be always visible in the field of view of the camera to avoid being lost or getting large odometry drift on a planar surface for example. It is why adding a RGB sensor could help to extract also visual features for better tracking. [Google Cartographer](https://github.com/googlecartographer/cartographer) and [ethzasl_icp_mapping](https://github.com/ethz-asl/ethzasl_icp_mapping) may be also other geometry-only based solutions for that kind of sensor.

cheers,
Mathieu