introlab/rtabmap #400: use ORBSLAM2 library in rtabmap

Garand0o0, 2019-05-25:
There is an error when I compile the OdometryORBSLAM2.cpp.
OdometryORBSLAM2.cpp:
```
#include <System.h>
#include <thread>
```
error:
fatal error: System.h: no such file or directory.

And I can't find System.h in rtabmap-master.Can someone help me?Thanks.

matlabbe, 2019-05-29:
System.h is part of ORB_SLAM2 repo: https://github.com/raulmur/ORB_SLAM2/blob/master/include/System.h

The ORB_SLAM2 include directory doesn't seem to be correctly set to compiler (https://github.com/introlab/rtabmap/blob/5b5b594f4aabd6384073c6bb8bbf9ee7b2345725/corelib/src/CMakeLists.txt#L450)

What is the output of `make VERSBOSE=1`?

Garand0o0, 2019-05-30:
Hi matiabbe,
 It works,thinks!

Garand0o0, 2019-05-30:
Hi matlabbe,
It works,thanks!