introlab/rtabmap #402: How to navigation  after mapping with RTAB-MAP ?

MattixLee, 2019-05-27:
Hello, 

Thank you for your work!
As shown in figure, the map has been built with RTAB-map.
I want to use this map or it's octomap for navigation, standalone version of RTAB-map was used, can you provide me with relevant tutorials?

Thank you !

[image]

nathangeorge1, 2019-05-29:
Hello, although @matlabbe can give you further info, I haven't found any "tutorials" outside of the wiki tutorials: https://github.com/introlab/rtabmap/wiki/Tutorials

But if you are familiar with ROS, the easiest way is to subscribe to the relevant topics in the rtabmap node: https://wiki.ros.org/rtabmap_ros.

My company also has free generic robot navigation code that is based on rtabmap: http://www.newmindrobotics.com/

MattixLee, 2019-05-29:
@nathangeorge1 
hi, Thank you for you help! I am trying to do the job with ROS.
I want to keep trying.

cheers

matlabbe, 2019-05-30:
Relevant tutorials on ROS are here: https://wiki.ros.org/rtabmap_ros/Tutorials

Even you load that map in rtabmap_ros, it is preferred to do navigation in a map taken from the same point of view than the robot (and with same sensors) to get good localization, so you may have to rebuild a new map with the robot.

cheers,
Mathieu

MattixLee, 2019-05-31:
@matlabbe 

Thank you for your reply!  I do not have a turtlrbot, so I finished mapping and navigation in gazebo.

cheers!

RashidAb, 2021-05-13:
What do I do if I am using a custom robot? not turtlebot

benz777, 2022-03-13:
any solution please @MattixLee

matlabbe, 2022-03-13:
For navigation with rtabmap, it is pretty straightforward if your robot is move_base ready (http://wiki.ros.org/navigation/Tutorials/RobotSetup), only /map topic would have to be remapped. Even if you are on a custom robot, many things from turtlebot robot could be transferred. Take a look at the gazebo turtlebot tutorial here: https://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot

Use rqt_graph to see how topics are connected. `rosrun tf2_tools view_frames` can be also used to see the resulting TF tree.

alp3327, 2023-01-26:
I have a zed camera and I have mapped out and localized. I have no idea how to navigate.

nsfhac, 2023-03-15:
@alp3327 hey have you succeeded in Navigating using ZED Camera? I'm also trying to do the same thing. Can you help me? thanks

> I have a zed camera and I have mapped out and localized. I have no idea how to navigate.

farhan-haroon, 2023-10-28:
Were you able to get Autonomous Navigation using the RTAB Map with ZED camera?

matlabbe, 2023-10-29:
If you are on ROS2, it is possible to use nav2. There is an example here with an rgb-d camera (if you have a zed camera, you would have to remap the correct topics): https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_demos/launch/turtlebot3_rgbd.launch.py