introlab/rtabmap #403: Stall at startup

atnTeosit, 2019-05-27:
RTabMap get stuck in cloudViewer constructor.

Here the stack:
	vtkRenderingOpenGL2-8.0.dll!vtkWin32OpenGLRenderWindow::Frame() Ligne 378	C++
 	vtkRenderingCore-8.0.dll!vtkRenderWindow::Render() Ligne 477	C++
 	rtabmap_gui.dll!rtabmap::CloudViewer::CloudViewer(QWidget * parent, rtabmap::CloudViewerInteractorStyle * style) Ligne 181	C++
 	rtabmap_gui.dll!rtabmap::MainWindow::MainWindow(rtabmap::PreferencesDialog * prefDialog, QWidget * parent, bool showSplashScreen) Ligne 218	C++
 	RTABMap.exe!main(int argc, char * * argv) Ligne 53	C++

Freeze on "::SwapBuffers(this->DeviceContext);"

I'm using Visual Studio 2019, VTK 8.0.0 or 8.2.0, PCL 1.8.1 or 1.9.1

Does someone have a clue ?
Thanks

matlabbe, 2019-05-30:
Those configurations have not been tested on Windows. See the appveyor build here: https://github.com/introlab/rtabmap/blob/master/.appveyor.yml
For convenience, all dependencies that we know are working are pre-built. PCL 1.8 or 1.9 should work. You may try VTK 6.3 first as it seems to be a VTK related error, which has been more tested for all rtabmap functionalities.

Regards,
Mathieu

atnTeosit, 2019-06-07:
It was a graphic driver issue, nothing related to rtabmap.
Thanks

closed 2019-06-07 by atnTeosit