introlab/rtabmap #404: Manual map modifications

nathangeorge1, 2019-05-29:
Hello,

Are there any parameters or tools to manually modify the OccupancyGrid? I wanted to box out an area and make everything outside of it "occupied" so the robot doesn't move there.

I also might want to have users manually make areas free or occupied by a GUI.

Thanks,
Nathan

malerv, 2019-05-29:
If you are using ros you could subscribe to the occupancy grid topic do the modification and republish it on an other topic.

nathangeorge1, 2019-05-29:
I thought about that; but then, how would I save it? I thought rtabmap does that internally.

matlabbe, 2019-05-30:
If you are going in localization mode (`Mem/IncrementalMemory=false`), memory management is disabled (`Rtabmap/TimeThr=0`, default), optimization is not done from graph end (`RGBD/OptimizeFromGraphEnd=false`, default) the global occupancy grid saved to database is reloaded "as is" without regenerating it. That occupancy grid can be edited in rtabmap-databaseViewer, see File->Edit optimized 2D map...

When reloading the map with the conditions above, the map published by rtabmap_ros should contain the edits.

cheers,
Mathieu

nathangeorge1, 2019-06-03:
sounds perfect; I'll check this out in a month or so