introlab/rtabmap #405: Issue with top-level CMakeLists.txt and Realsense2

srinath-iko, 2019-05-29:
Please change the top-level cmakelists code to as below to work with realsense2 cameras.

(lines 410-419)
`IF(WITH_REALSENSE2)
    IF(WIN32)
        FIND_PACKAGE(RealSense2 QUIET)
	ELSE()
        FIND_PACKAGE(RealSense2 QUIET)
    ENDIF()
    IF(RealSense2_FOUND)
       MESSAGE(STATUS "Found RealSense2: ${RealSense2_INCLUDE_DIRS}")
    ENDIF(RealSense2_FOUND)
ENDIF(WITH_REALSENSE2)
`

matlabbe, 2019-05-30:
Are you on Windows or Linux? On linux, the latest release 2.22 still generate a config file named `realsense2Config.cmake`. For Windows, there could be indeed an error as `realsense2_FOUND` would be `RealSense2_FOUND`:
https://github.com/introlab/rtabmap/blob/cd6e51a9683b293cdcf431f5fd33846dee43e3ce/CMakeLists.txt#L410-L419

srinath-iko, 2019-05-31:
I am on Linux. Although there wouldn't be a major error, the `FIND_PACKAGE(realsense2 QUIET)` would not call the FindRealSense2.cmake but rather directly find the realsense libraries and therefore not set the names accordingly.

matlabbe, 2019-05-31:
There is a hack here: 
https://github.com/introlab/rtabmap/blob/cd6e51a9683b293cdcf431f5fd33846dee43e3ce/cmake_modules/FindRealSense2.cmake#L26-L29
On linux, we use the realsenseConfig.cmake instead of FindRealSense2.cmake. On windows, they don't provide realsenseConfig.cmake, so the hack is to find it ourself with FindRealSense2.cmake of this repo. Do you have errors or realsense2 is not found because of this approach?

closed 2019-06-13 by srinath-iko