introlab/rtabmap #406: RTABMAP with HSV color for RDB-D

elpimous, 2019-06-01:
ROBOT : QBO version 1 (robot odometry)
board : TX2
Cam : D435
ubuntu 18.04
Ros melodic
                                     -------------------------------------------

Hello the team, hello all,
Working on a fully autonomous robot.
Need the best possible navigation, and RTABMAP seems to be the best for me.

I see that my navigation is very good just after building map, but poor later in the journey.
It seems to be due to changes in the day ambient colors.

I had idea to create the map with a HSV color topic instead of a RGB one, but it doesn't seems to work (rtabmap waits after "rtabmap 0.17.6 started...")

1/ Could anyone tell me how to use HSV with rtabmap ?! and is it a good idea ?

                                     -------------------------------------------

2/ I had idea to create a map in the morning, and restart in the evening, over the existant one.
like this, i should obtain a map containing 2 different contrast maps of same environment.

Could rtabmap memorize both colors periods for same localization

Thanks all.
Vincent

matlabbe, 2019-06-02:
Hi Vincent,

Actually, RGB is converted to grayscale for feature detection and descriptor extraction, the color information is not used. So not sure how converting to HSV would help. You may try to convert HSV to grayscale.

You can map at different time and merge the two sessions in the same map. On localization, rtabmap will use both versions of the same location. The tricky part is you don't want to much lighting change between the sessions to be able to close loop closures automatically between them. You may have to create more than two sessions of the same environment at different time to be able to merge them all together. On localization, you will then have multiple versions of the same location at different lighting variations to localize on. 

cheers,
Mathieu

elpimous, 2019-06-03:
Hi Mathieu.
Thanks for answer. 
OK, I think that I can forget HSV idea... 
I'll test 4 different mappings hours (09h-13h-19h-closed windows)
I'll report here result.
Thanks again Mathieu. 
Vincent (France)

elpimous, 2019-07-01:
Hi all,
Hi Mathieu. 
Well, multimap of same environment, with different contrasts doesn't seem to work correctly.
Ex: navigating in a morning map, good.
      Mix it with an evening one, good for evening, but localization very hard in the next morning

It seems that the last mixed map erases previous ones..

Am I right?

matlabbe, 2019-07-03:
You could try localizing the next morning only on the map taken the previous morning. If localization is not better or similar than with the one merged with morning+evening, it means that the lighting the next morning is different than lighting of previous morning.

I don't have the full results yet, but we did some experiments taking maps at each 20-30 minutes intervals during 3 hours at dawn or dusk to capture all lighting variations during that time. The localization performance done the next day was a lot higher with the merged map.

All data is kept in the database from all sessions, and rtabmap tries to localize on all of them independently if they are linked together or not in the database. It is preferred that all maps are linked in the databases (loop closures between sessions) to have only a single map, in order to avoid changing origin if the localization happens on an independent map.

Note that if memory management is enabled, depending on the trajectory, it may happen that there are more locations of the last map than the previous ones in the working memory. When launching in localization mode, you can force to use all locations in long-term memory with:
```ini
Mem/IncrementalMemory=false
Mem/InitWMWithAllNodes=true
```
cheers,
Mathieu