introlab/rtabmap #407: A error happened when the last step catkin_make -j1 (Ubuntu 18.04 / ROS Melodic)

BeautyBeast, 2019-06-04:
*** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_shape.so.3.2.0', needed by '/home/nvidia/catkin_ws/devel/lib/librtabmap_ros.so'.  Stop.
CMakeFiles/Makefile2:4923: recipe for target 'rtabmap_ros-melodic-devel/CMakeFiles/rtabmap_ros.dir/all' failed
make[1]: *** [rtabmap_ros-melodic-devel/CMakeFiles/rtabmap_ros.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

There was no file named libopencv_shape.so.3.2.0 or librtabmap_ros.so. What should I do ? Thank you.

matlabbe, 2019-06-04:
`librtabmap_ros.so` is the library being built, failing because `libopencv_shape.so.3.2.0` is not found. The docker image for melodic is able to build: https://hub.docker.com/r/introlab3it/rtabmap_ros/builds
I just checked, and the library `libopencv_shape.so.3.2.0` exists under `/usr/lib/x86_64-linux-gnu`. You are on `aarch64`, `libopencv_shape.so.3.2.0` may not be available. However, rtabmap doesn't use `libopencv_shape.so.3.2.0`, it should be a dependency given by the `/usr/share/OpenCV/OpenCVConfig.cmake` installed on your computer. If `libopencv_shape.so.3.2.0` doesn't exist, remove `opencv_shape` from the list of modules in `OpenCVConfig.cmake`.

cheers,
Mathieu

BeautyBeast, 2019-06-05:
> `librtabmap_ros.so` is the library being built, failing because `libopencv_shape.so.3.2.0` is not found. The docker image for melodic is able to build: [...]

Thank you for your answer but the issue still exists after I delete the opencv_shape from OpenCVConfig.cmake.

By the way, I ran the RTAB-Map standalone libraries in LINUX directly using command: rtapmap. There was something wrong with the GUI, it seemed like the RTAB worked but the simulation which should be on the right side of the interface didn't appear.
[image]

And the last question how can I save the data both recording data and simulation data when I use the standalone libraries? 

Thank you

matlabbe, 2019-06-05:
OpenGL seems to have difficulty to run on the platform. Rendering in 3D Map view is OpenGL. Does RVIZ have the same problem? When building `rtabmap_ros`, you can do `catkin_make VERBOSE=1` to see the linker inputs. There should be a dependency from `rtabmap_ros` that redirects the `libopencv_shape.so` library to linker.

constructive-realities, 2019-06-07:
FYI, i have both issues (libopencv_shape as well as the rtabmap UI render issue) 

GL seems to work just fine in Rviz

constructive-realities, 2019-06-07:
These are the files on the base jetson nano image.  
[image]

so is this just a versioning issue?

matlabbe, 2019-06-08:
If you do:
```bash
ln -s  /usr/lib/libopencv_shape.so.3.3.1 /usr/lib/aarch64-linux-gnu/libopencv_shape.so.3.2.0
```
does it work?

To really find the source of the problem, we would have to search which rtabmap's dependency is linked to wrong opencv library with `ldd`, or if the the problem is coming from the installed OpenCVConfig.cmake.

For the rendering issue, it is maybe something related to vtk/qt/pcl visualizer binaries for this platform. So if you are using ros, use rviz instead. For Jetson TX2, there are some issues listed [here](https://github.com/introlab/rtabmap_ros#build-from-source-for-nvidia-jetson-with-opencv-4-tegra) (in particular 7 and 8), not sure if they are applicable to Nano too.

constructive-realities, 2019-06-20:
Small incremental progress; but still blocked.

after setting a link for libopencv_shape.so

`catkin_make -j1`

```
[ 57%] Built target rtabmap_ros_generate_messages_lisp
[ 66%] Built target rtabmap_ros_generate_messages_py
[ 67%] Built target rtabmap_ros_gencfg
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_superres.so.3.2.0', needed by '/home/rosuser/rtabmap_ws/devel/lib/librtabmap_ros.so'.  Stop.
CMakeFiles/Makefile2:9275: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed
rosuser@cap-bot-003:~/rtabmap_ws$ sudo find / -iname 'libopencv_superres*'
^[[Afind: ‘/run/user/1000/gvfs’: Permission denied
/usr/lib/libopencv_superres.so
/usr/lib/libopencv_superres.so.3.3.1
/usr/lib/libopencv_superres.so.3.3

```
So i did the same linking as with `libopencv_shape.so` for `libopencv_superres.so`

which brings the similar error for `libopencv_videostab.so` (which was also in /usr/lib/):  same linking

but then (finally) stopped by `/libopencv_viz.so.3.2.0`

as libopencv_viz does not exist on the nano...   so not sure if this got dropped or rolled into some other resource for opencv 3.3

matlabbe, 2019-06-20:
Thx for the update. If rtabmap is building and working (minor the opengl problem in the 3D Map view), those opencv dependencies are maybe coming from cv_bridge package. You could try to uninstall cv_bridge: 
```bash
$ sudo apt-get remove ros-melodic-vision-opencv
```

Then rebuild vision_opencv stack from source (clone in your catkin_ws): https://github.com/ros-perception/vision_opencv, so that cv_bridge is correctly linking on the opencv 3.3.1 binaries installed instead of the missing opencv 3.2.0.

cheers,
Mathieu