introlab/rtabmap #408: can't merge maps ?!

elpimous, 2019-06-05:
hello all.

the tuto command doesn't wort, to fuse maps ?!!

`rtabmap-reprocess "map1;map2" fused_map.db`

the error : `Input database "map1;map2" dosen't exist !`

An idea ?

matlabbe, 2019-06-05:
The command would be more like:
```
$ rtabmap-reprocess "map1.db;map2.db" fused_map.db
```
but version >= 0.18 is required. I updated the [tutorial](https://github.com/introlab/rtabmap/wiki/Multi-Session-Mapping-with-RTAB-Map-Tango#processing-the-databases-without-gui) with the minimum version.

In 0.17.6, to merge databases, it should be done with the UI with the approach shown in section 5 here: https://github.com/introlab/rtabmap/wiki/Multi-Session-Mapping-with-RTAB-Map-Tango#5-processing-the-databases

cheers,
Mathieu

elpimous, 2019-06-05:
Yes,  it was my command, lol.
Thanks, I'll update melodic rtabmap_ros version

matlabbe, 2019-06-05:
Melodic binaries are still at 0.17.6. I [recently did a pull request for 0.19.3](https://github.com/ros/rosdistro/pulls) for next melodic release, not sure when osrf will update it. So for now,  if you want to use the command line approach, rtabmap should be built from source.

elpimous, 2019-06-05:
Good.
Thanks. 
I'll do it so.

elpimous, 2019-06-07:
Hello,
Can t wait melodic updates... 
I used Rtabmap GUI App, 
Opened all needed dB maps, 
Clicked on find more loop closures, 
And I finally obtained a nearly perfect final map, containing all necessary color informations, for mostly luminosity, in localization and navigation.
Thanks Mathieu.

closed 2019-07-01 by elpimous