introlab/rtabmap #409: T265 relocation conflicts with RTAB-Map relocation

matlabbe, 2019-06-06:
There is currently [no way to disable](https://github.com/IntelRealSense/librealsense/issues/3516#issuecomment-474828004) relocation feature from T265. A workaround is to integrate linear and angular velocities like in this post: https://github.com/IntelRealSense/realsense-ros/issues/779
Modifications would need to be done here:
https://github.com/introlab/rtabmap/blob/e09a872fa2149482e0e4ae3827176c8390da8517/corelib/src/camera/CameraRealSense2.cpp#L210-L230 

Linear and angular velocites can be fetched like [this](https://github.com/IntelRealSense/realsense-ros/blob/2b3365f5c96fcafa53412881e8e4484a6648bc23/realsense2_camera/src/base_realsense_node.cpp#L1304-L1320).

Note that on ROS, one may use [robot_localization](http://docs.ros.org/melodic/api/robot_localization/html/index.html) package to integrate the twist coming from realsense odometry.

macmason, 2019-06-19:
Upvoting here (and at #779) as it's not clear which issue is the "right" one.

matlabbe, 2019-10-02:
Realsense relocalization is now disabled, see the commit above.

commit 3ab5ec21 (2019-10-02): Fixed #409
closed 2019-10-02 by matlabbe