introlab/rtabmap #410: How to output the location of relasnese in the map that has been built

MattixLee, 2019-06-07:
@matlabbe Hello,

Thanks for your work! In win10, I installed RTAB-MAP.exe, I can build a map with good quality. 

when I use localization,  I can see a position in the window, but no specific location information is output at the terminal. I want to get a specific location information with C++ projects in win10 , so I  can navigation with the location information.

Please give me some help. Thank you!

Cheers

matlabbe, 2019-06-07:
Did you look at the [Event-based C++ example](https://github.com/introlab/rtabmap/tree/master/examples/RGBDMapping) and the [Non-event C++ example](https://github.com/introlab/rtabmap/tree/master/examples/NoEventsExample)?

For the first example, the odometry pose transformed in the /map frame can estimated like [here](https://github.com/introlab/rtabmap/blob/b5f5623af4fe23580d5d9077c454b615d81f0977/examples/RGBDMapping/MapBuilder.h#L156) in the odometry callback:
```cpp
Transform poseInMapFrame = odometryCorrection_*odom.pose();
```
`odometryCorrection_` is updated in the mapping statistics callback here:
https://github.com/introlab/rtabmap/blob/b5f5623af4fe23580d5d9077c454b615d81f0977/examples/RGBDMapping/MapBuilder.h#L275

For the second example, you could get the pose like this [here](https://github.com/introlab/rtabmap/blob/b5f5623af4fe23580d5d9077c454b615d81f0977/examples/NoEventsExample/main.cpp#L137):
```cpp
Transform poseInMapFrame = rtabmap.getMapCorrection()*pose;
```

cheers,
Mathieu

MattixLee, 2019-06-08:
@matlabbe hi,
Thank you for your help! I used vs2013 in win10, the project of the first example has been configured and completed.But a error occured, 503 error C2039: "getLastSignatureData": Not a member of "rtabmap:: Statistics". In win10, I installed RTABMap-0.18.0-win64.exe, MapBuilder.h and main.cpp were copied from introlab/rtabmap/examples/RGBDMapping.

I guess if the name of the function has changed, please help me to solve the problem. Thank you!

cheers,
Lee

MattixLee, 2019-06-08:
@matlabbe hi,

I ran the rtabmap-rgbd_mapping in ubuntu 16.04, I think it is got after the  event-based C++ example is compiled. When  rtabmap-rgbd_mapping was run, the window was appear, the window looked like in mapping mode, can this project achieve locaclization mode? eg:  the c++ projects read the bilt map, if realsense sees part of the map, the project can return to where realsense is in the map coordinate system at this time.

cheers,
Lee

matlabbe, 2019-06-10:
For windows, look at the examples under 0.18.0 tag: https://github.com/introlab/rtabmap/tree/0.18.0/examples

Yes it is possible to do localization in a previous map with small changes in the example. For example instead of:
https://github.com/introlab/rtabmap/blob/b5f5623af4fe23580d5d9077c454b615d81f0977/examples/RGBDMapping/main.cpp#L180
it could be:
```cpp
params.insert(ParametersPair(Parameters::kMemIncrementalMemory(), "false"))); // Localization Mode
rtabmap->init(params, my_previous_map.db); 
```

The map won't appear in the MapBuilder window, as only new clouds are created in this example. We would need to initialize MapBuilder with the data from the database before starting the loop, similar to:
https://github.com/introlab/rtabmap/blob/b5f5623af4fe23580d5d9077c454b615d81f0977/examples/RGBDMapping/MapBuilder.h#L197-L214
but with all signatures loaded from database with:
```cpp
std::map<int, Signature> signatures;
rtabmap.get3DMap(signatures, ...);
```

cheers,
Mathieu

MattixLee, 2019-06-10:
@matlabbe 

Thank you very much for your patient reply！Firstly, I want to run the example you gave，then I try to modify codes to realize localization. But I met a error, 503 error C2039: "getLastSignatureData": Not a member of "rtabmap:: Statistics". 

In win10, I installed RTABMap-0.18.0-win64.exe, MapBuilder.h and main.cpp were copied from introlab/rtabmap/examples/RGBDMapping, they were compiled with vs2013.

Thank you very much!
cheers,
Lee

MattixLee, 2019-06-10:
@matlabbe Thank you very much for your patient reply！Firstly, I want to run the example you gave，then I try to modify codes to realize localization. But I met a error, 503 error C2039: "getLastSignatureData": Not a member of "rtabmap:: Statistics".

Finally, I found that the functions in the Statistics.h were different. In 
  github/rtabmap/corelib/include/rtabmap/core/Statistics.h,
// getters
	bool extended() const {return _extended;}
	int refImageId() const {return _refImageId;}
	int refImageMapId() const {return _refImageMapId;}
	int loopClosureId() const {return _loopClosureId;}
	int loopClosureMapId() const {return _loopClosureMapId;}
	int proximityDetectionId() const {return _proximiyDetectionId;}
	int proximityDetectionMapId() const {return _proximiyDetectionMapId;}
	double stamp() const {return _stamp;}
        const Signature & getLastSignatureData() const {return _lastSignatureData;}

In win10, I installed RTABMap-0.18.0-win64.exe, MapBuilder.h and main.cpp were copied from introlab/rtabmap/examples/RGBDMapping, they were compiled with vs2013.
In my computer, D:\RTABMAP\include\rtabmap-0.18\rtabmap\core\Statistics.h,
	// getters
	bool extended() const {return _extended;}
	int refImageId() const {return _refImageId;}
	int loopClosureId() const {return _loopClosureId;}
	int proximityDetectionId() const {return _proximiyDetectionId;}
	double stamp() const {return _stamp;}
	const std::map<int, Signature> & getSignatures() const {return _signatures;}

Is getLastSignatureData()  and getSignatures() the same function? How should I deal with this problem?

Thank you very much!
cheers,
Lee

matlabbe, 2019-06-11:
Yes I said look in [0.18.0 tag branch](https://github.com/introlab/rtabmap/tree/0.18.0/examples), not the [master branch](https://github.com/introlab/rtabmap/tree/master/examples). The API changes over time. In 0.18.0, there is [Statistics::getSignatures()](https://github.com/introlab/rtabmap/blob/f670f71d47fc82776efd74f0743f590e2dd652b2/corelib/include/rtabmap/core/Statistics.h#L207), on master it is replaced by [Statistics::getLastSignatureData()](https://github.com/introlab/rtabmap/blob/b5f5623af4fe23580d5d9077c454b615d81f0977/corelib/include/rtabmap/core/Statistics.h#L218).

MattixLee, 2019-06-11:
Thank you for your reply! I'm sorry I didn't distinguish master from 0.18.0 tag branch,

I compiled the RGBD_Mapping(0.18.0), but another problem was encountered, I tried some methods on the internet, but there are still mistakes. I wonder if you've ever had the same problem before，Can you help solve this problem?
error LNK2001: Unparsable external symbols "public: static struct QMetaObject const MapBuilder::staticMetaObject" (?staticMetaObject@MapBuilder@@2UQMetaObject@@B)            
 VS2013-Projects\location\location\main.obj	location

I want to solve this problem by compiling source code，opencv was not found. Visual Studio command prompt (it was open with "Execute as Administrator")
But I installed opencv3.0, what should I add the path of opencv?

D:\RTAB-Map_CMAKE\rtabmap-0.19.3\rtabmap-0.19.3\build>cmake -G "Visual Studio 12 2013 Win64" ..
-- The C compiler identification is MSVC 18.0.21005.1
-- The CXX compiler identification is MSVC 18.0.21005.1
-- Check for working C compiler: D:/VS2013/VC/bin/x86_amd64/cl.exe
-- Check for working C compiler: D:/VS2013/VC/bin/x86_amd64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: D:/VS2013/VC/bin/x86_amd64/cl.exe
-- Check for working CXX compiler: D:/VS2013/VC/bin/x86_amd64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:64 (cmake_policy):
  The OLD behavior for policy CMP0020 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:67 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Error at CMakeLists.txt:196 (FIND_PACKAGE):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "D:/RTAB-Map_CMAKE/rtabmap-0.19.3/rtabmap-0.19.3/build/CMakeFiles/CMakeOutput.log". 

cheers,
Lee

matlabbe, 2019-06-12:
On Windows, you can refer to AppVeyor build to see how cmake is called (setting OpenCV_DIR path): https://github.com/introlab/rtabmap/blob/b5f5623af4fe23580d5d9077c454b615d81f0977/.appveyor.yml#L123

MattixLee, 2019-06-12:
Thank you for your patient reply! 
In win10, the files generated by RTAB-MAP.exe  were used, when Q_OBJECT was deleted in MapBuilder.h, the projects could be compiled successfully.
Although I don't know why,. 

cheers,
Lee

closed 2019-06-20 by MattixLee