introlab/rtabmap #411: Mapping accuracy vs Visual Odometry accuracy.

mpottinger, 2019-06-10:
Hello, this is not so much an issue but just a question that I am really curious about.

I  am developing an object detection/counting app that needs to keep track of objects and count them in 3d space. 

There are two ways I could go about it, one is to use mapping software such as rtabmap, map the space, then detect from a top down view of the map.

The other way is to use an approach similar to augmented reality where I do live object detection and place markers, tracking the position of the camera using a realsense T265 or any other visual odometry, with no mapping other than positions of the objects.

The thing I have noticed is that AR type apps including with the t265 tracking camera, there can be noticeable drift if the camera moves to much, which would invalidate the positions of my markers and make counting innacurate.

The question I have is if there is anything inherent to a mapping algorithm like rtabmap or anything else that makes it more accurate than the pose tracking of the camera? Or is the accuracy of the mapping totally dependent on the localization algorithm such as orb-slam or the t265?

The computational power needed for full mapping is obviously higher, I just wonder if there is any benefit that comes from that in terms of tracking the positions of things, or if lightweight tracking with just the t265 is as accurate as it can get?

Hope someone has some insight! Thanks!

matlabbe, 2019-06-11:
Actually the T265 does some kind of relocalization, not sure how they deal with all previous poses though when this happens. It is like the camera just jumps back on relocalization, without map corrections.

With rtabmap or orbslam2, the map is corrected on relocalization, so the localization of the markers will be more accurate overtime. 

The key difference between odometry and SLAM is that odometry will drift overtime, with no way to recorrect the poses when coming back to a seen location. With SLAM, the whole map will be deformed when coming back to a previous location, correcting the odometry drift.

Note that there is an open issue with T265 and rtabmap about conflicting relocation feature of T265 at the same time than rtabmap: https://github.com/introlab/rtabmap/issues/409

cheers,
Mathieu

mpottinger, 2019-06-11:
@matlabbe Ok, thank you, I am beginning to wrap my head around this! 

Yes I have seen the relocalization of the t265 in action. Sometimes it didn't seem perfect but it does suddenly jump sometimes as it corrects. It would be nice to have the option to disable it in a future update.

I am familiar with anchors in ARCore which I assume is the same kind of thing, just with a bit more control where you can say 'remember this spot right here.'

I found ARCore the most impressive at doing that, but still even then the positions can be off by a bit.

So what I get from this, I think, is if I am doing this object tracking over a large areas I will be best to use rtabmap, but in a small area, just visual odometry might be good enough, and that the advantage of rtabmap would be the relocalization accuracy over large distances, similar to arcore or better.

I am perhaps over analyzing and falling into the trap of 'premature optimization'. 

I only need to track in a small area the size of an individual warehouse rack anyway. Barely a square meter or two.

So maybe it makes no difference which route I go. I will try moving forward with just the tracking camera and if that fails at least I know rtabmap does something more under the hood that may help! Thanks!

matlabbe, 2019-10-02:
Update: https://github.com/introlab/rtabmap/issues/409 has been closed, relocalization of realsense is now disabled when running rtabmap.

closed 2019-10-02 by matlabbe