introlab/rtabmap #412: Point Cloud formation on empty space while taking a turn

yashagarwal9, 2019-06-14:
I have been using rtabmap for mapping the indoor environment. I have been making a map of a rectangular corridor but the problem that I'm facing is that while approaching the corners the point cloud is formed in empty space. I have tried filtering the point cloud but it doesn't work. Is there any other parameter using which I can solve this problem? Or any suggestions while mapping the environment.
[image]

matlabbe, 2019-06-16:
Is it a stereo camera? You could decrease the range of the cloud created in MapCloud display (look for Max Depth), this can filter far noisy data.

yashagarwal9, 2019-06-17:
Thanks @matlabbe  for your reply. I tried setting MaxDepth to 2 and that worked but the 2D map generated is the same.

matlabbe, 2019-06-17:
Hi,

For the occupancy grid, we should set `Grid/RangeMax` parameter of `rtabmap` node to 2 meters for example.

cheers,
Mathieu

yashagarwal9, 2019-06-17:
Hi, 
That worked just fine. 
I'm using the ZED mini and camera odometry as rgbg_odomety did not worked for me. It always gave me the loop closure rejection. Not enough inliers 0/20 even after tuning the suggested parameters. What can be done to update the map if new object appears in the incoming video stream?

matlabbe, 2019-06-20:
I think the ZED Mini can rely for some time on IMU when there are not enough visual features, which make it more robust in that case. However, there could be more drift in those areas, which will impact how good the map will be optimized after loop closure.

The default parameters won't clear dynamic obstacles after they moved (environment is assumed static). To clear them, we should enable ray tracing (`Grid/RayTracing=true`). This can be done in 2D or 3D. If you are moving in 2D, I recommend to set `Grid/3D=false`. 3D obstacles are still detected but ray tracing will be done in 2D. For the 3D case, an Octomap will be built, which requires a lot more computation power. See Figures 14-15-16-17 and Table 10 of this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf) for a computation/memory/quality comparison.

The 2D case I am referring (`Grid/RayTracing=true`, `Grid/3D=false`) is: GFD1→G3D0→GRT1→2D

cheers,
Mathieu