introlab/rtabmap #413: RGBD Sample bad_alloc Error

schmiran, 2019-06-20:
Hi guys, I'm trying to get the C++ examples to work and get a bad_alloc error in both examples.

I have installed RTABMap 0.18.0 over the Windows binaries. Below you can see what CMAKE output when configuring the "Cplusplus RGBD Mapping" example:

```
The C compiler identification is MSVC 19.16.27027.1
The CXX compiler identification is MSVC 19.16.27027.1
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
Found RTABMap: C:/Program Files/RTABMap/lib/rtabmap_core.lib;C:/Program Files/RTABMap/lib/rtabmap_utilite.lib;C:/Program Files/RTABMap/lib/rtabmap_gui.lib;C:/Program Files/zlib/lib/zlib.lib;C:/Program Files/g2o/lib/g2o_core.lib;C:/Program Files/g2o/lib/g2o_types_slam2d.lib;C:/Program Files/g2o/lib/g2o_types_slam3d.lib;C:/Program Files/g2o/lib/g2o_types_sba.lib;C:/Program Files/g2o/lib/g2o_stuff.lib;C:/Program Files/g2o/lib/g2o_solver_csparse.lib;C:/Program Files/g2o/lib/g2o_csparse_extension.lib;C:/Program Files/g2o/lib/g2o_ext_csparse.lib;C:/Intel/OpenCL/sdk/lib/x64/OpenCL.lib;C:/workspace/libfreenect2/build_nocuda/install/lib/freenect2.lib;C:/Program Files/OpenNI2/Lib/OpenNI2.lib;C:/Program Files/Point Grey Research/FlyCapture2/lib64/vs2015/FlyCapture2_v140.lib;C:/Program Files/FLIR Integrated Imaging Solutions/Triclops Stereo Vision SDK/lib64/triclops_v140.lib;C:/Program Files/FLIR Integrated Imaging Solutions/Triclops Stereo Vision SDK/lib64/flycapture2bridge_v140.lib;C:/Program Files/Realsense/lib/realsense.lib;C:/Program Files (x86)/Intel RealSense SDK 2.0/lib/x64/realsense2.lib;C:/Program Files/octomap-distribution/lib/octomap.lib;C:/Program Files/octomap-distribution/lib/octomath.lib;C:/Program Files/cpu_tsdf/lib/cpu_tsdf.lib;C:/Program Files/open_chisel/lib/open_chisel.lib  
OpenCV ARCH: x64
OpenCV RUNTIME: vc15
OpenCV STATIC: OFF
Found OpenCV: C:/opencv/build (found version "4.1.0") 
Found OpenCV 4.1.0 in C:/opencv/build/x64/vc15/lib
You might need to add C:\opencv\build\x64\vc15\bin to your PATH to be able to run your applications.
CMake Warning (dev) at CMakeLists.txt:25 (FIND_PACKAGE):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Environment variable PCL_ROOT is set to:

    C:\Program Files\PCL 1.9.1

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

Found Eigen: C:/Program Files/PCL 1.9.1/3rdParty/Eigen/eigen3  
Eigen found (include: C:/Program Files/PCL 1.9.1/3rdParty/Eigen/eigen3, version: 3.3.5)
Looking for pthread.h
Looking for pthread.h - not found
Found Threads: TRUE  
Boost version: 1.68.0
Found the following Boost libraries:
  system
  filesystem
  thread
  date_time
  iostreams
  serialization
  chrono
  atomic
  regex
Found FLANN: C:/Program Files/PCL 1.9.1/3rdParty/FLANN/lib/flann_cpp_s.lib  
FLANN found (include: C:/Program Files/PCL 1.9.1/3rdParty/FLANN/include, lib: optimized;C:/Program Files/PCL 1.9.1/3rdParty/FLANN/lib/flann_cpp_s.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/FLANN/lib/flann_cpp_s-gd.lib)
FLANN found (include: C:/Program Files/PCL 1.9.1/3rdParty/FLANN/include, lib: optimized;C:/Program Files/PCL 1.9.1/3rdParty/FLANN/lib/flann_cpp_s.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/FLANN/lib/flann_cpp_s-gd.lib)
Found OpenNI2: C:/Program Files/OpenNI2/Lib/OpenNI2.lib  
OpenNI2 found (include: C:/Program Files/OpenNI2/Include, lib: C:/Program Files/OpenNI2/Lib/OpenNI2.lib)
OpenNI2 found (include: C:/Program Files/OpenNI2/Include, lib: C:/Program Files/OpenNI2/Lib/OpenNI2.lib)
Found Qhull: optimized;C:/Program Files/PCL 1.9.1/3rdParty/Qhull/lib/qhullstatic.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Qhull/lib/qhullstatic_d.lib  
QHULL found (include: C:/Program Files/PCL 1.9.1/3rdParty/Qhull/include, lib: optimized;C:/Program Files/PCL 1.9.1/3rdParty/Qhull/lib/qhullstatic.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Qhull/lib/qhullstatic_d.lib)
looking for PCL_COMMON
Found PCL_COMMON: C:/Program Files/PCL 1.9.1/lib/pcl_common_release.lib  
looking for PCL_KDTREE
Found PCL_KDTREE: C:/Program Files/PCL 1.9.1/lib/pcl_kdtree_release.lib  
looking for PCL_OCTREE
Found PCL_OCTREE: C:/Program Files/PCL 1.9.1/lib/pcl_octree_release.lib  
looking for PCL_SEARCH
Found PCL_SEARCH: C:/Program Files/PCL 1.9.1/lib/pcl_search_release.lib  
[... 27 lines cut ...]
Found PCL_RECOGNITION: C:/Program Files/PCL 1.9.1/lib/pcl_recognition_release.lib  
looking for PCL_STEREO
Found PCL_STEREO: C:/Program Files/PCL 1.9.1/lib/pcl_stereo_release.lib  
looking for PCL_OUTOFCORE
Found PCL_OUTOFCORE: C:/Program Files/PCL 1.9.1/lib/pcl_outofcore_release.lib  
looking for PCL_PEOPLE
Found PCL_PEOPLE: C:/Program Files/PCL 1.9.1/lib/pcl_people_release.lib  
Found PCL: pcl_common;pcl_kdtree;pcl_octree;pcl_search;pcl_sample_consensus;pcl_filters;pcl_io;pcl_features;pcl_ml;pcl_segmentation;pcl_visualization;pcl_surface;pcl_registration;pcl_keypoints;pcl_tracking;pcl_recognition;pcl_stereo;pcl_outofcore;pcl_people;optimized;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_system-vc141-mt-x64-1_68.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_system-vc141-mt-gd-x64-1_68.lib;optimized;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_filesystem-vc141-mt-x64-1_68.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_filesystem-vc141-mt-gd-x64-1_68.lib;optimized;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_thread-vc141-mt-x64-1_68.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_thread-vc141-mt-gd-x64-1_68.lib;optimized;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_date_time-vc141-mt-x64-1_68.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_date_time-vc141-mt-gd-x64-1_68.lib;optimized;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_iostreams-vc141-mt-x64-1_68.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_iostreams-vc141-mt-gd-x64-1_68.lib;optimized;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_serialization-vc141-mt-x64-1_68.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_serialization-vc141-mt-gd-x64-1_68.lib;optimized;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_chrono-vc141-mt-x64-1_68.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_chrono-vc141-mt-gd-x64-1_68.lib;optimized;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_atomic-vc141-mt-x64-1_68.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_atomic-vc141-mt-gd-x64-1_68.lib;optimized;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_regex-vc141-mt-x64-1_68.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Boost/lib/libboost_regex-vc141-mt-gd-x64-1_68.lib;optimized;C:/Program Files/PCL 1.9.1/3rdParty/Qhull/lib/qhullstatic.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/Qhull/lib/qhullstatic_d.lib;C:/Program Files/OpenNI2/Lib/OpenNI2.lib;optimized;C:/Program Files/PCL 1.9.1/3rdParty/FLANN/lib/flann_cpp_s.lib;debug;C:/Program Files/PCL 1.9.1/3rdParty/FLANN/lib/flann_cpp_s-gd.lib;vtksys;vtkCommonCore;vtkCommonMath;vtkCommonMisc;vtkCommonSystem;vtkCommonTransforms;vtkCommonDataModel;vtkCommonColor;vtkCommonExecutionModel;vtkCommonComputationalGeometry;vtkFiltersCore;vtkFiltersGeneral;vtkImagingCore;vtkImagingFourier;vtkalglib;vtkFiltersStatistics;vtkFiltersExtraction;vtkInfovisCore;vtkFiltersGeometry;vtkFiltersSources;vtkRenderingCore;vtkzlib;vtkfreetype;vtkRenderingFreeType;vtkRenderingContext2D;vtkChartsCore;vtkDICOMParser;vtklz4;vtkIOCore;vtkIOLegacy;vtkexpat;vtkIOXMLParser;vtkDomainsChemistry;vtkIOXML;vtkParallelCore;vtkFiltersAMR;vtkFiltersFlowPaths;vtkFiltersG [... cut ...]
  behavior and not rely on setting a policy to OLD.

Configuring done
```

The generation of the solution file was no problem and after I cleaned up the dlls, it built without any problems. Now my Problem:

I have connected an Intel Realsense D415 and therefore choose the RealSense2 driver. In line 149 `camera = new CameraRealSense2("", 0, opticalRotation);` I get kicked out because of a std::bad_alloc error. I have no idea what I did wrong. 

I hope you can give me a hint as to why that is. Many thanks in advance!

matlabbe, 2019-06-20:
Is the standalone working on your computer with realsense2 driver and this camera?

matlabbe, 2019-06-20:
Note that you would also have to rebuild rtabmap from source as the binaries may not be built with the same versions of the dependencies you have on your computer. If not the same version of the dll is used, this can cause segmentation faults.

schmiran, 2019-06-21:
Hello, thanks for your answer!

Yes, the standalone version works with the sensor and also with others I have here. Thanks for the tip, that makes sensethen. I'll try again to build from source, that didn't work out for me at the beginning.

schmiran, 2019-07-02:
I have now built RTABMap from Source. Unfortunately I still get a `std_bad_alloc` error at the same line in the RGB-D mapping example. Do you have another idea why this might be?

closed 2019-07-02 by schmiran