introlab/rtabmap #416: Occupancy Grid map advise

xilef, 2019-06-23:
Hi all,

Not really an issue, I just need some advise. I am using the rtabmap_ros package on a simulated environment and robot in gazebo. During mapping whenever a loop closure happens the occupancy grid map seemingly 'jumps' all over the place.

How the world looks like:
[image]

Map in rviz:
[image]

For this instance I only mapped the main room for testing. If I ever enter one of the rooms the jumping gets even worse.

I am using the following launch file. Still trying out several params but I have not yet managed to get a decent map:
`<launch>
  
  <arg name="database_path"     default="rtabmap.db"/>
  <arg name="rgb_topic"   default="/camera/rgb/image_raw"/>
  <arg name="depth_topic" default="/camera/depth/image_raw"/>
  <arg name="camera_info_topic" default="/camera/rgb/camera_info"/>  

  
  <group ns="rtabmap">
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">

      
      <param name="database_path"       type="string" value="$(arg database_path)"/>
      <param name="frame_id"            type="string" value="robot_footprint"/>
      <param name="odom_frame_id"       type="string" value="odom"/>
      <param name="odom_tf_linear_variance"	type="string" value="0.001"/>
      <param name="odom_tf_angular_variance" type="string" value="0.001"/>
      <param name="subscribe_depth"     type="bool" value="true"/>
      <param name="subscribe_scan"      type="bool" value="true"/>
      <param name="approx_sync"			type="bool" value="true"/>

      
      <remap from="scan" to="/scan"/>
      <remap from="rgb/image" to="$(arg rgb_topic)"/>
      <remap from="depth/image" to="$(arg depth_topic)"/>
      <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>

      
      <remap from="grid_map" to="/map"/>

      
      <param name="Rtabmap/DetectionRate"		type="string" value="5"/>

      
      <param name="Reg/Force3DoF"				type="string" value="true"/>

      
      
      <param name="Reg/Strategy"				type="string" value="0"/>

      
      
      <param name="Kp/DetectorStrategy"			type="string" value="0"/>

      
      <param name="Kp/MaxFeatures"				type="string" value="600"/>

      
      <param name="SURF/HessianThreshold"		type="string" value="300"/>

      
      <param name="Vis/MinInliers"				type="string" value="5"/>
      <param name="Vis/FeatureType"				type="string" value="0"/>

      
      <param name="Mem/NotLinkedNodesKept"		type="string" value="false"/>
      
      <param name="RGBD/OptimizeMaxError"		type="string" value="0.0"/>
      
      <param name="RGBD/AngularUpdate"			type="string" value="0.1"/>
      <param name="RGBD/LinearUpdate"			type="string" value="0.1"/>
      <param name="RGBD/LoopClosureReextractFeatures" type="string" value="true"/>
      
      <param name="Odom/Holonomic"				type="string" value="false"/>
      
      <param name="Grid/FromDepth"				type="string" value="false"/>
      
      <param name="Optimizer/Strategy"			type="string" value="1"/>
      <param name="Optimizer/Robust"			type="string" value="true"/>
      
    </node>
    
    
    <node pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
        <param name="subscribe_depth"             type="bool" value="true"/>
        <param name="subscribe_scan"              type="bool" value="true"/>
        <param name="frame_id"                    type="string" value="robot_footprint"/>

        <remap from="rgb/image"       to="$(arg rgb_topic)"/>
        <remap from="depth/image"     to="$(arg depth_topic)"/>
        <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
        <remap from="scan"            to="/scan"/>
    </node>
  </group>
</launch>`

Here is the db:
[http://gofile.me/2AjJo/sio6RNsVO](url)

Thanks and regards,
Felix Isaac Palomares

matlabbe, 2019-06-24:
The link is "connecting..." endlessly. 

If you have a lidar, set `Reg/Strategy` to 1. `RGBD/LoopClosureReextractFeatures` parameter is not neeed anymore. I would not disable `RGBD/OptimizeMaxError`, keeping at 3 (while disabling `Optimizer/Robust`). If too many good loop closures are rejected, you may adjust the odometry variance instead (`odom_tf_linear_variance` and `odom_tf_angular_variance`). I would increase `Vis/MinInliers` to 15/20 to avoid bad loop closure transformations. `Rtabmap/DetectionRate` could be kept at 1 Hz unless your robot moves very fast.

cheers,
Mathieu

xilef, 2019-06-24:
Hi Mathieu,

Thank you for that! I'll try them out on see what happens.

I have also updated the DB link hopefully it works now:
[https://www.dropbox.com/s/7wv1xbztr4iq2jx/rtabmap.db?dl=0](url)

Thanks and regards,
Felix Isaac Palomares

matlabbe, 2019-06-24:
Looking at the database with `rtabmap-databaseViewer`, the camera is missing the optical rotation. The camera is then looking upward instead of forward, which may create very wrong loop closure transformations.
[image]

Are the frames defined by the urdf of the robot or they defined with `static_transform_publisher`? What are they? Which `frame_id` is set in RGB image topic?

The parameters referred in my previous post still apply.

cheers,
Mathieu

xilef, 2019-06-25:
Hi Matthieu,

I loaded my world, loaded rviz, added the camera display and verified it is pointing forward. Added the depth point cloud and lo and behold that thing is pointed upwards. After fixing the links, I ran the mapping again without changing any values yet and now it is generating the occupancy grid and 3d map correctly. Thank you very much for pointing out the frame_id. Next will be testing out your suggested params.

Thanks and regards,
Felix Palomares

closed 2019-06-25 by xilef