introlab/rtabmap #420: some errors about g2o while compiling

RichardYao1995, 2019-07-02:
My g2o version is the 2017.
when I run make -j4
it comes to the following problem 
can anyone help me？

[ 83%] Building CXX object tools/KittiDataset/CMakeFiles/kitti_dataset.dir/main.cpp.o
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::EdgeProjectP2SC::EdgeProjectP2SC()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::Vertex::getMinimalEstimateData(double*) const'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::EdgeSE2Prior::EdgeSE2Prior()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::SparseOptimizer::setAlgorithm(g2o::OptimizationAlgorithm*)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `typeinfo for g2o::VertexCam'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::ParameterSE3Offset::ParameterSE3Offset()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::MarginalCovarianceCholesky::computeCovariance(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >&, std::vector<int, std::allocator<int> > const&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > const&)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::CacheContainer::findCache(g2o::Cache::CacheKey const&)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::Vertex::updateCache()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::VertexSBAPointXYZ::VertexSBAPointXYZ()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cholmod_change_factor'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::RobustKernel::setDelta(double)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::Edge::setMeasurementFromState()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cs_di_pinv'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `vtable for g2o::RobustKernelHuber'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::SparseOptimizer::setVerbose(bool)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::Vertex::cacheContainer()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `typeinfo for g2o::OptimizableGraph::Vertex'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::Edge::Edge()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::MarginalCovarianceCholesky::setCholeskyFactor(int, int*, int*, double*, int*)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::EdgeSE2XYPrior::EdgeSE2XYPrior()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::SparseOptimizer::optimize(int, bool)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `typeinfo for g2o::EdgeProjectP2SC'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::EdgeSE2PointXY::EdgeSE2PointXY()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::Factory::instance()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cholmod_factorize'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::Solver::~Solver()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cholmod_analyze_p'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cs_di_nfree'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cs_di_symperm'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::writeCCSMatrix(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, int, int const*, int const*, double const*, bool)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cholmod_analyze'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::Solver::resizeVector(unsigned long)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::MarginalCovarianceCholesky::computeCovariance(double**, std::vector<int, std::allocator<int> > const&)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::G2OBatchStatistics::_globalStats'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::Vertex::minimalEstimateDimension() const'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::EdgeSBACam::EdgeSBACam()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::Edge::setRobustKernel(g2o::RobustKernel*)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::Cache::CacheKey::CacheKey(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<g2o::Parameter*, std::allocator<g2o::Parameter*> > const&)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `typeinfo for g2o::VertexSE3'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::EdgeProjectP2MC::EdgeProjectP2MC()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::SparseOptimizer::initializeOptimization(int)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::Parameter::setId(int)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::csparse_extension::writeCs2Octave(char const*, cs_di_sparse const*, bool)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cholmod_amd'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::ParameterSE2Offset::ParameterSE2Offset()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::Edge::~Edge()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `typeinfo for g2o::OptimizableGraph::Edge'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::EdgeSE3Prior::EdgeSE3Prior()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::EdgeSE3::EdgeSE3()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::VertexPointXY::VertexPointXY()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::Edge::setParameterId(int, int)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::EdgeSE3PointXYZ::EdgeSE3PointXYZ()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::SparseOptimizer::activeRobustChi2() const'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::Edge::measurementDimension() const'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `typeinfo for g2o::HyperGraph::Vertex'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::MarginalCovarianceCholesky::~MarginalCovarianceCholesky()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `typeinfo for g2o::Cache'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cholmod_finish'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cs_di_post'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::VertexSE2::VertexSE2()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::SparseOptimizer::computeMarginals(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > const&)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cs_di_etree'
[... 35 lines cut ...]
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::VertexSE3::VertexSE3()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::addVertex(g2o::HyperGraph::Vertex*, g2o::HyperGraph::Data*)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::internal::fromVectorMQT(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cs_di_malloc'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::Vertex::Vertex()'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `cs_di_free'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::OptimizableGraph::addEdge(g2o::HyperGraph::Edge*)'
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::HyperGraph::vertex(int)'
[... cut ...]
../../../bin/librtabmap_core.so.0.19.4: undefined reference to `g2o::RobustKernel::RobustKernel()'
collect2: error: ld returned 1 exit status
tools/ConsoleApp/CMakeFiles/consoleApp.dir/build.make:417: recipe for target '../bin/rtabmap-console' failed
make[2]: *** [../bin/rtabmap-console] Error 1
CMakeFiles/Makefile2:541: recipe for target 'tools/ConsoleApp/CMakeFiles/consoleApp.dir/all' failed
make[1]: *** [tools/ConsoleApp/CMakeFiles/consoleApp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

matlabbe, 2019-07-03:
I cannot reproduce those problems on my system with g2o 2017 release. What is your system?

srinath-iko, 2019-07-05:
Feels like either
a) You havent installed G2O correctly
b) Rtabmap isnt finding G2O while making.

If you could remove the CMakeCache and show the output of cmake .. , it can give a better idea of where this error is getting produced

RichardYao1995, 2019-07-05:
this happens from an incorrect installation on g2o which lacks some parts but returns a correct installation result. I reinstall the g2o after installing some lacking package and now it works.
thanks

closed 2019-07-05 by RichardYao1995