introlab/rtabmap #422: LNK2005:  ... already defined in libboost_serialization-vc140-mt-1_62.lib

leoteo1, 2019-07-09:
I've got error with boost library when I build "rtabmap_core", since there are several function have been declared in the same name of libboost_serialization-vc140-mt-1_62.lib and boost_serialization-vc140-mt-1_62.lib.

**Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2005	"protected: void __cdecl boost::serialization::extended_type_info::key_unregister(void)const " (?key_unregister@extended_type_info@serialization@boost@@IEBAXXZ) already defined in libboost_serialization-vc140-mt-1_62.lib(extended_type_info.obj)	rtabmap_core	G:\Room scanning\rtabmap\rtabmap-master\build\corelib\src\boost_serialization-vc140-mt-1_62.lib(boost_serialization-vc140-mt-1_62.dll)	1**	

Can anyone tell me how to solve this probem?
Thanks,

matlabbe, 2019-07-09:
You can take a look at the appveyor script for Windows compilation: https://github.com/introlab/rtabmap/blob/master/.appveyor.yml

I am not sure why there is a `boost_serialization-vc140-mt-1_62.lib` in the `src` directory of rtabmap, as it should not.

leoteo1, 2019-07-10:
> You can take a look at the appveyor script for Windows compilation: https://github.com/introlab/rtabmap/blob/master/.appveyor.yml
> 
> I am not sure why there is a `boost_serialization-vc140-mt-1_62.lib` in the `src` directory of rtabmap, as it should not.

Thanks for your quick response!!!

There is no any *.lib file in the src but the **boost_** and **Libboost_** are listed in Input of Linker, are there something wrong in my CMake configuration??

..\..\..\lib\rtabmap_utilite.lib
C:\opencv\build\x64\vc14\lib\opencv_world343.lib
**C:\PCL 1.8.1\3rdParty\boost_1_62_0\lib64-msvc-14.0\libboost_system-vc140-mt-1_62.lib
C:\PCL 1.8.1\3rdParty\boost_1_62_0\lib64-msvc-14.0\libboost_filesystem-vc140-mt-1_62.lib
C:\PCL 1.8.1\3rdParty\boost_1_62_0\lib64-msvc-14.0\libboost_thread-vc140-mt-1_62.lib
C:\PCL 1.8.1\3rdParty\boost_1_62_0\lib64-msvc-14.0\libboost_date_time-vc140-mt-1_62.lib
C:\PCL 1.8.1\3rdParty\boost_1_62_0\lib64-msvc-14.0\libboost_iostreams-vc140-mt-1_62.lib
C:\PCL 1.8.1\3rdParty\boost_1_62_0\lib64-msvc-14.0\libboost_serialization-vc140-mt-1_62.lib
C:\PCL 1.8.1\3rdParty\boost_1_62_0\lib64-msvc-14.0\libboost_chrono-vc140-mt-1_62.lib
C:\PCL 1.8.1\3rdParty\boost_1_62_0\lib64-msvc-14.0\libboost_atomic-vc140-mt-1_62.lib
C:\PCL 1.8.1\3rdParty\boost_1_62_0\lib64-msvc-14.0\libboost_regex-vc140-mt-1_62.lib**
C:\PCL 1.8.1\lib\pcl_common_release.lib
C:\PCL 1.8.1\lib\pcl_octree_release.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtksqlite-6.3.lib
C:\PCL 1.8.1\lib\pcl_io_release.lib
C:\PCL 1.8.1\3rdParty\FLANN\lib\flann_cpp_s.lib
C:\PCL 1.8.1\lib\pcl_kdtree_release.lib
C:\PCL 1.8.1\lib\pcl_search_release.lib
C:\PCL 1.8.1\3rdParty\Qhull\lib\qhull_p.lib
C:\PCL 1.8.1\lib\pcl_surface_release.lib
C:\PCL 1.8.1\lib\pcl_filters_release.lib
C:\PCL 1.8.1\lib\pcl_features_release.lib
C:\PCL 1.8.1\lib\pcl_registration_release.lib
C:\PCL 1.8.1\lib\pcl_sample_consensus_release.lib
C:\PCL 1.8.1\lib\pcl_ml_release.lib
C:\PCL 1.8.1\lib\pcl_segmentation_release.lib
C:\PCL 1.8.1\lib\pcl_visualization_release.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkDomainsChemistry-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkFiltersFlowPaths-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkFiltersGeneric-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkFiltersHyperTree-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkFiltersParallelImaging-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkFiltersProgrammable-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkFiltersSMP-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkFiltersSelection-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkFiltersVerdict-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkverdict-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkGUISupportQtOpenGL-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkGUISupportQtSQL-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOSQL-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkGeovisCore-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOAMR-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOEnSight-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOExodus-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOExport-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkRenderingGL2PS-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkRenderingContextOpenGL-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkgl2ps-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOImport-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOInfovis-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtklibxml2-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOLSDyna-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOMINC-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOMovie-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkoggtheora-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOPLY-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOParallel-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkjsoncpp-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOParallelXML-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkIOVideo-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkImagingMath-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkImagingMorphological-6.3.lib
[... 91 lines cut ...]
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkftgl-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkfreetype-6.3.lib
C:\PCL 1.8.1\3rdParty\VTK\lib\vtkzlib-6.3.lib
opengl32.lib
C:\Qt\Qt5.11.1\5.11.1\msvc2015_64\lib\Qt5Widgets.lib
C:\Qt\Qt5.11.1\5.11.1\msvc2015_64\lib\Qt5Gui.lib
C:\Qt\Qt5.11.1\5.11.1\msvc2015_64\lib\Qt5Core.lib
**C:\Libraries\boost_1_62_0\lib64-msvc-14.0\boost_serialization-vc140-mt-1_62.lib
C:\Libraries\boost_1_62_0\lib64-msvc-14.0\boost_system-vc140-mt-1_62.lib
C:\Libraries\boost_1_62_0\lib64-msvc-14.0\boost_filesystem-vc140-mt-1_62.lib
C:\Libraries\boost_1_62_0\lib64-msvc-14.0\boost_thread-vc140-mt-1_62.lib
C:\Libraries\boost_1_62_0\lib64-msvc-14.0\boost_date_time-vc140-mt-1_62.lib
C:\Libraries\boost_1_62_0\lib64-msvc-14.0\boost_timer-vc140-mt-1_62.lib
C:\Libraries\boost_1_62_0\lib64-msvc-14.0\boost_chrono-vc140-mt-1_62.lib**
C:\Rtabmap\3rdParty\GTSAM\lib\metis.lib
kernel32.lib
user32.lib
gdi32.lib
winspool.lib
shell32.lib
ole32.lib
oleaut32.lib
uuid.lib
comdlg32.lib
advapi32.lib