introlab/rtabmap #424: Keep reinstalling rtabmap

Williamc17, 2019-07-23:
It's **weird** I have to reinstall rtabmap a few days every time. After reinstalling, everythins seems **working well** but I got tedius kept doing this! 

> Just following the tutorial to reinstall rtabmap and rtabmap_ros.

But cannot `make` directly. So those cmake files just like disappered (I have no ideal why I have to `camke` again):

```
~/rtabmap/build$ make -j4
[  4%] Built target rtabmap_utilite
[  5%] Built target res_tool
[  5%] Built target imagesJoiner
[  6%] Built target extractObject
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libfreenect.so', needed by '../bin/librtabmap_core.so.0.19.1'.  Stop.
CMakeFiles/Makefile2:2386: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```

---

When running `roslaunch rtabmap_ros rtabmap.launch`, the only error I got is:
```
process[rtabmap/rgbd_odometry-2]: started with pid [24091]
/home/william/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry: error while loading shared libraries: libfreenect.so.0.5: cannot open shared object file: No such file or directory
process[rtabmap/rtabmap-3]: started with pid [24092]
/home/william/catkin_ws/devel/lib/rtabmap_ros/rtabmap: error while loading shared libraries: libfreenect.so.0.5: cannot open shared object file: No such file or directory
process[rtabmap/rtabmapviz-4]: started with pid [24093]
/home/william/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz: error while loading shared libraries: libfreenect.so.0.5: cannot open shared object file: No such file or directory
```

Here is the whole version of errors: (sorry for I don't know how to get the code folded)
```
~/catkin_ws/src/rtabmap_ros$ roslaunch rtabmap_ros rtabmap.launch 
... logging to /home/william/.ros/log/f8c15398-acfd-11e9-bd17-23b321bd4f48/roslaunch-william-System-Product-Name-24050.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://william-System-Product-Name:40743/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/frame_id: camera_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 1.0
 * /rtabmap/rtabmap/odom_tf_linear_variance: 1.0
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_normal_k: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmapviz/approx_sync: True
 * /rtabmap/rtabmapviz/frame_id: camera_link
 * /rtabmap/rtabmapviz/odom_frame_id: 
 * /rtabmap/rtabmapviz/queue_size: 10
 * /rtabmap/rtabmapviz/subscribe_depth: True
 * /rtabmap/rtabmapviz/subscribe_odom_info: True
 * /rtabmap/rtabmapviz/subscribe_rgbd: False
 * /rtabmap/rtabmapviz/subscribe_scan: False
 * /rtabmap/rtabmapviz/subscribe_scan_cloud: False
 * /rtabmap/rtabmapviz/subscribe_stereo: False
 * /rtabmap/rtabmapviz/wait_for_transform_duration: 0.2

NODES
  /rtabmap/
    rgbd_odometry (rtabmap_ros/rgbd_odometry)
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)

auto-starting new master
process[master]: started with pid [24060]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f8c15398-acfd-11e9-bd17-23b321bd4f48
process[rosout-1]: started with pid [24073]
started core service [/rosout]
process[rtabmap/rgbd_odometry-2]: started with pid [24091]
/home/william/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry: error while loading shared libraries: libfreenect.so.0.5: cannot open shared object file: No such file or directory
process[rtabmap/rtabmap-3]: started with pid [24092]
/home/william/catkin_ws/devel/lib/rtabmap_ros/rtabmap: error while loading shared libraries: libfreenect.so.0.5: cannot open shared object file: No such file or directory
process[rtabmap/rtabmapviz-4]: started with pid [24093]
/home/william/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz: error while loading shared libraries: libfreenect.so.0.5: cannot open shared object file: No such file or directory
[rtabmap/rgbd_odometry-2] process has died [pid 24091, exit code 127, cmd /home/william/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info rgbd_image:=rgbd_image_relay odom:=odom __name:=rgbd_odometry __log:=/home/william/.ros/log/f8c15398-acfd-11e9-bd17-23b321bd4f48/rtabmap-rgbd_odometry-2.log].
log file: /home/william/.ros/log/f8c15398-acfd-11e9-bd17-23b321bd4f48/rtabmap-rgbd_odometry-2*.log
[rtabmap/rtabmap-3] process has died [pid 24092, exit code 127, cmd /home/william/catkin_ws/devel/lib/rtabmap_ros/rtabmap rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info rgbd_image:=rgbd_image_relay left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud user_data:=/user_data user_data_async:=/user_data_async gps/fix:=/gps/fix tag_detections:=/tag_detections odom:=odom __name:=rtabmap __log:=/home/william/.ros/log/f8c15398-acfd-11e9-bd17-23b321bd4f48/rtabmap-rtabmap-3.log].
log file: /home/william/.ros/log/f8c15398-acfd-11e9-bd17-23b321bd4f48/rtabmap-rtabmap-3*.log
[rtabmap/rtabmapviz-4] process has died [pid 24093, exit code 127, cmd /home/william/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz -d ~/.ros/rtabmap_gui.ini rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info rgbd_image:=rgbd_image_relay left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud odom:=odom __name:=rtabmapviz __log:=/home/william/.ros/log/f8c15398-acfd-11e9-bd17-23b321bd4f48/rtabmap-rtabmapviz-4.log].
log file: /home/william/.ros/log/f8c15398-acfd-11e9-bd17-23b321bd4f48/rtabmap-rtabmapviz-4*.log
^C[rosout-1] killing on exit
^C[master] killing on exit
^Cshutting down processing monitor...
... shutting down processing monitor complete
done
```

matlabbe, 2019-07-23:
Hi,

Can you do `$ /home/william/catkin_ws/devel/lib/rtabmap_ros/rtabmap`?

It seems libfreenect has been uninstalled or upgraded to a different version. It could also be that you may have to update LD_LIBRARY__PATH environment variable with directory containing libfreenect.so so that the linker can find it.

Cheers,
Mathieu

Williamc17, 2019-07-24:
> Hi,
>
> Can you do `$ /home/william/catkin_ws/devel/lib/rtabmap_ros/rtabmap`?
>
> It seems libfreenect has been uninstalled or upgraded to a different version. It could also be that you may have to update [...]

When I do `$ /home/william/catkin_ws/devel/lib/rtabmap_ros/rtabmap`, error occurs:

```
/home/william/catkin_ws/devel/lib/rtabmap_ros/rtabmap: error while loading shared libraries: libfreenect.so.0.5: cannot open shared object file: No such file or directory
```

And I can locate `libfreenect.so`, it just like disappeared.

matlabbe, 2019-07-24:
Sorry, I meant `$ ldd /home/william/catkin_ws/devel/lib/rtabmap_ros/rtabmap` to see what the linker can find and cannot find. 

Can you run freenect_launch package?
`$ roslaunch freenect_launch freenect.launch`
To see if other apps have also problem finding that library.

Williamc17, 2019-07-24:
I got this error when running freenect_launch package.
```
$ roslaunch freenect_launch freenect.launch
[freenect.launch] is neither a launch file in package [freenect_launch] nor is [freenect_launch] a launch file name
The traceback for the exception was written to the log file
```

matlabbe, 2019-07-24:
Hi,

For freenect_launch, you can install it with: `sudo apt-get install ros-kinetic-freenect-launch` 

So this time rtabmap is ok? As I see in the ldd log:
`libfreenect.so.0.5 => /usr/lib/x86_64-linux-gnu/libfreenect.so.0.5 (0x00007f8107e0e000)` 

Your `libfreenect.so.0.5` exists in `/usr/lib/x86_64-linux-gnu`. Maybe sometimes the path `/usr/lib/x86_64-linux-gnu` is not correctly searched on runtime by the linker for some obscure reasons. Next time it fails, you could try do this before lanching rtabmap:
`$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/x86_64-linux-gnu`

or make sure you always source `setup.bash` from ros:
```bash
$ source /opt/ros/kinetic/setup.bash
$ source ~/catkin_ws/devel/setup.bash
```

Williamc17, 2019-07-31:
Sorry for the late reply, it's been tested both for freenect and rtabmap. I still can't find out why it would be like this. **But** now they are both working well after these days. 

Thanks for your suggestions.

closed 2019-07-31 by Williamc17