introlab/rtabmap #425: Build issue on Arch Linux using ROS Melodic

parkerlreed, 2019-07-27:
Installed the full melodic-desktop package and went to compile rtabmap.

Arch Linux
gcc 9.1.0-2

```
/home/parker/build/rtabmap/guilib/src/CloudViewer.cpp: In member function ‘bool rtabmap::CloudViewer::addCloudTextureMesh(const string&, const Ptr&, const cv::Mat&, const rtabmap::Transform&)’:
/home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:788:73: error: invalid use of incomplete type ‘class vtkTexture’
  788 |    _visualizer->getCloudActorMap()->find(id)->second.actor->GetTexture()->SetInterpolate(1);
      |                                                                         ^~
In file included from /usr/include/vtk/vtkLODActor.h:60,
                 from /usr/include/pcl-1.9/pcl/visualization/common/actor_map.h:45,
                 from /usr/include/pcl-1.9/pcl/visualization/interactor_style.h:44,
                 from /home/parker/build/rtabmap/guilib/src/../include/rtabmap/gui/CloudViewerInteractorStyle.h:15,
                 from /home/parker/build/rtabmap/guilib/src/../include/rtabmap/gui/CloudViewer.h:35,
                 from /home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:28:
/usr/include/vtk/vtkActor.h:41:7: note: forward declaration of ‘class vtkTexture’
   41 | class vtkTexture;
      |       ^~~~~~~~~~
/home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:789:73: error: invalid use of incomplete type ‘class vtkTexture’
  789 |    _visualizer->getCloudActorMap()->find(id)->second.actor->GetTexture()->SetBlendingMode(vtkTexture::VTK_TEXTURE_BLENDING_MODE_REPLACE);
      |                                                                         ^~
In file included from /usr/include/vtk/vtkLODActor.h:60,
                 from /usr/include/pcl-1.9/pcl/visualization/common/actor_map.h:45,
                 from /usr/include/pcl-1.9/pcl/visualization/interactor_style.h:44,
                 from /home/parker/build/rtabmap/guilib/src/../include/rtabmap/gui/CloudViewerInteractorStyle.h:15,
                 from /home/parker/build/rtabmap/guilib/src/../include/rtabmap/gui/CloudViewer.h:35,
                 from /home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:28:
/usr/include/vtk/vtkActor.h:41:7: note: forward declaration of ‘class vtkTexture’
   41 | class vtkTexture;
      |       ^~~~~~~~~~
/home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:789:103: error: incomplete type ‘vtkTexture’ used in nested name specifier
  789 |    _visualizer->getCloudActorMap()->find(id)->second.actor->GetTexture()->SetBlendingMode(vtkTexture::VTK_TEXTURE_BLENDING_MODE_REPLACE);
      |                                                                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/parker/build/rtabmap/guilib/src/CloudViewer.cpp: In member function ‘bool rtabmap::CloudViewer::addTextureMesh(const pcl::TextureMesh&, const cv::Mat&, const string&, int)’:
/home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:1142:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
 1142 |   if (!tex_manager)
      |   ^~
/home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:1145:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
 1145 |     vtkSmartPointer<vtkTexture> texture = vtkSmartPointer<vtkTexture>::New ();
      |     ^~~~~~~~~~~~~~~
/home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:1150:9: error: invalid use of incomplete type ‘class vtkTexture’
 1150 |  texture->SetInputConnection(cvImageToVtk->GetOutputPort());
      |         ^~
In file included from /usr/include/vtk/vtkLODActor.h:60,
                 from /usr/include/pcl-1.9/pcl/visualization/common/actor_map.h:45,
                 from /usr/include/pcl-1.9/pcl/visualization/interactor_style.h:44,
                 from /home/parker/build/rtabmap/guilib/src/../include/rtabmap/gui/CloudViewerInteractorStyle.h:15,
                 from /home/parker/build/rtabmap/guilib/src/../include/rtabmap/gui/CloudViewer.h:35,
                 from /home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:28:
/usr/include/vtk/vtkActor.h:41:7: note: forward declaration of ‘class vtkTexture’
   41 | class vtkTexture;
      |       ^~~~~~~~~~
In file included from /usr/include/pcl-1.9/pcl/visualization/point_cloud_geometry_handlers.h:48,
                 from /usr/include/pcl-1.9/pcl/visualization/point_cloud_handlers.h:41,
                 from /usr/include/pcl-1.9/pcl/visualization/common/actor_map.h:40,
                 from /usr/include/pcl-1.9/pcl/visualization/interactor_style.h:44,
                 from /home/parker/build/rtabmap/guilib/src/../include/rtabmap/gui/CloudViewerInteractorStyle.h:15,
                 from /home/parker/build/rtabmap/guilib/src/../include/rtabmap/gui/CloudViewer.h:35,
                 from /home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:28:
/usr/include/vtk/vtkSmartPointer.h: In instantiation of ‘static vtkSmartPointer<T> vtkSmartPointer<T>::New() [with T = vtkTexture]’:
/home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:1145:72:   required from here
/usr/include/vtk/vtkSmartPointer.h:155:37: error: incomplete type ‘vtkTexture’ used in nested name specifier
  155 |     return vtkSmartPointer<T>(T::New(), NoReference());
      |                               ~~~~~~^~
/usr/include/vtk/vtkSmartPointer.h: In instantiation of ‘T* vtkSmartPointer<T>::operator->() const [with T = vtkTexture]’:
/home/parker/build/rtabmap/guilib/src/CloudViewer.cpp:1150:9:   required from here
/usr/include/vtk/vtkSmartPointer.h:130:12: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkTexture*’
  130 |     return static_cast<T*>(this->Object);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 43 lines cut ...]
                 from /usr/include/boost/random/linear_congruential.hpp:30,
                 from /usr/include/boost/random/additive_combine.hpp:27,
                 from /usr/include/boost/random.hpp:36,
                 from /usr/include/pcl-1.9/pcl/filters/boost.h:49,
                 from /usr/include/pcl-1.9/pcl/filters/filter.h:46,
                 from /home/parker/build/rtabmap/examples/Export/main.cpp:37:
/usr/include/boost/pending/integer_log2.hpp:7:1: note: #pragma message: This header is deprecated. Use <boost/integer/integer_log2.hpp> instead.
    7 | BOOST_HEADER_DEPRECATED("<boost/integer/integer_log2.hpp>");
      | ^~~~~~~~~~~~~~~~~~~~~~~
[ 69%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/OdometryViewer.cpp.o
make[2]: *** [guilib/src/CMakeFiles/rtabmap_gui.dir/build.make:425: guilib/src/CMakeFiles/rtabmap_gui.dir/CloudViewer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 70%] Linking CXX executable ../../../bin/rtabmap-export
In file included from /usr/include/boost/detail/endian.hpp:9,
                 from /usr/include/pcl-1.9/pcl/PCLPointCloud2.h:11,
                 from /usr/include/pcl-1.9/pcl/pcl_base.h:55,
                 from /home/parker/build/rtabmap/corelib/include/rtabmap/core/util3d_transforms.h:35,
                 from /home/parker/build/rtabmap/guilib/src/OdometryViewer.cpp:30:
/usr/include/boost/predef/detail/endian_compat.h:11:161: note: #pragma message: The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead
   11 | #pragma message("The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead")
      |                                                                                                                                                                 ^
[ 70%] Built target export
make[1]: *** [CMakeFiles/Makefile2:331: guilib/src/CMakeFiles/rtabmap_gui.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

parkerlreed, 2019-07-27:
Cmake output

```
-- The C compiler identification is GNU 9.1.0
-- The CXX compiler identification is GNU 9.1.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:64 (cmake_policy):
  The OLD behavior for policy CMP0020 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:67 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Checking for module 'eigen3'
--   Found eigen3, version 3.3.7
-- Found Eigen: /usr/include/eigen3  
-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- Found libusb-1.0: /usr/include  
-- Checking for module 'flann'
--   Found flann, version 1.9.1
-- FLANN found (include: , lib: flann;flann_cpp)
-- FLANN found (include: , lib: flann;flann_cpp)
-- Found Qhull: optimized;/usr/lib/libqhull_p.so;debug;/usr/lib/libqhull_p.so  
-- QHULL found (include: /usr/include, lib: optimized;/usr/lib/libqhull_p.so;debug;/usr/lib/libqhull_p.so)
-- Found PCL_COMMON: /usr/lib/libpcl_common.so  
-- Found PCL_OCTREE: /usr/lib/libpcl_octree.so  
-- Found PCL_IO: /usr/lib/libpcl_io.so  
-- Found PCL_KDTREE: /usr/lib/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/lib/libpcl_search.so  
-- Found PCL_SURFACE: /usr/lib/libpcl_surface.so  
-- Found PCL_FILTERS: /usr/lib/libpcl_filters.so  
-- Found PCL_2D: /usr/include/pcl-1.9  
-- Found PCL_FEATURES: /usr/lib/libpcl_features.so  
-- Found PCL_REGISTRATION: /usr/lib/libpcl_registration.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/lib/libpcl_sample_consensus.so  
-- Found PCL_GEOMETRY: /usr/include/pcl-1.9  
-- Found PCL_ML: /usr/lib/libpcl_ml.so  
-- Found PCL_SEGMENTATION: /usr/lib/libpcl_segmentation.so  
-- Found PCL_VISUALIZATION: /usr/lib/libpcl_visualization.so  
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/libsqlite3.so
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found OpenMP
-- Found OpenCV: /usr/include/opencv4
-- Found PCL: /usr/include/pcl-1.9;/usr/include/eigen3;/usr/include;/usr/include/vtk;/usr/include/freetype2;/usr/include/double-conversion;/usr/include/libxml2;/usr/include/python3.7m
-- Found ZLIB: /usr/include
[... 145 lines cut ...]
--  *With g2o                  = NO (g2o not found)
--  *With GTSAM                = NO (GTSAM not found)
--   With VERTIGO              = NO (GTSAM or g2o required)
--   With cvsba                = NO (cvsba not found)
--  *With libpointmatcher      = NO (libpointmatcher not found)
--   With loam_velodyne        = NO (loam_velodyne not found)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = NO (librealsense2 not found)
--   With OCTOMAP              = NO (octomap not found)
--   With CPUTSDF              = NO (CPUTSDF not found)
--   With OpenChisel           = NO (open_chisel not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With okvis                = NO (okvis not found)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (VINS-Fusion not found)
--   With ORB_SLAM2            = NO (ORB_SLAM2 not found, make sure environment variable ORB_SLAM2_ROOT_DIR is set)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/parker/build/rtabmap/build
```

discord-intech, 2019-07-28:
This is due to a modification in the VTK lib. PCL is not properly including the vtkTexture header and instead relies on another header to have imported it beforehand. Since this is a PCL problem, this is not the place to fix it, but you can mitigate it by adding the vtkTexture import in CloudViewer.h, like so :

```diff
diff --git a/guilib/include/rtabmap/gui/CloudViewer.h b/guilib/include/rtabmap/gui/CloudViewer.h
index 526f4bc2..56701b59 100644
--- a/guilib/include/rtabmap/gui/CloudViewer.h
+++ b/guilib/include/rtabmap/gui/CloudViewer.h
@@ -28,6 +28,8 @@ SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 #ifndef RTABMAP_CLOUDVIEWER_H_
 #define RTABMAP_CLOUDVIEWER_H_
 
+#include <vtkTexture.h>
 #include "rtabmap/gui/RtabmapGuiExp.h" // DLL export/import defines
```

matlabbe, 2019-07-29:
Thx @discord-intech for the hint. I've made the change in the commit above. I cannot test it on my testing computer (using PCL binaries, not latest version from source), let me know if this works.

discord-intech, 2019-07-29:
I tested it on my Archlinux machine, it should work for @parkerlreed too.

parkerlreed, 2019-07-29:
Thanks! Will test compiling now

parkerlreed, 2019-07-29:
:+1: 

[image]

closed 2019-07-29 by parkerlreed

parkerlreed, 2019-07-29:
@discord-intech Quick question before I submit another issue: If you view point cloud and then exit the window, does it also crash for you? It's weird because the main window pops up on top of the point cloud view. You can use it as much as you want, but as soon as you exit

```
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
Generic Warning: In /build/vtk/src/VTK-8.2.0/GUISupport/Qt/QVTKWidget.cxx, line 83
QVTKWidget was deprecated for VTK 8.1 and will be removed in a future version.

Generic Warning: In /build/vtk/src/VTK-8.2.0/GUISupport/Qt/QVTKPaintEngine.cxx, line 25
QVTKPaintEngine was deprecated for VTK 8.1 and will be removed in a future version.

X Error of failed request:  BadWindow (invalid Window parameter)
  Major opcode of failed request:  3 (X_GetWindowAttributes)
  Resource id in failed request:  0x2006c7
  Serial number of failed request:  22319
  Current serial number in output stream:  22320
```

https://www.youtube.com/watch?v=7-KzYUcujZo

parkerlreed, 2019-07-29:
Oh I realized it may be slightly related to the CloudViewer that was fixed hmm

discord-intech, 2019-07-31:
I cannot reproduce the same crash with my system. For info, I use i3wm, so it might be related to your DE and VTK having a spat and forcing Xorg to kill rtabmapviz. I do not think this a problem related to rtabmap, but I could be wrong.

parkerlreed, 2019-07-31:
Ok Ill take a stab at it later. Now that you mention it, last time I used this a lot natively was on i3wm heh.

parkerlreed, 2019-08-01:
@discord-intech Doesn't fully crash on i3 but does lock up completely.

https://www.youtube.com/watch?v=V7-jl5ogS8Q

vtk 8.2.0-5

matlabbe, 2019-08-01:
Did you try with vtk 6.3.0 (default version on 18.04)?

parkerlreed, 2019-08-07:
Currently building vtk6. Will try an rtabmap compile once that finishes.

parkerlreed, 2019-08-07:
vtk6 seems to not have the vtkm folder under cmake

[image]

I've compared the build script for vtk and vtk6 and don't notice any obvious things that would include or not include vtkm

parkerlreed, 2019-08-07:
I tried with both set to the /opt/vtk6/cmake and it compiled but crashes out on start

```
[parker@stealth build]$ rtabmap
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Generic Warning: In /home/parker/build/vtk6/src/VTK-6.3.0/Rendering/OpenGL/vtkOpenGLPolyDataMapper.cxx, line 68
vtkOpenGLPolyDataMapper::vtkOpenGLPolyDataMapper was deprecated for VTK 6.2 and will be removed in a future version.

KCrash: crashing... crashRecursionCounter = 2
KCrash: Application Name = rtabmap path = /usr/bin pid = 3406
KCrash: Arguments: /usr/bin/rtabmap 
KCrash: Attempting to start /usr/lib/drkonqi from kdeinit
sock_file=/run/user/1000/kdeinit5__0

[1]+  Stopped                 rtabmap
```

matlabbe, 2019-08-13:
You may look at this post: https://github.com/introlab/rtabmap/issues/427#issuecomment-516914817

On Arch Linux, Qt5 seems built with OpenGL ES instead of OpenGL, causing issues with vtk binaries that seem built for OpenGL, not OpenGL ES. rtabmap binaries then crash if the UI is used. The rtabmap node would be okay, so the workaround is to use RVIZ instead of rtabmapviz for visualization.

parkerlreed, 2019-08-14:
Thanks, will take a look. That vtk6 was compiled from source so I would hope it would match the system libs/OpenGL support. Arch has tried pretty hard to move away from everything QT4.