introlab/rtabmap #426: Make error; looks like issue with libnss3 version?

eric-schleicher, 2019-07-27:
Looks like an issue with libnss3 on (pretty vanilla) Ubuntu 18.04LTS; mostly setup for ROS melodic

[image]

In researching, it appears that 16.04 uses version `2:3.28.4-0ubuntu0.16.04.6` and 18.04 uses `2:3.35-2ubuntu2.3`

This [page](https://abi-laboratory.pro/index.php?view=timeline&l=nss) however lists interface changes, but that nothing has been removed since listed previous versions. 

not sure how to chase down the issue.

matlabbe, 2019-07-29:
The [rtabmap docker image for bionic](https://cloud.docker.com/u/introlab3it/repository/docker/introlab3it/rtabmap) builds correctly, see the log at the end of this post.

I am currently building a fresh docker image on my computer to see if the cache of the docker buildfarm should be updated. I also modified Travis config to build not jsut for precise, but for xenial and bionic as well. See this [job](https://travis-ci.org/introlab/rtabmap/jobs/564764379) for bionic (currently building).

Docker:
```
Explore
Repositories
Organizations
Repositories
introlab3it / rtabmap
Builds
Using 0 of 0 private repositories. Get more
SUCCESS

Build in 'master:/docker/bionic' (a90356f0)
bionic
introlab/rtabmap
CREATED
2 days ago
DURATION
93 min
USER
introlab3it
LOCATION
unknown
IP
140.82.115.70
Building in Docker Cloud's infrastructure...
Cloning into '.'...
Warning: Permanently added the RSA host key for IP address '192.30.253.112' to the list of known hosts.
Reset branch 'master'
Your branch is up-to-date with 'origin/master'.
Pulling cache layers for index.docker.io/introlab3it/rtabmap:bionic...
Done!
Executing build hook...
Sending build context to Docker daemon 6.656kB
Step 1/15 : FROM ubuntu:18.04
18.04: Pulling from library/ubuntu
7413c47ba209: Already exists
0fe7e7cbb2e8: Already exists
1d425c982345: Already exists
344da5c95cec: Already exists
Digest: sha256:c303f19cfe9ee92badbbbd7567bc1ca47789f79303ddcef56f77687d4744cd7a
Status: Downloaded newer image for ubuntu:18.04
---> 3556258649b2
Step 2/15 : RUN apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y libsqlite3-dev libpcl-dev git cmake libopencv-dev libproj-dev libqt5svg5-dev libfreenect-dev libopenni2-dev ffmpeg software-properties-common
---> Using cache
---> c02887bb8b52
Step 3/15 : WORKDIR /root/
---> Using cache
---> 8cb9efca71bc
Step 4/15 : RUN git clone https://bitbucket.org/gtborg/gtsam.git
---> Using cache
---> 9e6b4d57428b
Step 5/15 : RUN cd gtsam && git checkout 4.0.0-alpha2 && mkdir build && cd build && cmake -DMETIS_SHARED=ON -DGTSAM_BUILD_STATIC_LIBRARY=OFF -DGTSAM_USE_SYSTEM_EIGEN=ON -DGTSAM_BUILD_TESTS=OFF -DGTSAM_BUILD_EXAMPLES_ALWAYS=OFF -DCMAKE_BUILD_TYPE=Release .. && make -j$(nproc) && make install && cd && rm -r gtsam
---> Using cache
---> 54c75b58b136
Step 6/15 : RUN git clone https://github.com/RainerKuemmerle/g2o.git
---> Using cache
---> 3b56e7c611a4
Step 7/15 : RUN cd g2o && git checkout 20170730_git && mkdir build && cd build && cmake -DBUILD_LGPL_SHARED_LIBS=ON -DG2O_BUILD_APPS=OFF -DBUILD_WITH_MARCH_NATIVE=OFF -DG2O_BUILD_EXAMPLES=OFF -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=Release .. && make -j$(nproc) && make install && cd && rm -r g2o
---> Using cache
---> ee2f3a6ac3e6
Step 8/15 : RUN git clone https://github.com/ethz-asl/libnabo.git
---> Using cache
---> 1865e9d2a84e
Step 9/15 : RUN cd libnabo && git checkout 7e378f6765393462357b8b74d8dc8c5554542ae6 && mkdir build && cd build && cmake -DCMAKE_BUILD_TYPE=Release .. && make -j$(nproc) && make install && cd && rm -r libnabo
---> Using cache
---> f29a43a095f2
[... 1112 lines cut ...]
cde2cd537412: Layer already exists
b079b3fa8d1b: Layer already exists
a31dbd3063d7: Layer already exists
12915dd200c2: Layer already exists
c56e09e1bd18: Layer already exists
543791078bdb: Layer already exists
18.04: digest: sha256:6c63cb027ba9435899cb6dc36ebd62fb1b32d91de1f56f37fe0e828bd955de2e size: 3477
Build finished
EXPLORE
Docker Editions
Containers
Plugins
ACCOUNT
My Content
Billing
PUBLISH
Publisher Center
RESOURCES
Engineering Blog
SUPPORT
Feedback
Documentation
Copyright © 2019 Docker, Inc. All rights reserved.
LegalDocker
```

eric-schleicher, 2019-09-29:
determines this was cause by loading mainline kernel.

closed 2019-09-29 by eric-schleicher

roncapat, 2020-03-23:
@eric-schleicher which version?