introlab/rtabmap #427: 3D map is not loading

aguilaair, 2019-07-29:
Hardware: Jetson Nano
OS: Ubuntu 18.04 w/ ROS Melodic and Jetpack

I cannot view the 3d map generated.  The 3d map tab is completely empty. I've tried to install from source but the problem is still there. I believe this issue might be related to ther map not being saved on ROS.
Could you guys shed some light?

Thanks!

aguilaair, 2019-07-29:
[image]
 Here is how it looks

matlabbe, 2019-07-30:
Very strange, even with nothing running, the 3D Map view should be black with an OpenGL rendering view. The OpenGL renderer is not showing, not sure if it is a VTK/Qt problem. I have a nano on my desk I didn't have time to test with yet, I'll try it soon.

aguilaair, 2019-07-30:
Ok, thank you for the info. Keep me updated!

matlabbe, 2019-07-31:
I've found a way to make it working. We can replace the vtk libraries that are related to Qt5 by the ones built with from source with Qt4. Here the full walk-through:

* Install/uninstall `ros-melodic-rtabmap-ros` to get all dependencies installed:
```bash
$ sudo apt install ros-melodic-rtabmap-ros
$ sudo apt remove ros-melodic-rtabmap
```

* Build VTK 6.3.0 from source with Qt4 (not Qt5!) **OR** for without recompiling, download this archive [vtk6.3.0-arm64-qt4-libs.zip](https://github.com/introlab/rtabmap/files/3452792/vtk6.3.0-arm64-qt4-libs.zip) with already compiled  libraries:
```bash
$ cd ~
$ cd git clone https://github.com/Kitware/VTK.git
$ cd VTK
$ git checkout v6.3.0
$ mkdir build
$ cd build
$ cmake -DVTK_Group_Qt=ON -DVTK_QT_VERSION=4 -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Release ..
```

* Remove all Qt5 related vtk libraries installed in `/usr/lib/aarch64-linux-gnu`:
```bash
$ sudo rm /usr/lib/aarch64-linux-gnu/libvtkGUISupportQt*
$ sudo rm /usr/lib/aarch64-linux-gnu/libvtkRenderingQt*
$ sudo rm /usr/lib/aarch64-linux-gnu/libvtkViewsQt*
$ sudo rm /usr/lib/cmake/vtk-6.3/Modules/vtkGUISupportQtWebkit.cmake
```

Copy the newly compiled ones with Qt4 support:
```bash
$ sudo cp ~/VTK/build/lib/libvtkGUISupportQt* /usr/lib/aarch64-linux-gnu/.
$ sudo cp ~/VTK/build/lib/libvtkRenderingQt* /usr/lib/aarch64-linux-gnu/.
$ sudo cp ~/VTK/build/lib/libvtkGUISupportQtSQL* /usr/lib/aarch64-linux-gnu/.
$ sudo cp ~/VTK/build/lib/libvtkViewsQt* /usr/lib/aarch64-linux-gnu/.
```

Create symbolic links to match binaries version:
```bash
$ cd /usr/lib/aarch64-linux-gnu
$ sudo ln -s  libvtkGUISupportQtOpenGL-6.3.so.1 libvtkGUISupportQtOpenGL-6.3.so.6.3.0
$ sudo ln -s  libvtkGUISupportQt-6.3.so.1 libvtkGUISupportQt-6.3.so.6.3.0
$ sudo ln -s  libvtkRenderingQt-6.3.so.1 libvtkRenderingQt-6.3.so.6.3.0
$ sudo ln -s  libvtkGUISupportQtSQL-6.3.so.1 libvtkGUISupportQtSQL-6.3.so.6.3.0
$ sudo ln -s  libvtkViewsQt-6.3.so.1 libvtkViewsQt-6.3.so.6.3.0

$ sudo ln -s libvtkInteractionStyle-6.3.so.6.3.0 libvtkInteractionStyle-6.3.so.1
$ sudo ln -s libvtkRenderingOpenGL-6.3.so.6.3.0 libvtkRenderingOpenGL-6.3.so.1
$ sudo ln -s libvtkRenderingCore-6.3.so.6.3.0 libvtkRenderingCore-6.3.so.1
$ sudo ln -s libvtkFiltersExtraction-6.3.so.6.3.0 libvtkFiltersExtraction-6.3.so.1
$ sudo ln -s libvtkCommonDataModel-6.3.so.6.3.0 libvtkCommonDataModel-6.3.so.1
$ sudo ln -s libvtkCommonCore-6.3.so.6.3.0 libvtkCommonCore-6.3.so.1
```
* Install optional rtabmap depdencies (like GTSAM, realsense, zed, etc.)

* Build RTAB-Map with same Qt version (4) used by VTK (use latest rtabmap from source to have `RTABMAP_QT_VERSION` option):
```bash
$ cd ~
$ git clone https://github.com/introlab/rtabmap.git
$ cd rtabmap/build
$ cmake -DRTABMAP_QT_VERSION=4 ..
$ make
```

NOTES: 
 * rtabmap_ros not yet tested (there could be a problem with RVIZ plugins and rtabmapviz using different Qt versions). If you are going to use ROS, avoid all those recompilations and use rtabmap binaries directly and just don't use rtabmapviz but RVIZ instead!
 * I didn't retest the walk-through above on a fresh nano (it is quite long to test! >6 hours)

Nap2501, 2019-09-21:
Hello Matthieu or anyone who could help me,
I also use a jetson nano, and i use libfreenect2 (neither g2o nor GTSAM), I've flashed the last linux release by nvidia : version JP 4.2.2 | 2019/08/26
i had what i believe to be the same issue : i have a freezed image on the 3D display window, and rtabmap crashes when i try to use the start button.
i'm a newbie at linux (it's been only 6 month i'm using it).

I can't build VTK from source (i have errors when i do cmake) , so i've had downloaded your files, and made up a ```~/VTK/build/lib/``` to drop the files into (with a previous `git clone VTK`).
<details>
  <summary>VTK cmake Errors</summary>
   
  ```shell
~/VTK/build$ cmake -DVTK_Group_Qt=ON -DVTK_QT_VERSION=4 -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Release ..
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAVE_GCC_ERROR_RETURN_TYPE
-- Performing Test HAVE_GCC_ERROR_RETURN_TYPE - Success
-- Performing Test HAVE_GCC_VISIBILITY
-- Performing Test HAVE_GCC_VISIBILITY - Success
CMake Error at CMake/vtkCompilerExtras.cmake:47 (if):
  if given arguments:

    "cc: error: ARGS: No such file or directory

  cc (Ubuntu/Linaro 7.4.0-1ubuntu1~18.04.1) 7.4.0

  Copyright (C) 2017 Free Software Foundation, Inc.

  This is free software" " see the source for copying conditions.  There is
  NO

  warranty" " not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR
  PURPOSE.

  

  " "VERSION_GREATER" "4.2.0" "AND" "BUILD_SHARED_LIBS" "AND"
  "HAVE_GCC_VISIBILITY" "AND" "VTK_USE_GCC_VISIBILITY" "AND" "NOT" "MINGW"
  "AND" "NOT" "CYGWIN"

  Unknown arguments specified
Call Stack (most recent call first):
  CMakeLists.txt:283 (include)

-- Configuring incomplete, errors occurred!
See also "/home/supergitan/VTK/build/CMakeFiles/CMakeOutput.log".
  ```
  
</details>
all the procedure went good, except on part 6:

> Remove all references to Qt5 stuff in /usr/lib/cmake/vtk-6.3/VTKTargets.cmake and /usr/lib/cmake/vtk-6.3/VTKTargets-none.cmake.

i removed every reference to Qt5 using gedit, it seems to have issue with attached metadata.
i'm unsure my issue is related to the way i had edited this file, but maybe i'm wrong.

after building and installing Rtabmap (which was fine, i did not have more compilation errors than usual) , i can't run rtabmap by running
`$ rtabmap`
i get the error `bash: rtabmap: command not found`
i've tried `echo $PATH `, multiple results but no `:/home/rtabmap/bin`
so i tried `export PATH=$PATH:/home/rtabmap/bin`
i know this issue is probably just a PATH issue, but i've tried all day long and i still have no solution.
i will retry with the before last ubuntu release from nvidia

matlabbe, 2019-09-22:
Just to run rtabmap from `rtabmap/bin` directory, you have to do `./rtabmap`, not `rtabmap`:
```bash
cd rtabmap/bin
./rtabmap
```

Nap2501, 2019-09-24:
I have done the procedure again on a freshly flashed Sd, to be sure nothing is interfering with the result.
At the end, I tried to do
```shell
cd rtabmap/bin
./rtabmap
```

i have the following error: `bash: ./rtabmap: No such file or directory`
it seems that `./rtabmap` is not into the bin directory, i tried `ls -a` , it returned:
```shell
.                             rtabmap-console
..                            rtabmap-detectMoreLoopClosures
data                          rtabmap-euroc_dataset
.gitignore                    rtabmap-extractObject
librtabmap_core.so            rtabmap-imagesJoiner
librtabmap_core.so.0.19       rtabmap-kitti_dataset
librtabmap_core.so.0.19.5     rtabmap-recovery
librtabmap_utilite.so         rtabmap-reprocess
librtabmap_utilite.so.0.19    rtabmap-res_tool
librtabmap_utilite.so.0.19.5  rtabmap-res_tool-0.3.0
rtabmap-bow_mapping           rtabmap-rgbd_dataset
rtabmap-camera                rtabmap-stereoEval
```

i have archived the cmake make and install text from shell, in case it could be useful.

I guess i have missed something on Part6:

> Remove all references to Qt5 stuff in /usr/lib/cmake/vtk-6.3/VTKTargets.cmake and /usr/lib/cmake/vtk-6.3/VTKTargets-none.cmake.

plus, when installing/desinstalling `ros-melodic-rtabmap-ros` , i have initialized rosdep, could be a source of issues?

<details>
  <summary>Cmake</summary>
    
  ```shell
supergitan@supergitan-cuivre:~/rtabmap/build$ cmake -DRTABMAP_QT_VERSION=4 -Dfreenect2_DIR=$HOME/freenect2/lib/cmake/freenect2 ..
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:64 (cmake_policy):
  The OLD behavior for policy CMP0020 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Checking for module 'eigen3'
--   Found eigen3, version 3.3.4
-- Found eigen: /usr/include/eigen3  
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- The imported target "vtkGUISupportQtWebkit" references the file
   "/usr/lib/aarch64-linux-gnu/libvtkGUISupportQtWebkit-6.3.so.6.3.0"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
[... 677 lines cut ...]
-- Installing: /usr/local/include/rtabmap-0.19/rtabmap/core/optimizer/OptimizerTORO.h
-- Installing: /usr/local/include/rtabmap-0.19/rtabmap/core/optimizer/OptimizerGTSAM.h
-- Installing: /usr/local/include/rtabmap-0.19/rtabmap/core/optimizer/OptimizerCeres.h
-- Installing: /usr/local/include/rtabmap-0.19/rtabmap/core/optimizer/OptimizerCVSBA.h
-- Installing: /usr/local/bin/rtabmap-console
-- Set runtime path of "/usr/local/bin/rtabmap-console" to ""
-- Installing: /usr/local/bin/rtabmap-extractObject
-- Set runtime path of "/usr/local/bin/rtabmap-extractObject" to ""
-- Installing: /usr/local/bin/rtabmap-camera
-- Set runtime path of "/usr/local/bin/rtabmap-camera" to ""
-- Installing: /usr/local/bin/rtabmap-kitti_dataset
-- Set runtime path of "/usr/local/bin/rtabmap-kitti_dataset" to ""
-- Installing: /usr/local/bin/rtabmap-rgbd_dataset
-- Set runtime path of "/usr/local/bin/rtabmap-rgbd_dataset" to ""
-- Installing: /usr/local/bin/rtabmap-euroc_dataset
-- Set runtime path of "/usr/local/bin/rtabmap-euroc_dataset" to ""
-- Installing: /usr/local/bin/rtabmap-recovery
-- Set runtime path of "/usr/local/bin/rtabmap-recovery" to ""
-- Installing: /usr/local/bin/rtabmap-reprocess
-- Set runtime path of "/usr/local/bin/rtabmap-reprocess" to ""
-- Installing: /usr/local/bin/rtabmap-detectMoreLoopClosures
-- Set runtime path of "/usr/local/bin/rtabmap-detectMoreLoopClosures" to ""
  ```
  
</details>

matlabbe, 2019-09-24:
cmake didn't find Qt:
```
With Qt                   = NO (Qt not found)
```
so `rtabmap` app is not built (well all gui stuff). Initializing rosdep would not be an issue. Is qt4 installed?

aguilaair, 2019-10-21:
Anyway to fix it in a way I could still use Rviz?

Thanks in advance!

matlabbe, 2019-10-22:
The issue was related to rtabmapviz. Rviz should work directly without any recompilation on rtabmap side.

zzxxtt, 2019-12-02:
All jetson with ubuntu 18.04 have the same problem, including TX2, AGX. Are there be any better way to solve it now?

axeotron, 2020-02-16:
The problem still exists on Jetson TX2 running JetPack 4.3 with ZED SDK 3.0. I can't get past making VTK. Reading the error "cmake -DVTK_Group_Qt=ON -DVTK_QT_VERSION=4 -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Release .."

Googling this shows that it is caused by C compiler version which 7.4.0 in my case and one solution is to downgrade it. This is another case where it highlights that this is caused by the version GCC compiler.
https://github.com/ufz/ogs/issues/1274#issuecomment-227650988

Other things I tried and didn't work: 
Commenting the lines as shown here https://github.com/mxe/mxe/issues/1536#issuecomment-509168351

I am not sure if this helps, I would love to contribute in solving this as I have used rtabmap for some time, however, on Ubuntu 16 based JetPacks.

CluelessIT, 2020-03-10:
Hi, @matlabbe  I also encountered the same issue and when i followed your steps, I got this error:

rtabmap: error while loading shared libraries: libvtkGUISupportQt-6.3.so.6.3: cannot open shared object file: No such file or directory

any idea what is happening?

matlabbe, 2020-04-02:
# Update: Jetpack 4.3

## Standalone
The instructions of this [post](https://github.com/introlab/rtabmap/issues/427#issuecomment-516914817) above are still working. Tested with ZED SDK 3.1.2 driver and a zedm camera.

## ROS
The ros binaries are still working out of the box with the original issue (rtabmapviz blank 3D View). Recommendation is still to use rviz for visualization.

If rtabmap has to be rebuilt from source. The new jetpack added more incompatibilities. By default `libopencv-dev` version from jetpack image is now 4.1.1. This is causing problems when building `rtabmap_ros` with `cv_bridge`, which is still linked to standard 3.2 version from bionic. `cv_bridge` has to be also rebuilt from source. The master branch of [cv_bridge](https://github.com/ros-perception/vision_opencv) repo is not compatible with opencv 4, we should use this [pull request](https://github.com/ros-perception/vision_opencv/issues/291). In summary:

```bash
sudo apt remove ros-melodic-cv-bridge
cd ~/catkin_ws/src
git clone https://github.com/BrutusTT/vision_opencv.git
git clone https://github.com/ros-perception/image_transport_plugins
```
Build rtabmap library with `-DWITH_QT=OFF` if you are using RVIZ. If rtabmapviz needs to be used, you would have to follow the [instructions](https://github.com/introlab/rtabmap/issues/427#issuecomment-516914817) of the standalone version, including the second note at the bottom of the same post.

When running [robot mapping demo bag](http://wiki.ros.org/rtabmap_ros#Robot_mapping), if you get error "`Unsupported image format: 32FC1; compressedDepth`", change those [lines](https://github.com/ros-perception/image_transport_plugins/blob/0162f74e389d139b571d395c0bbb3c8b9d013a94/compressed_depth_image_transport/src/codec.cpp#L81-L84) to:
```cpp
    if (format.find("compressedDepth rvl") != std::string::npos) {
      compression_format = "rvl";
    } else if (format.find("compressedDepth") != std::string::npos) {
      compression_format = "png"; // fallback default png for old rosbags
```

matlabbe, 2020-04-02:
@axeotron Are you using VTK v6.3.0? Note that in my update just above, I tested with the [precompiled vtk libraries with Qt4](https://github.com/introlab/rtabmap/files/3457605/vtk6.3.0-arm64-qt4-libs-cmake.zip) of this [post](https://github.com/introlab/rtabmap/issues/427#issuecomment-516914817), I didn't try to rebuild VTK on Jetpack 3.4.

@CluelessIT make sure you did step 7 in that [post](https://github.com/introlab/rtabmap/issues/427#issuecomment-516914817) above.

tkircher, 2020-04-02:
@matlabbe I also made a post [here](https://github.com/ros-perception/vision_opencv/issues/272) about my own experiences getting ROS working under Jetpack 4.3.

a-xtron, 2020-04-03:
@matlabbe Indeed I was trying to rebuild it. Not sure if worth mentioning, I have changed my setup from Jetson TX2 to AGX Xavier running Jetpack 4.3 and ZED SDK 3.1.2  After trying without recompiling vtk and using the compiled libraries attached in the post above I managed to reach the building process. However, make fails with the following error:
`[ 44%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OctoMap.cpp.o
make[2]: *** No rule to make target 'vtkGUISupportQtOpenGL-NOTFOUND', needed by '../bin/librtabmap_core.so.0.19.6'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 44%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/imufilter/MadgwickFilter.cpp.o
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
nvidia@axtron-xavier:~/rtabmap/build$ 
`
cmake output:
`The system is: Linux - 4.9.140-tegra - aarch64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/nvidia/rtabmap/build/CMakeFiles/3.10.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/nvidia/rtabmap/build/CMakeFiles/3.10.2/CompilerIdCXX/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/nvidia/rtabmap/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_8248b/fast"
/usr/bin/make -f CMakeFiles/cmTC_8248b.dir/build.make CMakeFiles/cmTC_8248b.dir/build
make[1]: Entering directory '/home/nvidia/rtabmap/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_8248b.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_8248b.dir/testCCompiler.c.o   -c /home/nvidia/rtabmap/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_8248b
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_8248b.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_8248b.dir/testCCompiler.c.o  -o cmTC_8248b 
make[1]: Leaving directory '/home/nvidia/rtabmap/build/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/nvidia/rtabmap/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_f1142/fast"
/usr/bin/make -f CMakeFiles/cmTC_f1142.dir/build.make CMakeFiles/cmTC_f1142.dir/build
make[1]: Entering directory '/home/nvidia/rtabmap/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_f1142.dir/CMakeCCompilerABI.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_f1142.dir/CMakeCCompilerABI.c.o   -c /usr/share/cmake-3.10/Modules/CMakeCCompilerABI.c
Linking C executable cmTC_f1142
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_f1142.dir/link.txt --verbose=1
/usr/bin/cc     -v -rdynamic CMakeFiles/cmTC_f1142.dir/CMakeCCompilerABI.c.o  -o cmTC_f1142 
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/aarch64-linux-gnu/7/lto-wrapper
Target: aarch64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu/Linaro 7.5.0-3ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=aarch64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libquadmath --disable-libquadmath-support --enable-plugin --enable-default-pie --with-system-zlib --enable-multiarch --enable-fix-cortex-a53-843419 --disable-werror --enable-checking=release --build=aarch64-linux-gnu --host=aarch64-linux-gnu --target=aarch64-linux-gnu
Thread model: posix
gcc version 7.5.0 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 
COMPILER_PATH=/usr/lib/gcc/aarch64-linux-gnu/7/:/usr/lib/gcc/aarch64-linux-gnu/7/:/usr/lib/gcc/aarch64-linux-gnu/:/usr/lib/gcc/aarch64-linux-gnu/7/:/usr/lib/gcc/aarch64-linux-gnu/
LIBRARY_PATH=/usr/lib/gcc/aarch64-linux-gnu/7/:/usr/lib/gcc/aarch64-linux-gnu/7/../../../aarch64-linux-gnu/:/usr/lib/gcc/aarch64-linux-gnu/7/../../../../lib/:/lib/aarch64-linux-gnu/:/lib/../lib/:/usr/lib/aarch64-linux-gnu/:/usr/lib/../lib/:/usr/lib/gcc/aarch64-linux-gnu/7/../../../:/lib/:/usr/lib/
[... 1156 lines cut ...]
   return ((int*)(&Q_WS_MAC))[argc];
                   ^~~~~~~~
                   Q_WS_X11
CMakeFiles/cmTC_bda42.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_bda42.dir/CheckSymbolExists.cxx.o' failed
make[1]: *** [CMakeFiles/cmTC_bda42.dir/CheckSymbolExists.cxx.o] Error 1
make[1]: Leaving directory '/home/nvidia/rtabmap/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_bda42/fast' failed
make: *** [cmTC_bda42/fast] Error 2

File /home/nvidia/rtabmap/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx:
/* */
#include <QtCore/qglobal.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef Q_WS_MAC
  return ((int*)(&Q_WS_MAC))[argc];
#else
  (void)argc;
  return 0;
#endif
}

`

matlabbe, 2020-04-03:
On my side, rtabmap is linked on `libvtkGUISupportQt-6.3.so.1`, not `libvtkGUISupportQt-6.3.so.6.3`. See full ldd log here:
<details> <summary> $ ldd rtabmap | grep vtk </summary>

```bash
ldd bin/rtabmap | grep vtk
	libvtkIOMPIImage-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkIOMPIImage-6.3.so.6.3 (0x0000007f8229a000)
	libvtkRenderingFreeTypeFontConfig-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkRenderingFreeTypeFontConfig-6.3.so.6.3 (0x0000007f82280000)
	libvtkRenderingLOD-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkRenderingLOD-6.3.so.6.3 (0x0000007f82257000)
	libvtkRenderingMatplotlib-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkRenderingMatplotlib-6.3.so.6.3 (0x0000007f82237000)
	libvtkRenderingVolumeOpenGL-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkRenderingVolumeOpenGL-6.3.so.6.3 (0x0000007f82187000)
	libvtkGUISupportQt-6.3.so.1 => /usr/lib/aarch64-linux-gnu/libvtkGUISupportQt-6.3.so.1 (0x0000007f81819000)
	libvtkRenderingOpenGL-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkRenderingOpenGL-6.3.so.6.3 (0x0000007f81576000)
	libvtkInteractionStyle-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkInteractionStyle-6.3.so.6.3 (0x0000007f814e4000)
	libvtkRenderingAnnotation-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkRenderingAnnotation-6.3.so.6.3 (0x0000007f81396000)
	libvtkRenderingFreeType-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkRenderingFreeType-6.3.so.6.3 (0x0000007f812c6000)
	libvtkRenderingCore-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkRenderingCore-6.3.so.6.3 (0x0000007f81050000)
	libvtkFiltersSources-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkFiltersSources-6.3.so.6.3 (0x0000007f80f86000)
	libvtkFiltersGeneral-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkFiltersGeneral-6.3.so.6.3 (0x0000007f80c80000)
	libvtkCommonExecutionModel-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkCommonExecutionModel-6.3.so.6.3 (0x0000007f80bc7000)
	libvtkCommonDataModel-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkCommonDataModel-6.3.so.6.3 (0x0000007f80885000)
	libvtkCommonMath-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkCommonMath-6.3.so.6.3 (0x0000007f80857000)
	libvtkCommonCore-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkCommonCore-6.3.so.6.3 (0x0000007f80543000)
	libvtkIOMPIParallel-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkIOMPIParallel-6.3.so.6.3 (0x0000007f7715b000)
	libvtkIOPLY-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkIOPLY-6.3.so.6.3 (0x0000007f77130000)
	libvtkIOLegacy-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkIOLegacy-6.3.so.6.3 (0x0000007f76fc7000)
	libvtkIOGeometry-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkIOGeometry-6.3.so.6.3 (0x0000007f76ead000)
	libvtkIOImage-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkIOImage-6.3.so.6.3 (0x0000007f76d31000)
	libvtkIOCore-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkIOCore-6.3.so.6.3 (0x0000007f76cb0000)
	libvtkImagingCore-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkImagingCore-6.3.so.6.3 (0x0000007f76af7000)
	libvtkFiltersModeling-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkFiltersModeling-6.3.so.6.3 (0x0000007f767dd000)
	libvtkFiltersCore-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkFiltersCore-6.3.so.6.3 (0x0000007f763b2000)
	libvtkRenderingContextOpenGL-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkRenderingContextOpenGL-6.3.so.6.3 (0x0000007f76372000)
	libvtkViewsContext2D-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkViewsContext2D-6.3.so.6.3 (0x0000007f7634f000)
	libvtkChartsCore-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkChartsCore-6.3.so.6.3 (0x0000007f760f6000)
	libvtkRenderingContext2D-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkRenderingContext2D-6.3.so.6.3 (0x0000007f760a3000)
	libvtkInteractionWidgets-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkInteractionWidgets-6.3.so.6.3 (0x0000007f75d90000)
	libvtkImagingSources-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkImagingSources-6.3.so.6.3 (0x0000007f75d35000)
	libvtkCommonColor-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkCommonColor-6.3.so.6.3 (0x0000007f75d0a000)
	libvtkFiltersExtraction-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkFiltersExtraction-6.3.so.6.3 (0x0000007f75bf9000)
	libvtkCommonTransforms-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkCommonTransforms-6.3.so.6.3 (0x0000007f75bbd000)
	libvtkParallelMPI-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkParallelMPI-6.3.so.6.3 (0x0000007f75b85000)
	libvtkParallelCore-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkParallelCore-6.3.so.6.3 (0x0000007f75a0e000)
	libvtksys-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtksys-6.3.so.6.3 (0x0000007f759b9000)
	libvtkPythonInterpreter-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkPythonInterpreter-6.3.so.6.3 (0x0000007f7599c000)
	libvtkWrappingPython27Core-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkWrappingPython27Core-6.3.so.6.3 (0x0000007f7595a000)
	libvtkRenderingVolume-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkRenderingVolume-6.3.so.6.3 (0x0000007f750c3000)
	libvtkCommonSystem-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkCommonSystem-6.3.so.6.3 (0x0000007f7509f000)
	libvtkImagingHybrid-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkImagingHybrid-6.3.so.6.3 (0x0000007f74eb6000)
	libvtkftgl-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkftgl-6.3.so.6.3 (0x0000007f74e2f000)
	libvtkFiltersGeometry-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkFiltersGeometry-6.3.so.6.3 (0x0000007f74d98000)
[... cut ...]
	libvtkFiltersParallel-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkFiltersParallel-6.3.so.6.3 (0x0000007f6faef000)
	libvtkIONetCDF-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkIONetCDF-6.3.so.6.3 (0x0000007f6fa7c000)
	libvtkImagingFourier-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkImagingFourier-6.3.so.6.3 (0x0000007f6f8e8000)
	libvtkalglib-6.3.so.6.3 => /usr/lib/aarch64-linux-gnu/libvtkalglib-6.3.so.6.3 (0x0000007f6f8b6000)
```
</details>

Try cleaning your cmake cache so that cmake remove previous links to old libraries.
`rm rtabmap/build/CMakeCache.txt`

a-xtron, 2020-04-03:
Thanks, now I managed to get ./rtabmap launch. I did kind of repeat the steps from step 4 just to make sure no other library is miss-linked and emptied the whole build file. Now its time to get the camera running. I hope that having ZED 2 is not a problem :P

Cremzy, 2020-04-05:
After performing the steps on this [post](https://github.com/introlab/rtabmap/issues/427#issuecomment-608052821) , i am getting stuck since I get an error using make after : cmake .. --DWITH_QT=OFF

Current error when trying to make
`~/rtabmap/build$ make
[  8%] Built target rtabmap_utilite
[ 10%] Built target res_tool
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRealSense2.cpp.o
/home/cremzy/rtabmap/corelib/src/camera/CameraRealSense2.cpp: In member function ‘virtual bool rtabmap::CameraRealSense2::init(const string&, const string&)’:
/home/cremzy/rtabmap/corelib/src/camera/CameraRealSense2.cpp:492:32: error: variable ‘std::ifstream in’ has initializer but incomplete type
    std::ifstream in(jsonConfig_);
                                ^
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:547: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRealSense2.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRealSense2.cpp.o] Error 1
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
`

Current setup :
- Jetson Nano Jetpack 4.3 latest
- Intel Realsense D435i

Steps tooken:
1. installed realsense libraries 
https://www.jetsonhacks.com/2019/05/16/jetson-nano-realsense-depth-camera/
./installDependencies
./installRealSense
./buildRealSense

2. Installed ros-melodic-desktop via apt-get
3. installed ros-melodic-rtabmap by apt-get and ros-melodic-rtabmap-ros
4. Uninstalled ros-melodic-cv-bridge
5. Uninstalled ros-meloidc-rtabmap
6. Installed rtabmap by git @ ~/rtabmap
7. cd ~/rtabmap/build
8. cmake .. --DWITH_QT=OFF
9. make
10. error

tkircher, 2020-04-05:
Looks like some json code was introduced recently that broke it.

matlabbe, 2020-04-06:
Indeed as @tkircher shown, this has been fixed yesterday in this commit: https://github.com/introlab/rtabmap/commit/e83838f744fe95f77d35961e52539a61a76fbdec

patrickpoirier51, 2020-07-02:
Bonjour Mathieu,

I finally received my Xavier NX and it comes with JETPACK 4.4
I got stuck trying compiling VTK 6.3 with  gcc (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 7.5.0 and I don t really want to get back to older release :-(

Do you have any plan to update VTK to newer release and using QT5 soon ?

tkircher, 2020-07-03:
@patrickpoirier51 I'm currently running RTAB-Map on an Xavier NX with Qt 5.14.2, VTK 7.1.1, libPCL-1.11, and OpenCV 4.3. Works perfectly.

patrickpoirier51, 2020-07-03:
@tkircher oh great!
Could you share the method to build ?
That would be greatly appreciated

tkircher, 2020-07-03:
Pull down the Qt source from git://code.qt.io/qt/qt5.git, and check out branch 5.14. Build and install according to the directions. Then when you configure VTK 7, make sure to specify -DVTK_QT_VERSION=5. When you subsequently build OpenCV, specify the CUDA arch for Xavier: -DCUDA_ARCH_BIN=7.2, also make sure to specify -DOPENCV_ENABLE_NONFREE=1.

patrickpoirier51, 2020-07-03:
@tkircher Thanks, I will give it a try
What cmake command do you issue to build RTABMAP then ?

tkircher, 2020-07-03:
cmake ..

patrickpoirier51, 2020-07-03:
LOL !!! Hard to make it  simpler  :-)
Thanks I'll report back

tkircher, 2020-07-03:
Incidentally, I got VTK 9.0.20200702 built and tested, and with a few modifications, got RTAB-Map built against it.

@matlabbe The only change that really needs to be made for VTK 9 is to replace QVTKWidget with QVTKOpenGLWidget, in guilib/include/rtabmap/gui/CloudViewer.h and guilib/src/CloudViewer.cpp. This could probably be conditional on VTK_MAJOR_VERSION > 7 since it's present in VTK 8 but wasn't deprecated yet. Also, VTK_LEGACY_REMOVE has to be set for that one place it lives in the code.

Redyiyi, 2020-07-06:
I'm sorry that I have some questions.
I'm new to Jetson Nano and I want to "Remove all references to Qt5 stuff in VTKTargets.cmake......", but I have some problems now. It will be appreciated if anyone tells me how to solve these problems...
First, these cmake files are read-only and I don't know how to remove the references in .cmake...
Second, which references are associated with Qt5? Are the sentences which contain "Qt5"? Or anything else?

s3668134, 2020-11-04:
Hi,

I was getting this issue when running the make command after executing the cmake 

`make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libvtkGUISupportQtWebkit-6.3.so.6.3.0', needed by '../bin/librtabmap_core.so.0.20.6'.  Stop.
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2`

I checked the /usr/lib/aarch64-linux-gnu/ folder for the files and it currently points this file ` libvtkGUISupportQt-6.3.so.6.3.0` -> `libvtkGUISupportQt-6.3.so.1` so I am not sure as to how I can solve this issue. 

One thing that could've caused this issue is "removing all references to Qt5 stuff in /usr/lib/cmake/vtk-6.3/VTKTargets.cmake and /usr/lib/cmake/vtk-6.3/VTKTargets-none.cmake" as I wasn't sure which ones to remove specifically. 

If anyone could help me on how to solve this error, it would be very much appreciated.

marcostrullato, 2020-11-25:
Hi everyone, is anyone of you so kind to assemble a dockerfile with the steps?
I'm going to start working on it, however saving time would be nice.

matlabbe, 2020-11-30:
For the "Remove all references to Qt5 stuff in VTKTargets.cmake......" problem, if copy/paste of the prebuilt files don't work anymore and it is not enough clear to remove all Qt5 related stuff from VTK files (because of a new jetpack for example), as stated in the [post](https://github.com/introlab/rtabmap/issues/427#issuecomment-516914817), the main step to follow is "Build VTK 6.3.0 from source with Qt4 (not Qt5!)". You can rebuild VTK with Qt4, then PCL with that VTK, then Rtabmap with that PCL version.

ChemicalNRG, 2020-12-29:
> Pull down the Qt source from git://code.qt.io/qt/qt5.git, and check out branch 5.14. Build and install according to the directions. Then when you configure VTK 7, make sure to specify [...]

Can you explain how you did this? I am using an AGX with 30GB onboard storage and with de QT5 source on a network drive the build of QT5.14 alone is more than 14GB. It cancelled because there was no space left so maybe it is even 20GB or more. And following the directions you cant use make install, so i cant build it on the NAS and then use make install.

tkircher, 2020-12-29:
@ChemicalNRG I recommend installing a 512GB NVMe drive. The onboard 32GB eMMC is only intended to be a boot volume.

ChemicalNRG, 2020-12-29:
> @ChemicalNRG I recommend installing a 512GB NVMe drive. The onboard 32GB eMMC is only intended to be a boot volume.

I was already looking for that indeed, but because this is a jetson nano topic which dont have that possibility and also the qt5 files which are installed with ros or rtabmap are using much less space, i thought maybe i was doing something wrong or it can be installed with less space needed.

VisionaryMind, 2020-12-30:
I am encountering issues running slam rtabmap.launch for Azure Kinect on Jetson Nano with Jetpack 4. Here are the steps taken:

1. Installed Azure Kinect SDK and ran k4aviewer successfully.
2. Installed ros-melodic-desktop via apt-get
3. Created workspace and git cloned BrutusTT/vision_opencv.git and ros-perception/image_transport_plugins
4. Installed ros-melodic-rtabmap and ros-melodic-rtabmap-ros via apt-get
5. Uninstalled ros-melodic-cv-bridge
6. Uninstalled ros-meloidc-rtabmap
7. Built standalone rtabmap WITH_QT=OFF and make install to usr/local

I am able to "rosrun rviz rviz" and then "roslaunch azure_kinect_ros_driver driver.launch", however nothing appears in RVIZ. Further, if I attempt to "roslaunch azure_kinect_ros_driver slam_rtabmap.launch rviz:=true", it connects to the Kinect camera, however throws the following error:

[ERROR] [1609349860.812967146]: **Failed to load nodelet [/k4a/rectify_rgb] of type [image_proc/rectify] even after refreshing the cache: According to the loaded plugin descriptions the class image_proc/rectify with base class type nodelet::Nodelet does not exist.** Declared types are  Azure_Kinect_ROS_Driver/K4AROSBridgeNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid rtabmap_ros/data_odom_sync rtabmap_ros/data_throttle rtabmap_ros/disparity_to_depth rtabmap_ros/icp_odometry rtabmap_ros/imu_to_tf rtabmap_ros/obstacles_detection rtabmap_ros/obstacles_detection_old rtabmap_ros/point_cloud_aggregator rtabmap_ros/point_cloud_assembler rtabmap_ros/point_cloud_xyz rtabmap_ros/point_cloud_xyzrgb rtabmap_ros/pointcloud_to_depthimage rtabmap_ros/rgb_sync rtabmap_ros/rgbd_odometry rtabmap_ros/rgbd_relay rtabmap_ros/rgbd_sync rtabmap_ros/rgbdicp_odometry rtabmap_ros/rtabmap rtabmap_ros/stereo_odometry rtabmap_ros/stereo_sync rtabmap_ros/stereo_throttle rtabmap_ros/undistort_depth
[ERROR] [1609349860.813439970]: The error before refreshing the cache was: According to the loaded plugin descriptions the class image_proc/rectify with base class type nodelet::Nodelet does not exist. Declared types are  Azure_Kinect_ROS_Driver/K4AROSBridgeNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid rtabmap_ros/data_odom_sync rtabmap_ros/data_throttle rtabmap_ros/disparity_to_depth rtabmap_ros/icp_odometry rtabmap_ros/imu_to_tf rtabmap_ros/obstacles_detection rtabmap_ros/obstacles_detection_old rtabmap_ros/point_cloud_aggregator rtabmap_ros/point_cloud_assembler rtabmap_ros/point_cloud_xyz rtabmap_ros/point_cloud_xyzrgb rtabmap_ros/pointcloud_to_depthimage rtabmap_ros/rgb_sync rtabmap_ros/rgbd_odometry rtabmap_ros/rgbd_relay rtabmap_ros/rgbd_sync rtabmap_ros/rgbdicp_odometry rtabmap_ros/rtabmap rtabmap_ros/stereo_odometry rtabmap_ros/stereo_sync rtabmap_ros/stereo_throttle rtabmap_ros/undistort_depth
[FATAL] [1609349860.814003056]: Failed to load nodelet '/k4a/rectify_rgb` of type `image_proc/rectify` to manager `manager'

I attempted to "sudo ldconfig", but this did not help. Why doesn't ROS see the image_proc/rectify class? What have I missed?

ChemicalNRG, 2020-12-30:
> I am encountering issues running **slam_rtabmap.launch** for Azure Kinect on Jetson Nano with Jetpack 4. Here are the steps taken:
> 
> 1. Installed Azure Kinect SDK and ran k4aviewer successfully.
> 2. Installed ros-melodic-desktop via apt-get
> 3. Created workspace and git cloned BrutusTT/vision_opencv.git and ros-perception/image_transport_plugins
> 4. Installed ros-melodic-rtabmap and ros-melodic-rtabmap-ros via apt-get
> 5. Uninstalled ros-melodic-cv-bridge
> 6. Uninstalled ros-melodic-rtabmap
> 7. Built standalone rtabmap WITH_QT=OFF and "sudo make install" to usr/local
> 
> I am able to "rosrun rviz rviz" and then "roslaunch azure_kinect_ros_driver driver.launch", however nothing appears in RVIZ. Further, if I attempt to "roslaunch azure_kinect_ros_driver slam_rtabmap.launch rviz:=true", it connects to the Kinect camera, however throws the following error:
> 
> [ERROR] [1609349860.812967146]: **Failed to load nodelet [/k4a/rectify_rgb] of type [image_proc/rectify] even after refreshing the cache: According to the loaded plugin descriptions the class image_proc/rectify with base class type nodelet::Nodelet does not exist.** Declared types are Azure_Kinect_ROS_Driver/K4AROSBridgeNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid rtabmap_ros/data_odom_sync rtabmap_ros/data_throttle rtabmap_ros/disparity_to_depth rtabmap_ros/icp_odometry rtabmap_ros/imu_to_tf rtabmap_ros/obstacles_detection rtabmap_ros/obstacles_detection_old rtabmap_ros/point_cloud_aggregator rtabmap_ros/point_cloud_assembler rtabmap_ros/point_cloud_xyz rtabmap_ros/point_cloud_xyzrgb rtabmap_ros/pointcloud_to_depthimage rtabmap_ros/rgb_sync rtabmap_ros/rgbd_odometry rtabmap_ros/rgbd_relay rtabmap_ros/rgbd_sync rtabmap_ros/rgbdicp_odometry rtabmap_ros/rtabmap rtabmap_ros/stereo_odometry rtabmap_ros/stereo_sync rtabmap_ros/stereo_throttle rtabmap_ros/undistort_depth
> [ERROR] [1609349860.813439970]: The error before refreshing the cache was: According to the loaded plugin descriptions the class image_proc/rectify with base class type nodelet::Nodelet does not exist. Declared types are Azure_Kinect_ROS_Driver/K4AROSBridgeNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid rtabmap_ros/data_odom_sync rtabmap_ros/data_throttle rtabmap_ros/disparity_to_depth rtabmap_ros/icp_odometry rtabmap_ros/imu_to_tf rtabmap_ros/obstacles_detection rtabmap_ros/obstacles_detection_old rtabmap_ros/point_cloud_aggregator rtabmap_ros/point_cloud_assembler rtabmap_ros/point_cloud_xyz rtabmap_ros/point_cloud_xyzrgb rtabmap_ros/pointcloud_to_depthimage rtabmap_ros/rgb_sync rtabmap_ros/rgbd_odometry rtabmap_ros/rgbd_relay rtabmap_ros/rgbd_sync rtabmap_ros/rgbdicp_odometry rtabmap_ros/rtabmap rtabmap_ros/stereo_odometry rtabmap_ros/stereo_sync rtabmap_ros/stereo_throttle rtabmap_ros/undistort_depth
> [FATAL] [1609349860.814003056]: Failed to load nodelet '/k4a/rectify_rgb`of type`image_proc/rectify`to manager`manager'
> 
> I attempted to "sudo ldconfig", but this did not help. Why doesn't ROS see the image_proc/rectify class? What have I missed?

I have had the same problem. Check if k4aviewer is working **without sudo**. 
1. Could be that you didnt copy the file "99-k4a.rules" in /etc/udev/rules.d (k4aviewer will only see the camera when you do "sudo k4aviewer"
2. Could be that a program that started the kinect did not stop it correctly. 
3. Could be you are using usb 2 or an usb 3 connection that does not work well with the kinect. (but if k4aviewer is working that is not likely)
4. Could be that you did not install Azure Kinect ROS Driver (git clone https://github.com/microsoft/Azure_Kinect_ROS_Driver.git in your catkin source folder)

ChemicalNRG, 2020-12-30:
The most rediculous thing is that everyone with a jetson, is trying to make rtabmap and or ROS work with the included opencv version (me included), because we all think it is build with cuda support, while it is most likely not.
I found that on another forum and searched it up and indeed there are many reports that the NVIDIA SDK manager failes in delivering an opencv version with cuda support.
You can check this with jtop (sudo -H pip install -U jetson-stats) do a reboot after that and start with "jtop"
Or with python:
1. import cv2
2. print cv2.getBuildInformation()

tkircher, 2020-12-30:
It's true, the default system libraries on Jetpack 4.4 leave much to be desired.

VisionaryMind, 2020-12-30:
The first thing I did before even attempting this project was to compile OpenCV with CUDA support. I believe the problem here is that Jetson Nano is simply unable to accommodate a K4A stream. For that matter, it is really unable to accommodate any stream, including from RealSense cameras, which either lock it up or cause it to turn off spontaneously. I am reluctant to purchase a Jetson AGX, since it will likely suffer from the same shortcomings of all other NVDIA IoT devices. Glad to hear AGX performs better, though I would sooner go with an Intel NUC or something else that is more fit for heavy-lifting.

I finally tabled even trying to get RTtabMap working on Nano today when seeing, in RViz, that it begins to drop frames about 60 seconds into capture. I chalk this up to both lacking hardware (Azure Kinect) combined with insufficient memory and GPU power (Nano). I will likely be returning the Azure Kinect while still within the 30-day purchase window. Unless anyone here has achieved anything interesting with it -- I find it to be underperforming and overpriced. The Nano simply emphasizes its weaknesses, as it is not fit for hardcore IoT applications and, in most benchmarks we have run, is actually much slower than an RPi4 running native Ubuntu 20.10.

I believe we should all question the viability of these cheap consumer IoT devices. They have not delivered, nor do their manufacturers stand behind them, as you have pointed out.

VisionaryMind, 2020-12-30:
Just a few clarifications here, before I tag this topic as unresolvable:

> I have had the same problem. Check if k4aviewer is working **without sudo**.

K4aViewer works perfectly fine without sudeo, the 99-k4a.rules were copied to /etc/udev/rules.d. In fact, the viewer works fine on Nano. No lock-ups. It is ROS that cannot handle K4a, and to be honest, I haven't seen any evidence of anyone getting it work work aside from some advice on what to install. No screenshot, it's as good as theory.

> 3. Could be you are using usb 2 or an usb 3 connection that does not work well with the kinect. (but if k4aviewer is working that is not likely)
All USB ports on the Jetson Nano B01 are USB3.

> 4. Could be that you did not install Azure Kinect ROS Driver (git clone https://github.com/microsoft/Azure_Kinect_ROS_Driver.git in your catkin source folder) OR did not source your catkin workplace so ROS cant find the driver (you can check this with "echo $ROS_PACKAGE_PATH", which should return your catkin workplace and the ros folder in /opt)

No, I can run "roslaunch azure_kinect_ros_driver driver.launch", open RViz, and subscribe to its Pointcloud2 topic. It will display one point cloud and then the mouse will start to lag and eventually freeze, and the console will say "queue full". In other words --- ROS simply doesn't work with Azure Kinect and Nano --- period. 

> I also recommend that you use the slam_rtabmap.launch that Matlabbe edited if you not do that already. If you do, you have to install ros-melodic-imu-filter-madgwick in order for it to work. The edit is a huge improvement over the original file, for me the difference between no pointcloud at all (with a lattepanda alpha 864) and a really good pointcloud.

That was also the first thing I applied. It makes no difference. Just to re-iterate, the probem is not hardware or configuration related. Everything is setup properly. RTabMap-ROS can't find the image_proc/rectify class. That is what the error is showing, and this particular error has never been seen nor addressed before, which is why I brought it up here. Go and search Google for it. Apart from there being close to nothing on "Jetson Nano" + "Azure Kinect" and "RtabMap", this particular error has surfaced, but not in this particular context.

My hope was that someone in the Introlab repository would actually know the code deeply enough to immediately see what the problem is. It's becoming gradually clear to me that all of these tools have just been glued together with sticky notes.

ChemicalNRG, 2020-12-30:
I am glad i found that at least. If i didnt had all these problems i most likely never used the true potential of the AGX, because .. who would think that libraries created/compiled by nvidia, for a development board of nvidia, does not include cuda support :p

Also, i believe the reports of this issue also include Jetpacks before 4.4 but i am not sure.

tkircher, 2020-12-31:
Strictly speaking, Jetpack is just generic arm64 Ubuntu plus the firmware required for the system to boot, CUDA, L4T, TensorRT, etc., NVidia doesn't actually provide OpenCV libraries.

ChemicalNRG, 2020-12-31:
> Just a few clarifications here, before I tag this topic as unresolvable:
> 
> > I have had the same problem. Check if k4aviewer is working **without sudo**.
> 
> K4aViewer works perfectly fine without sudo, the 99-k4a.rules were copied to /etc/udev/rules.d. In fact, the viewer works fine on Nano. No lock-ups. It is ROS that cannot handle K4a, and to be honest, I haven't seen any evidence of anyone getting it work work aside from some advice on what to install. No screenshot, it's as good as theory.
> 
> > 1. Could be you are using usb 2 or an usb 3 connection that does not work well with the kinect. (but if k4aviewer is working that is not likely)
> 
> I am using exclusively one of the four Jetson Nano B01 ports that are USB3.
> 
> > 1. Could be that you did not install Azure Kinect ROS Driver (git clone https://github.com/microsoft/Azure_Kinect_ROS_Driver.git in your catkin source folder) OR did not source your catkin workplace so ROS cant find the driver (you can check this with "echo $ROS_PACKAGE_PATH", which should return your catkin workplace and the ros folder in /opt)
> 
> No, I can run "roslaunch azure_kinect_ros_driver driver.launch", open RViz, and subscribe to its Pointcloud2 topic. It will display one point cloud and then the mouse will start to lag and eventually freeze, and the console will say "queue full". In other words --- ROS simply doesn't work with Azure Kinect and Nano --- period.
> 
> > I also recommend that you use the slam_rtabmap.launch that Matlabbe edited if you not do that already. If you do, you have to install ros-melodic-imu-filter-madgwick in order for it to work. The edit is a huge improvement over the original file, for me the difference between no pointcloud at all (with a lattepanda alpha 864) and a really good pointcloud.
> 
> That was also the first thing I applied. It makes no difference. Just to re-iterate, the probem is not hardware or configuration related. Everything is setup properly. RTabMap-ROS can't find the image_proc/rectify class. That is what the error is showing, and this particular error has never been seen nor addressed before, which is why I brought it up here. Go and search Google for it. Apart from there being close to nothing on "Jetson Nano" + "Azure Kinect" and "RtabMap", this particular error has surfaced, but not in this particular context.
> 
> My hope was that someone in the Introlab repository would actually know the code deeply enough to immediately see what the problem is. It's becoming gradually clear to me that all of these tools have just been glued together with sticky notes.

First of all, i have had help and answers on all my questions from Matlabbe, which is one or the only developer of Rtabmap.
Give it a couple of days and he will try to help you when he reads it. 

I understand your frustrations because ... well read [my](https://answers.ros.org/question/366177/need-some-help-with-mapping-and-azure-kinect/) [posts](https://answers.ros.org/question/366177/need-some-help-with-mapping-and-azure-kinect/).

It has cost me weeks to get even the smallest results, mostly because everything was new for me, lots of cmake errors which i didnt understand but now i can troubleshoot a little bit.
Did you try this software setup that now fails on a laptop or computer?
Because having a new devices like the kinect, new software (for me at least) and the nvidia (with arm64) is not an easy way to try everything at ones and find the problem.
You can also try the standalone version of Rtabmap and see how that works on the nano.
I know that Matlabbe has it working but i dont know how a rosbag, which i think he used, compares to a Azure Kinect.

And did you try something other than gdm3 to free up memory? You mentioned "one of the B01 ports" so you do have the 4GB version of the Nano right?

I agree with you that i think Microsoft is marketing the Azure Kinect too much as a product that is working allready or for tweakers (so works with a couple of hours). Then all the media displaying good results doesn't help either, which was never the case with the first Kinects because it was allways for the xbox and also marketed like that. The software is nowhere near ready (at least not a package for the use case i had in mind), but then again the Kinect is always been a product which was great through the efforts of the opensource community (from what i've seen way better than the D455 you mention, even a used Kinect v2) and i think this will be no exception. It just needs some time. I think now the Kinect v2 is also the much better option because you can get it for a couple of bucks and the community has allready spend a lot off time on that device.

I don't agree with you about the NUC, because the Lattepanda is basically that and the results are nowhere near that of the jetson agx and my hp elitebook 8760w (which has an old quadro 4000m, cuda disabled). So i believe you need gpu power or at least you have more use for that. But maybe there are some nucs that have that, i dont know.

ChemicalNRG, 2020-12-31:
> Strictly speaking, Jetpack is just generic arm64 Ubuntu plus the firmware required for the system to boot, CUDA, L4T, TensorRT, etc., NVidia doesn't actually provide OpenCV libraries.

I understand that, that's why i did say "/compile" because it was installed through the Nvidia SDK. That only makes sense if it is used by one of the things in the Jetpack, which dont make sense to not compile that with gpu acceleration your product is marketing and produced for (aside from ML).

ChemicalNRG, 2020-12-31:
Maybe here we can find some thing to reverse the qt5 install by the JetsonPack?
https://www.qt.io/blog/2016/11/10/qt-nvidia-jetson-tx1-device-creation-style
https://doc.qt.io/QtForDeviceCreation/qtee-preparing-hardware-jetsontx2.html

in the first article it looks like it is installed besides a 'normal' qt5 package.
I started allready a fresh install with the oldest sdk i could find for the AGX.
Also installed just the Jetson OS  + cuda and developer tools (so everything till the agreement) and now triing to install the programs i need before i will try to start install the jetpack. Hope that will solve a lot of problems.

VisionaryMind, 2020-12-31:
I had a chance today to install RTabMap with CUDA on a Windows machine with a GeForce RTX 2070 GPU and AMD Ryzen Threadripper 2950X 16-Core Processor. This computer can handle literally anything, and we use it frequently for photogrammetry and GPU-intensive 3D modeling. First, this clearly could not be used for true SLAM (moving around a room or a large area), as the camera gets lost with any slight movement beyond a centimeter (constantly showing red). We were able to capture a reasonably accurate point cloud, but at ~1 minute mark, it suddenly flips around and there are skewed points (double point clouds perpendicular to each other). 

I would have thought that Nano was playing a larger part in the poor scan results, but it appears to be the Azure and its low-performance Windows driver. It sounds like it may perform better on Xavier AGX, but I would have thought that a CUDA-enabled RTX 2070 would be powerful enough to handle this pipeline. Clearly not. 

I will attempt to work with the settings, but the performance for Windows + Azure Kinect  + RTabMap CUDA is so lacking, I am skeptical that any significant improvements may be realized. KinectFusion, for example, is much faster and does not get lost, even with faster movements.

tkircher, 2020-12-31:
@VisionaryMind It's definitely a problem on your end. I regularly scan areas outside for ten to fifteen minutes with a Jetson Xavier NX and Azure Kinect with RTAB-Map. I did find with the UDOO Bolt v8 that the USB controller has trouble keeping up with the Kinect and has buffer issues, with theoretically faster CPU, memory, and I/O.

VisionaryMind, 2020-12-31:
> @VisionaryMind It's definitely a problem on your end. I regularly scan areas outside for ten to fifteen minutes with a Jetson Xavier NX and Azure Kinect with RTAB-Map. I did find with the UDOO Bolt [...]

Which odometry method are you using? Kinect does not have proper IMU (only gyro and accel), and my own experiments with larger scans outdoors using RealSense cameras have proven the mandatory need for tracking (with T265, for example). Are you saying you are getting <1% loop closure with Xavier NX +Kinect + RTAB-Map? If so, some specifics would be very appreciated. 

I have a feeling that Kinect does not play well with AMD arch. Xavier NX / AGX will both likely be more compatible. I don't think there is an issue with the USB 3.0 port on this computer, as I use it with other devices, including D455 / T265 without an issue. But I'll look into that possibility.

Pursuant to this post's topic, I still contend that Nano isn't powerful enough to perform proper SLAM with Kinect. So far, I see success stories only with Tx2 and Xavier series. Our volumetric capture team is using high-end i9 Intel NUCs, but that's a different animal. If AGX + external tracking with T265 + Kinect are viable, then we'll pursue that route. We do scans with drones as well, and I fear loop closure will be all but absent in such an application.

VisionaryMind, 2020-12-31:
Yes, I took a look at the specs for UDOO Bolt v8 and it implements an AMD Ryzen as well. I don't think it's the USB controller --- it's the AMD arch. Microsoft claims support for AMD, but I have seen nothing but problems on any and all computers with this arch. I think for Kinect to work properly --- it "prefers" Intel. I could be wildly off, but that's the anecdotal account.

tkircher, 2021-01-01:
@VisionaryMind I boot and run my NX off an NVMe SSD, yeah.

ChemicalNRG, 2021-01-02:
> Pull down the Qt source from git://code.qt.io/qt/qt5.git, and check out branch 5.14. Build and install according to the directions. Then when you configure VTK 7, make sure to specify [...]

Can you share your exact qt5 configure command please?

When i use: ../qt5/configure -prefix /usr/local/opt/qt5 -nomake tests -nomake examples -opengl dynamic

I always get this error:
Note: The following modules are not being compiled in this configuration:
    3dcore
    3drender

ERROR: The OpenGL functionality tests failed!
You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],
QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.

When i use: -opengl desktop or -opengl es2 or not use -opengl .. in the config i get no errors, but i am afraid non of these are what is needed right?

tkircher, 2021-01-02:
Use -opengl desktop. I also recommend using Qt 5.15 now. This thread is several months old so many of the previous comments are obsolete.

ChemicalNRG, 2021-01-02:
> Use -opengl desktop. I also recommend using Qt 5.15 now. This thread is several months old so many of the previous comments are obsolete.

I am currently not using anything from jetpack, but if i want to, isnt opengl es2 not required? So is -opengl dynamic required then?

ChemicalNRG, 2021-01-03:
> I had a chance today to install RTabMap with CUDA on a Windows machine with a GeForce RTX 2070 GPU and AMD Ryzen Threadripper 2950X 16-Core Processor. This computer can handle literally anything, and we use it frequently for photogrammetry and GPU-intensive 3D modeling. First, this clearly could not be used for true SLAM (moving around a room or a large area), as the camera gets lost with any slight movement beyond a centimeter (constantly showing red). We were able to capture a reasonably accurate point cloud, but at ~1 minute mark, it suddenly flips around and there are skewed points (double point clouds perpendicular to each other). This makes the entire scan unusable.

Now i am sure you are not using the launch file i did mention, because that is the behaviour with the original launch file.
And i also got lesser results on windows than i got on ubuntu.

Have a look at [this](https://github.com/OpenKinect/libfreenect2/issues/807) for usb buffer issue. I dont know if it is needed though, i want to test it, but i dont know how i can visualize that buffer. I dont believe it is needed for 720p because that is running fine for me, but maybe 1080p and beyond will have improvements.

VisionaryMind, 2021-01-04:
@ChemicalNRG, I've been running a series of experiments. Many of these subjects are off topic for this thread, so perhaps we can start a new one, but for now, I'll respond here. I have come to the conclusion that RTabMap with Azure Kinect is too much for the Nano to handle. The primary issue is reliance on microSD for swap and storage, and this is likely the primary bottleneck. An NVMe SSD would be helpful, but Nano doesn't support it, so I have gone ahead and procured a Xavier AGX with 1Tb of SSD via M.2. I need it for other visual-AI projects, so now it can be used for RTabMap. 

If you will have a few minutes, could you provide some quick guidance on what your AGX configuration looks like for RTabMap? Jetpack 4.4.1 for AGX appears to be installing all the CUDA libraries, but your previous comments would lead me to believe OpenCV is not compiled for it. I will perform the recommended test to verify. 

> Now i am sure you are not using the [launch](https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/166) file i did mention (or not installed the imu-madgwick filter), because that is the behaviour with the original launch file.

Are you saying that you resolved this behavior _on a Jetson Nano_ with that launch file? I have both installed the IMU-Madgwick-Filter and the launch file you refer to, but the problem becomes worse if I turn on IMU filtering in RTapMap with Madgwick. I haven't scrutinized the launch file, but I presume that's what it is doing.

> And i also got lesser results on windows than i got on ubuntu.

Well, Windows isn't particularly suited to CV or AI. Actually, it's not really suited to anything. I have more headaches when I use Windows, but am "forced" to have it around to run certain software such as Unity, Unreal Engine, Houdini, etc. Going forward, our team has made a commitment to avoid Windows at all costs for any CV work. My problem was actually an AMD USB host controller, as Microsoft states K4A does not work with such controllers. I've procured an Intel PCI controller board with 3.1 ports, and will test again to see if that helps.

> 
> Have a look at [this](https://github.com/OpenKinect/libfreenect2/issues/807) for usb buffer issue. I dont know if it is needed though, i want to test it, but i dont know how i can visualize that buffer. I dont believe it is needed for 720p because that is running fine for me, but maybe 1080p and beyond will have improvements. But i have seen only a mention if this in case of multiple kinects, and i think that if k4aviewer with the same sensors and resolution is running fine, that should also be the case for anything else, because it should be the same amount of data.

Really, what I want is the full UHD RGB with WFOV_UNBINNED depth. That all but cripples Nano. Have you gotten it working on the AGX? I don't think the USB buffer is issue, at least not on the Nano. Nano simply isn't powerful enough to handle high-speed Kinect streams --- on the bus itself. The main roadblock, as I've proposed above, is the microSD. Even with swap, it doesn't seem to be fast enough to stream large BAGs or MKVs.
>
> What i also like to know if it makes a difference to use jpeg instead of bgra (which i believe is the default), so it maybe relies more on gpu with NVJPG

That's what we are using. Actually, this brings up another topic --- it occurred to me that the "proper" way to do high-resolution scans with Kinect (on any hardware) is to first record an MKV using K4ARecord and then stream it RTabMap. Unfortunately, RTabMap seems to take every 5th or 10th frame, so it misses a lot of keypoints in the process. Perhaps it would do better with raw RGB + depth images, extracted from a Kinect MKV, but that brings up another issue --- Open3D (the easiest way to do that) is yet unable to extract RGB / depth images from Kinect MKV's. It's a known bug at this point, and I haven't found any repo's on Github that implement RGB + Depth image streaming. It's all pointcloud oriented.

Again, if you have a chance to share the basics of your AGX config, I will try to duplicate here. Nano is not viable on RTabMap from my perspective. If you end up getting it to work (and I mean "really" work, not just updating the point cloud every 5-6 seconds), please let me know. It would be useful to know it is usable in this pipeline, as AGX seems to be ESD sensitive, and probably wouldn't be a good candidate for mobile, handheld scans without rigging a custom enclosure.

VisionaryMind, 2021-01-05:
Just a quick update here, as this will be the last. For anyone reading, the Xavier AGX is no improvement over Jetson Nano. It suffers from the same inability to support OpenCV 4 CUDA + ROS, and the hacks that work for  Jetson Nano may not be applied to AGX. It would appear NVIDIA has not employed top-tier engineers, as OpenCV is included on the Jetpack without CUDA compilation --- for an edge device that is allegedly "built for CUDA". I don't see that any of these AI-vision libraries will work on NVIDIA hardware, despite claims that ROS Melodic may be compiled (with Noetic cross-references) to support OpenCV. None of those hacks work at all. Lots of people saying they've got it working but no results to show. NVIDIA knows their devices are not compatible with CUDA-compiles OpenCV, as integration is quite straightfoward _were the hardware able to accommodate it_.

ChemicalNRG, 2021-01-06:
> Just a quick update here, as this will be the last. For anyone reading, the Xavier AGX is no improvement over Jetson Nano. It suffers from the same inability to support OpenCV 4 CUDA + ROS, and the hacks that work for Jetson Nano may not be applied to AGX. It would appear NVIDIA has not employed top-tier engineers, as OpenCV is included on the Jetpack without CUDA compilation --- for an edge device that is allegedly "built for CUDA". I don't see that any of these AI-vision libraries will work on NVIDIA hardware, despite claims that ROS Melodic may be compiled (with Noetic cross-references) to support OpenCV.
> 
> None of those hacks work at all. Lots of people saying they've got it working but no results to show. NVIDIA knows their devices are not compatible with CUDA-compiled OpenCV, as integration is quite straightfoward _were the hardware able to accommodate it_. We have run multiple ML neural nets on the AGX as well, and it fails to deliver results on 80% of the tests -- and these are quite simple applications. Vision-oriented libraries should be the easiest to implement. All in all, very disappointed to see there are no options available for SLAM in this category. Perhaps @matlabbe would care to share his own hardware configurations in detail. I am dubious that NVIDIA hardware is present in his work to any significant degree.

If you so convinced about that, why should anyone bother to convince you otherwise? No one here works for Nvidia or has a gain from it, if you drop nvidia for something else. Only in this thread there are a number of people that have at least opencv with cuda support working. 2 of them already said that it are your settings that are wrong. You dont bother to post them, and want to spend time to anything but the most important thing, your launch or config file (of you want to try the standalone Rtabmap).

Aside from that, why Uhd while the depth pointcloud is only 512x512 / 640x576 @ 30fps? And if you had even read the article in one of the post i gave you, you already knew that Matlabbe has tested that binned gives better results then unbinned. So the choise is 1024x1024 @ 15fps or 512x512 @ 30fps, which probably results in using 512x512. So beyond 1080p is useless, especially if your device also strucles to run propperly.

[53 more replies]

closed 2025-01-12 by matlabbe