introlab/rtabmap #428: Interfacing and viewing ROS RTAB map on android?

GrantDare, 2019-07-30:
Hi Team,

I am doing a project where I am using the RTAB map algorithm with ROS to produce a 3D map and am using a custom android application to interface with the robot. We are using ROSJava to connect to the robot and so can access the PointCloud2 topics. I am requiring a way to view the resulting RTAB map on android based on the ROS output. I see there is an RTABMap application for the Tango however I was wondering if there was a way to leverage just the viewer from the application? 

Thanks team for any assistance you can provide me :)

matlabbe, 2019-07-30:
The android app doesn't use ROS and the visualization is done in cpp (jni). See [RTABMapApp](https://github.com/introlab/rtabmap/blob/master/app/android/jni/RTABMapApp.cpp)::Render() for the main rendering loop. This is where we process rtabmap events, creating clouds and updating the graph. On Java side, it is simply a [GLSurfaceView](GLSurfaceView) view, for which we do the rendering on jni side. It would be easier if you receive the PointCloud2 on jni side already (is ROSJava using jni under the hood for subscriber and publisher?), so there is no conversion between java and cpp. For the cpp side, the app follows mainly the c++ example here: https://github.com/introlab/rtabmap/wiki/Cplusplus-RGBD-Mapping

cheers,
Mathieu

GrantDare, 2019-07-30:
Thanks @matlabbe,

I am not the android developer so I have passed your feedback to him to assess the applicability. I have found a [point cloud android viewer](http://wiki.ros.org/android_pointcloud_viewer). Is there a way to view the optimized map as a PointCloud2 message? I really appreciate your support :)

matlabbe, 2019-07-31:
I am not sure what is the application, but the code I referred is to do SLAM directly in the android app. If rtabmap can be started independently of the android UI (so like a standard ROS node), you could subscribe to `/rtabmap/cloud_map` (sensor_msgs/PointCloud2) topic generated from rtabmap to see the map. http://wiki.ros.org/android_pointcloud_viewer may be then used directly.