introlab/rtabmap #429: apriltags on ceiling

molinadavid, 2019-07-30:
Hi Mathieu,

I am now trying to use apriltags with the external package as you suggested in a different issue, my current setup is as follow.
- Apriltags are on the ceiling looking down at a distance of about 4 Mts from the camera.
- The TF is provided using the latest version of apriltags_ros.
- The camera looking at the tags is pointing up (the TF on the robot description is correct)
- The camera that RtabMap uses is pointing front (is a different from the one used for the tags)

Keeping this setup in mind is it possible to successfully connect the tag_detection topic to RtabMap? I did a quick database for testing and the constrains on the mapGraph looked very strange and the localization got very messed up.

I am providing a database in the hope that it can help you and the launch file used to make the map, if is of any use I could also provide a rosbag with the related topics.

[rtabmap_ceilingTags.db](https://www.dropbox.com/s/857lw36pdqhxd6n/rtabmap_ceilingTags.db?dl=0)
[rtabmap_map.launch](https://www.dropbox.com/s/mizs12hu2zbd7ay/rtabmap_map.launch?dl=0)

As always thank you very much for your help.

Best regards,
David

matlabbe, 2019-07-30:
Hi David,

There is indeed something strange, it seems that tag 2 is only ~1/2 meters over the ground while the tag 1 is ~1/2 meters **under** the ground. See screenshot:
[image]

It could be a TF problem. If you use RVIZ and add TF, are the frames published by apriltag_ros make sense? Are both tags 1 and 2 over the ground about 4 meters?

I tried enabling marker detection (`RGBD/MarkerDetection=true`) when replaying this database, but tags are indeed detected with the front camera, not the one pointing up.
[image]
Without tag optimization:
[image]

molinadavid, 2019-07-31:
Hi Mathieu,

Thank you very much for your response.

The tags in reality are about 3.8 mts from the ground, on Rviz the tags do make sense and the relative position is accurate enough when measuring from the camera (about 20~30 cm error margin)

See the following screenshot
[image]

I took a rosbag including the relevant topics, if you want some different topics please do let me know.
[rtabmap_ceilingTags.bag](https://www.dropbox.com/s/6fjl3lndv9i1txo/rtabmap_ceilingTags.bag?dl=0)

As always, thank you very much for all your help

Best regards,
David

matlabbe, 2019-08-01:
The frame_id in `tag_detection` message is the same than `frame_id` set in both cameras:
```bash
$ rostopic echo /camera/left/camera_info/header/frame_id 
"zed_left_camera_optical_frame"

$ rostopic echo /up/camera/left/camera_info/header/frame_id
"zed_left_camera_optical_frame"

$ rostopic echo /tag_detections/header/frame_id
"zed_left_camera_optical_frame"

```

The wrong frame may be then used to transform back the tag in `base_footprint`.

cheers,
Mathieu