introlab/rtabmap #430: anyone uses ceres in the project?

RichardYao1995, 2019-08-01:
I found ceres hard to use in the project. When I call function or class from ceres, it shows undefined reference to that.
can anyone help me with that? I am sure that my CMakeLists.txt is right and ceres can be used in other project.

matlabbe, 2019-08-01:
It is like the libraries are not linked on compilation. What changes did you make to CMakeLists.txt?

RichardYao1995, 2019-08-02:
> It is like the libraries are not linked on compilation. What changes did you make to CMakeLists.txt?

I add three lines in example/Export/CMakeLists.txt 
FIND_PACKAGE(Ceres REQUIRED)
${CERES_INCLUDE_DIR}
${CERES_LIBS}
in the suitable place which can be run correctly in another project which has the same code.

matlabbe, 2019-08-09:
You may update those lines:
https://github.com/introlab/rtabmap/blob/262d74d1e323afcf64145b6389f336b642cac347/examples/Export/CMakeLists.txt#L31-L41

to:
```cmake
FIND_PACKAGE(Ceres REQUIRED)

SET(INCLUDE_DIRS
	${RTABMap_INCLUDE_DIRS}
    ${OpenCV_INCLUDE_DIRS}
    ${PCL_INCLUDE_DIRS}
    ${CERES_INCLUDE_DIR}
)

SET(LIBRARIES
	${RTABMap_LIBRARIES}
	${OpenCV_LIBRARIES}
	${PCL_LIBRARIES}
        ${CERES_LIBS}
)
```
Only code in the example `main.cpp` will have access to ceres.

RichardYao1995, 2019-08-12:
In fact, I have added these as you said and I don't think that's the real reason.
I linked the libs to .cpp files and other libs can be used correctly.
When I include .h files from Ceres, it comes no problem. But when I try to use it, the problem comes out.
Another project with the same code and CMakeLists.txt except rtabmap part, the Ceres runs correctly. That's why I thought your code may have conflicts with Ceres.

matlabbe, 2019-08-17:
Hi,

It is difficult to debug without any errors posted here. Are you linking on the static or shared library? I just [implemented](https://github.com/introlab/rtabmap/blob/master/corelib/src/optimizer/OptimizerCeres.cpp) **OptimizerCeres**  to see if it can work. I don't have any problems of compiling or running with it. I built Ceres 1.12 from source with shared library.

cheers,
Mathieu