introlab/rtabmap #431: Getting pose instead of odometry for inside-out tracking

pringithub, 2019-08-01:
Hello,

I'm attempting to use the RealSense d435 + rtabmap for inside-out localization within a moving vehicle (point cloud cropped at 1m so not to capture points outside of vehicle). The method works well enough at 30fps no less .. until the odometry is lost due to a jerky camera or not enough features are captured. 

I care about instantaneous pose more than full odometry, and was wondering if there was a way to get that without having to click on 'reset odometry' every time. Perhaps increasing the variance of the pose estimate such that the estimate is not lost?

Any and all help is much appreciated.

matlabbe, 2019-08-01:
We can enable auto-reset of odometry for convenience. In standalone, set Preferences->Odometry->"Automatically reset odometry..." at 1, so that after 1 frame being lost, the odometry is reset.

On ROS, for rtabmap's odometry nodes, set parameter `Odom/ResetCountdown` to 1.

cheers,
Mathieu

pringithub, 2019-08-01:
Wow - Thanks!

closed 2019-08-01 by pringithub