introlab/rtabmap #432: rtabmap make error

ppwwwwwpp, 2019-08-05:
[  4%] Built target rtabmap_utilite
[  4%] Built target res_tool
[ 42%] Built target rtabmap_core
[ 80%] Built target rtabmap_gui
[ 80%] Linking CXX executable ../../../bin/rtabmap
../../../bin/librtabmap_core.so.0.19.4：对‘visensor::ViSensorDriver::~ViSensorDriver()’未定义的引用
../../../bin/librtabmap_core.so.0.19.4：对‘visensor::ViSensorDriver::ViSensorDriver()’未定义的引用
../../../bin/librtabmap_core.so.0.19.4：对‘visensor::ViSensorDriver::getListOfCameraIDs() const’未定义的引用
../../../bin/librtabmap_core.so.0.19.4：对‘visensor::ViSensorDriver::init[abi:cxx11]()’未定义的引用
../../../bin/librtabmap_core.so.0.19.4：对‘visensor::ViSensorDriver::getCameraCalibration(visensor::SensorId::SensorId, visensor::ViCameraCalibration&, bool*, int) const’未定义的引用
collect2: error: ld returned 1 exit status
app/src/CMakeFiles/rtabmap.dir/build.make:545: recipe for target '../bin/rtabmap' failed
make[2]: *** [../bin/rtabmap] Error 1
CMakeFiles/Makefile2:416: recipe for target 'app/src/CMakeFiles/rtabmap.dir/all' failed
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

matlabbe, 2019-08-09:
[libvisensor](https://github.com/ethz-asl/libvisensor) is not integrated in official rtabmap repo, from which fork are you building rtabmap?

ppwwwwwpp, 2019-08-10:
I don't known  which fork, I'm build normally, but keep reporting this error.

matlabbe, 2019-08-16:
Then it is strange that it tries to link on libvisensor library. Can you do `ldd librtabmap_core.so.0.19.4`? Can you also show the output of `make VERBOSE=1` when making rtabmap?