introlab/rtabmap #435: Could not find acc data!

o0ozz, 2019-08-15:
Hi,

The sensor I used was D435i, running rtabmap, and the terminal hinted that the acceleration data could not be found. The driver I choosed is RealSense2.

```
[ WARN] (2019-08-15 11:53:46.216) CameraRealSense2.cpp:368::getPoseAndIMU() Could not find acc data to interpolate at time 1565841226149.322266 after waiting 30 ms (last is 1565841226137.336182)...
[ WARN] (2019-08-15 11:53:46.310) CameraRealSense2.cpp:368::getPoseAndIMU() Could not find acc data to interpolate at time 1565841226251.613037 after waiting 30 ms (last is 1565841226233.803467)...
[ WARN] (2019-08-15 11:53:46.350) CameraRealSense2.cpp:368::getPoseAndIMU() Could not find acc data to interpolate at time 1565841226284.072510 after waiting 30 ms (last is 1565841226265.957764)...
[ WARN] (2019-08-15 11:53:46.476) CameraRealSense2.cpp:368::getPoseAndIMU() Could not find acc data to interpolate at time 1565841226427.035645 after waiting 30 ms (last is 1565841226394.571289)...
```
What should I do? Thank you.

matlabbe, 2019-08-17:
Make sure you are plugged in an USB3. The acceleration buffer is filled here: 
https://github.com/introlab/rtabmap/blob/536f00147905f76f714d5157458c7f5a226b9eb0/corelib/src/camera/CameraRealSense2.cpp#L222-L226
And the waiting time for interpolation is done here:
https://github.com/introlab/rtabmap/blob/536f00147905f76f714d5157458c7f5a226b9eb0/corelib/src/camera/CameraRealSense2.cpp#L358-L365

Based on the warnings, it seems that you may increase the wait time to 35-40 ms [here](https://github.com/introlab/rtabmap/blob/536f00147905f76f714d5157458c7f5a226b9eb0/corelib/src/camera/CameraRealSense2.cpp#L306) (`1565841226427.035645-1565841226394.571289 = 32.46 ms`) instead of 30 ms. Are the warnings continuously appearing or just for a couple of seconds on initialization? If it is on initialization, you may ignore those warnings. If increasing the wait time to 35-40 fixes the problem, I may increase it and/or add a parameter.

o0ozz, 2019-08-26:
Hi matlabbe,
I increased the wait time to 35 ms. Now the terminal has no warning. Thank you!

commit 380fffb1 (2019-08-27): fixed #435
closed 2019-08-27 by matlabbe