introlab/rtabmap #436: [Feature request] Add support for new Azure Kinect camera (k4a)

brookman, 2019-08-24:


brookman, 2019-08-27:
Thank you for the reply! I have tried ROS and the K4A ros package. Unfortunately I wasn't able to compile/run rtabmap_ros on Windows. Is that even possible?

matlabbe, 2019-09-01:
I've never tried to build rtabmap_ros on Windows. I would suggest to try with an Ubuntu computer first.

scimad, 2020-01-17:
Any update on this?

spbwilson, 2020-02-13:
Also interested in any development on this!

matlabbe, 2020-02-21:
The kit cannot be bought from Canada, unfortunately.

tkircher, 2020-03-05:
> The kit cannot be bought from Canada, unfortunately.

I'll send you one if you're willing to put some time into supporting it.

matlabbe, 2020-03-10:
Yes sure! Many camera drivers have been integrated in rtabmap with someone/company sending me a development kit of their product. Sensors can be posted at this [address](https://www.usherbrooke.ca/3it/en/contact/) with my name "Mathieu Labbe".

Regards,
Mathieu

tkircher, 2020-03-17:
I'm sending one out this week.

matlabbe, 2020-04-04:
@tkircher Great thx, can you send me the tracking number if you have one (my github username @ gmail.com), the university closed since 2 weeks ago because of covid situation. The package may be held at Fedex/Ups/Post office...

tenneck, 2020-11-11:
Is the Azure Kinect camera now supported in rtabmap?

matlabbe, 2020-11-11:
Yes, even in latest windows binaries: https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts

closed 2020-11-11 by matlabbe