introlab/rtabmap #441: Questions about Velodyne Map with AprilTags

kylxyz, 2019-08-28:
Hi Mathew,

I created a map with Velodyne + Visual.
I just find something I could not understand,
I also upload my launcher if you could take a look at it and give some comments or hints to improve it.

I directly mapping instead of making from a bag, (actually I would like to if it is possible.)
and then I optimized the map. Here, I set `MarkerDetection = false`

then I re-created a map by using rtabmap gui, and this time I turned MarkerDetection = true in rtabmap gui. then a new map with apriltags is generated.

here I found some green string is not connected on the traverse blue lines, as far as I understand, green strings and yellow strings should be all connected with the blue lines.

[Launch File](https://drive.google.com/open?id=19nAVvGoxh_pzhp3_RUMYuEqgKyafTHGg)

[Raw db.](https://drive.google.com/open?id=1e0_eHoZmDCnYEGuE2bkYPllSyJYjY7_q)

[Optimized db without tags.](https://drive.google.com/open?id=1ewsf_k4qrC6ThdlLvd8U9YzRTo4zT_wt)
[screen shot](https://drive.google.com/open?id=1ercZ8QteVpNWNvB--k9qLR16jttsyl_3)

[re-created db with Tags.](https://drive.google.com/open?id=1DjQeZ6qGaSr5vPvg5UbFU95_j0xxz_RF)
[screen shot](https://drive.google.com/open?id=1QixTzqGRXi7jJ6t-Oj3aECu8b6-uPvgd)

Cheers,
Jin

matlabbe, 2019-09-01:
Hi Jin,

Thank you for the observation. Those blue nodes not linked to graph (only by landmark links) are nodes that were removed from the graph because the robot didn't move enough (`RGBD/LinearUpdate` and `RGBD/LinearUpdate` are not zero). I forgot to remove also the landmark links when that happens. It sohuld be fixed by the commit above.

cheers,
Mathieu

kylxyz, 2019-09-03:
Hi Mathieu, 

thank you for the kind reply, I will update the commit.
Cheers.
Jin

closed 2019-09-03 by kylxyz