introlab/rtabmap #443: Integrate Okvis to RGBDMapping

ahmadhasan2k8, 2019-08-29:
The example eurocDataset code shows how to use imu data for okvis but it's not event based. Is there any simple way to integrate OKVIS to the RGBD mapping example?

matlabbe, 2019-09-01:
It would require another thread (like IMUThread class but for your sensor) publishing the IMU events. 

When it comes to integrate more sensors, I would suggest to look at the ros implementation of rtabmap, in particular this [example](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/euroc_datasets.launch).

cheers,
Mathieu

ahmadhasan2k8, 2019-09-03:
@matlabbe Thanks for your response. It does seem like a lot of processing done again where there already is a process function in IMU class. I was looking at OdometryRos.cpp implementation. I might have been looking into the wrong place. But I looked into https://github.com/introlab/rtabmap/blob/e09a872fa2149482e0e4ae3827176c8390da8517/corelib/src/IMUThread.cpp where it seems like it only assumes data from a csv file. Do you think if I rewrite/ inherit the same way but replace the data reading directly from imu instead of csv file. Will it work?

matlabbe, 2019-09-06:
Yes, the idea is to publish an `IMUEvent` from a thread reading your imu data, like on those lines:https://github.com/introlab/rtabmap/blob/e09a872fa2149482e0e4ae3827176c8390da8517/corelib/src/IMUThread.cpp#L143-L144

The IMUThread class was very specific to txt file input format of EuRoC dataset, you can indeed rewrite the code inside for your sensor.

cheers,
Mathieu

ahmadhasan2k8, 2019-09-26:
@matlabbe I tried doing that and setting cameraThread FrameRate to 20 and ImuThread rate to 100. Now I get this 
```
W0926 15:56:27.364990  1766 Frontend.cpp:282] - Skipping propagation as only one IMU measurement has been given to frontend. Normal when starting up.
W0926 15:56:27.365177  1764 ThreadedKFVio.cpp:423] Frame is newer than oldest IMU measurement. Dropping it.imu time: 16908.654275000  frame time: 16908.650000000
```

Also I cannot generate Odometry and get this

```
I0926 15:56:29.262804  1765 Frontend.cpp:761] Initializing pose from 2D-2D RANSAC: orientation only
E0926 15:56:29.339404  1764 FrameSynchronizer.cpp:104] Dropping frame with id 1
 cam captureTime: 1.69112e+10
I0926 15:56:29.367509  1765 Frontend.cpp:761] Initializing pose from 2D-2D RANSAC: orientation only

```

Could you please help me figure this out? 

Thanks

matlabbe, 2019-09-27:
Can you try it under ros? I think it would be easier to debug (like if you can reproduce the same error, at least you could record a rosbag with frames  + imu to share). I've never tried something like this with the standalone. The closer I did is recently I updated realsense2 driver to publish IMU data faster than frames for VINS-Fusion. I didn't try okvis.