introlab/rtabmap #445: make error with pcl

ZhouSongxin, 2019-09-21:
hello, everyone. i have a qustion about pcl,as follows.

~/rtabmap/build$ cmake ..
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP
-- Found OpenCV: /opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-6.2;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/include/tcl;/usr/include/hdf5/openmpi;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/jsoncpp;/usr/include/x86_64-linux-gnu;/usr/include/libxml2;/usr/include/python2.7
-- Found ZLIB: /usr/include
-- VTK_RENDERING_BACKEND=OpenGL
-- Found Freenect: /opt/ros/kinetic/include/libfreenect
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Old g2o version detected with c++03 interface (config file: /opt/ros/kinetic/include/g2o/config.h).
-- Found g2o: /opt/ros/kinetic/include;/home/zhousongxin/ros/Kinetic/ros_ws/devel_isolated/suitesparse/include/suitesparse;/home/zhousongxin/ros/Kinetic/ros_ws/devel_isolated/suitesparse/include/suitesparse
-- GTSAM include directory:  /home/zhousongxin/ros/Kinetic/ros_ws/devel_isolated/gtsam/lib/cmake/GTSAM/../../../include
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- A library with BLAS API found.
-- Found CUDA: /usr/local/cuda-10.0 (found suitable version "10.0", minimum required is "10") 
-- Found ZED sdk: /usr/local/zed/include
-- Found CUDA: /usr/local/cuda-10.0 (found version "10.0") 
-- Found CUDA: /usr/local/cuda-10.0/include
-- Found RealSense2: 
-- Found octomap 1.8.1: /opt/ros/kinetic/include
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 2.0.0 installed in: /home/zhousongxin/ros/Kinetic/ros_ws/devel_isolated/ceres-solver with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations]
-- Architecture: x86_64
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.19.5
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DEIGEN_USE_NEW_STDVECTOR;-DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET;-DFLANN_STATIC;-Dqh_QHpointer
--   PCL_VERSION = 1.7.2
-- 
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Qt5                  = YES (License: Open Source or Commercial)
--   With VTK 6.2              = YES (License: BSD)
--   With external SQLite3     = YES (License: Public Domain)
--   With ORB OcTree           = YES (License: GPLv3)
--   With Madgwick             = YES (License: GPL)
--   With FastCV               = NO (FastCV not found)
-- 
--  Solvers:
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--  *With g2o                  = YES (License: BSD)
--  *With GTSAM                = YES (License: BSD)
--  *With Ceres                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--  *With libpointmatcher      = NO (libpointmatcher not found)
-- 
--  Reconstruction Approaches:
--   With OCTOMAP              = YES (License: BSD)
[... 275 lines cut ...]
                                                       ^
[ 36%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/frame_projector.cpp.o
[ 37%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/slam_calibrator.cpp.o
[ 37%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/opencv/ORBextractor.cc.o
[ 37%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/opencv/solvepnp.cpp.o
[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/toro3d/posegraph3.cpp.o
[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/toro3d/treeoptimizer3_iteration.cpp.o
[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/toro3d/treeoptimizer3.cpp.o
[ 38%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/toro3d/posegraph2.cpp.o
[ 39%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/toro3d/treeoptimizer2.cpp.o
[ 39%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/g2o/edge_se3_xyzprior.cpp.o
[ 39%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/edge_se2Switchable.cpp.o
[ 40%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/edge_se3Switchable.cpp.o
[ 40%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/edge_switchPrior.cpp.o
[ 40%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/types_g2o_robust.cpp.o
[ 41%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/vertigo/g2o/vertex_switchLinear.cpp.o
[ 41%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OctoMap.cpp.o
[ 41%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/gtsam/GravityFactor.cpp.o
[ 42%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/imufilter/MadgwickFilter.cpp.o
[ 42%] Linking CXX shared library ../../../bin/librtabmap_core.so
[ 42%] Built target rtabmap_core
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
 
I need your help. thanks.

matlabbe, 2019-09-22:
Hi,
If you comment this line: https://github.com/introlab/rtabmap/blob/c437375f0851b1e1ca9ff27d72aca2c2fc6d1083/tools/CMakeLists.txt#L4
Do everything else build?

matlabbe, 2019-10-14:
Try cleaning the build directory of rtabmap before rebuilding it, at least remove the CMakeCache.txt if you installed another version of PCL. The linker may still be used headers form the old PCL version, while trying to link on the new PCL version. By cleaning the build directory, cmake will be processed again and hopefully adjust to PCL correct paths.

Levi568, 2019-10-16:
@matlabbe Hi, the problem still remains. D you know if I can add the correct path to the CMakeList.txt file?

matlabbe, 2019-10-21:
You can do "`make VERBOSE=1`" to see to which library it is linking on.

Levi568, 2019-10-31:
@matlabbe It is linked to /usr/include/pcl-1.7

matlabbe, 2019-11-03:
What is your computer? Here is the normal output on Ubuntu 16.04: https://travis-ci.org/introlab/rtabmap/jobs/606827302