introlab/rtabmap #448: help: need to open 0.19.3 database on Windows (where only 0.18.0 is released)

frank26080115, 2019-10-07:
This is dumb but I took some 3D scans of stuff with 0.19.3 running on a Raspberry Pi, a bare minimum build so I can't even follow the tutorial to export the mesh, because the dependancies are simply missing. The plan is to simply copy over the database to my Windows computer and then do the mesh export.

But the problem is that there's only 0.18.0 for Windows. When I tried to open the database, it gave me an error saying that it's from a future version and thus cannot be opened. I tried to edit the version number inside the database but then RTAB-Map straight up crashed trying to read it. (which I totally understand)

I know my obvious options: build for Windows from source, build for my Ubuntu VM from source, downgrade my Raspberry Pi's version and retake the scans, or maybe use a SQLite editor and rip out the offending tags... (ok that's a stretch...)

But I am really hoping for a pre-release binary of 0.19.3 for Win64... err... is this something convenient to upload?

matlabbe, 2019-10-07:
Download latest windows binaries here: https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts

They don't have all camera drivers like the official releases, but at least you can visualize/export/replay databases.

Cheers,
Mathieu

frank26080115, 2019-10-07:
Thank you very much!

closed 2019-10-07 by frank26080115