introlab/rtabmap #450: make error

Levi568, 2019-10-10:
### Output using **cmake -DWITH_VINS=OFF ..** 
-- The C compiler identification is GNU 6.5.0
-- The CXX compiler identification is GNU 6.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.10
-- Found eigen: /usr/local/include/eigen3  
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- Checking for module 'libopenni2'
--   Found libopenni2, version 2.2.0.3
-- Found OpenNI2: /usr/lib/libOpenNI2.so  
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- Found libusb-1.0: /usr/include  
-- Checking for module 'flann'
--   Found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- Found PCL_COMMON: /usr/lib/x86_64-linux-gnu/libpcl_common.so  
-- Found PCL_OCTREE: /usr/lib/x86_64-linux-gnu/libpcl_octree.so  
-- Found PCL_IO: /usr/lib/x86_64-linux-gnu/libpcl_io.so  
-- Found PCL_KDTREE: /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/lib/x86_64-linux-gnu/libpcl_search.so  
-- Found PCL_SURFACE: /usr/lib/x86_64-linux-gnu/libpcl_surface.so  
-- Found PCL_FILTERS: /usr/lib/x86_64-linux-gnu/libpcl_filters.so  
-- Found PCL_FEATURES: /usr/lib/x86_64-linux-gnu/libpcl_features.so  
-- Found PCL_REGISTRATION: /usr/lib/x86_64-linux-gnu/libpcl_registration.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so  
-- Found PCL_GEOMETRY: /usr/include/pcl-1.7  
-- Found PCL_SEGMENTATION: /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so  
-- Found PCL_VISUALIZATION: /usr/lib/x86_64-linux-gnu/libpcl_visualization.so  
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found OpenMP
[... 89 lines cut ...]
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wei/rtabmap/build

### Error message
[  4%] Built target rtabmap_utilite
[  5%] Built target res_tool
[ 43%] Built target rtabmap_core
[ 81%] Built target rtabmap_gui
[ 82%] Linking CXX executable ../../../bin/rtabmap
../../../bin/librtabmap_core.so.0.19.5: undefined reference to `pcl::search::Search<pcl::PointNormal>::getName[abi:cxx11]() const'
../../../bin/librtabmap_gui.so.0.19.5: undefined reference to `pcl::search::Search<pcl::PointXYZRGBNormal>::getName[abi:cxx11]() const'
../../../bin/librtabmap_core.so.0.19.5: undefined reference to `pcl::search::Search<pcl::PointXYZI>::getName[abi:cxx11]() const'
../../../bin/librtabmap_core.so.0.19.5: undefined reference to `pcl::search::Search<pcl::PointXYZRGB>::getName[abi:cxx11]() const'
../../../bin/librtabmap_core.so.0.19.5: undefined reference to `pcl::search::Search<pcl::PointXYZ>::getName[abi:cxx11]() const'
collect2: error: ld returned 1 exit status
app/src/CMakeFiles/rtabmap.dir/build.make:529: recipe for target '../bin/rtabmap' failed
make[2]: *** [../bin/rtabmap] Error 1
CMakeFiles/Makefile2:416: recipe for target 'app/src/CMakeFiles/rtabmap.dir/all' failed
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

Same as issue #380, I installed VINS-Mono and VINS-Fusion on Ubuntu 16.04, ROS Kinetic successfully. I also installed the pcl and test it with c++ code, but not sure why there is an issue when building on rabmap

matlabbe, 2019-10-14:
Is it the PCL binaries or built from source? Has rtabmap been built with another PCL version previously? If so, clean the build directory, relaunch cmake and build again.

Levi568, 2019-10-31:
@matlabbe my pcl-1.7 is built from source. May I ask that since ROS has default pcl-1.7, can I just uninstall my pcl-1.7 and use the default one? I think there may be some confliction issue.

matlabbe, 2019-11-03:
Yes, it is even recommended to use PCL binaries installed with ROS. This is the default build. We don't test against all PCL versions built from source with all different options.