introlab/rtabmap #451: Kinect Studio?

farray81, 2019-10-14:
Hello,

I'm trying Rtabmap in Windows, and when I conect the Kinect v2, there's no problem, but when I disconnect and try to use Kinect Studio playing a sequence recorded previously, Rtabmap doesn't work,  this error message appears: 

[ERROR] (2019-10-14 15:24:08) CameraFreenect2.cpp:313::rtabmap::CameraFreenect2::init() CameraFreenect2: no device connected or failure opening the default one! Note that rtabmap should link on libusb of libfreenect2. Tip, before starting rtabmap: "$ export LD_LIBRARY_PATH=~/libfreenect2/depends/libusb/lib:$LD_LIBRARY_PATH"

Anyone knows if there's any chance that Rtabmap works with Kinect Studio?

Thanks

matlabbe, 2019-10-15:
Hi, not sure what "Kinect Studio playing a sequence recorded previously" does. Is it simulating a kinect2? You selected the Freenect2 driver, a kinect v2 must be plugged to work, it won't use any feed coming from Kinect Studio. Kinect for Windows SDK 2.0 driver can be selected (instead of Freenect2) in RTAB-Map if RTAB-Map has been built with it.

farray81, 2019-10-15:
Yes, Kinect Studio is included in Kinect SDK, and it allows you to record data from tje Kinect and, later, you can play this sequence any time, with the Kinect disconnected, simulating that a Kinect is connected. 

I guess the problem can be I can't select Kinect for Windows SDK 2.0 driver in RTAB-Map, this option appears in gray, I can only select Freenect2. I directly installed the RTABMap-0.18.0-win64-CUDA10.exe, I had installed Kinect for Windows SDK 2.0 + Kinect for Windows Runtime 2.0, and just in case, I installed UsbDK v1.00-12, but the option "Kinect for Windows SDK 2" appears in gray, I don't know how to do to enable it.

matlabbe, 2019-10-21:
The released 0.18.0 windows binaries are not built with K4W2 sdk to simplify the installation for those not needing K4W2 (so we force using Freenect2 instead on windows). Try the version 0.17.5 which has support for K4W2: https://github.com/introlab/rtabmap/releases/tag/0.17.5.

To get K4W2 working in recent versions, we should build RTAB-Map from source. See here for Visual Studio: https://github.com/introlab/rtabmap/blob/master/.appveyor.yml (note that those [steps](https://github.com/introlab/rtabmap/blob/6435c74bfe3f7b78eb42ba692d1a75dac9fb9c0a/.appveyor.yml#L90-L117) are not required)

cheers,
Mathieu

farray81, 2019-10-22:
> The released 0.18.0 windows binaries are not built with K4W2 sdk to simplify the installation for those not needing K4W2 (so we force using Freenect2 instead on windows). Try the version 0.17.5 which has support for K4W2: https://github.com/introlab/rtabmap/releases/tag/0.17.5.

This works! Thank you, matlabee