introlab/rtabmap #453: Question: What is happening in the align code of CameraRealsense2.cpp

srinath-iko, 2019-10-15:
Hey @matlabbe !

https://github.com/introlab/rtabmap/blob/dd8e21e05d2ed8cf40af26b86111f2f8ebf4a0a3/corelib/src/camera/CameraRealSense2.cpp#L153-L204

In the alignment code that aligns the depth space to the colour space, I saw that the depth values are aligned, converted to mm and saved back into another matrix called "depth". 
1) Are the 16 bit int cast as 8 int  because opencv returns a uchar pointer when calling for data or does it have a greater purpose?
2) The 8 bit ints are multiplied with the depth scale, but it would result in different values when compared to multiplying 16 bit int values with the depth scale. The drift increases monumentally when this technique is removed. Why so?

Thanks!

matlabbe, 2019-10-15:
OpenCV's matrix data pointer is an unsigned char: https://docs.opencv.org/3.4/d3/d63/classcv_1_1Mat.html#a4d33bed1c850265370d2af0ff02e1564

Each depth value is casted to 16bits, then converted to float before processing: https://github.com/introlab/rtabmap/blob/dd8e21e05d2ed8cf40af26b86111f2f8ebf4a0a3/corelib/src/camera/CameraRealSense2.cpp#L165

srinath-iko, 2019-10-16:
Hey @matlabbe. 
Thanks! I did understand the operations happening with the depth.
Sorry for not making myself clear.

I had referred to these lines.

https://github.com/introlab/rtabmap/blob/dd8e21e05d2ed8cf40af26b86111f2f8ebf4a0a3/corelib/src/camera/CameraRealSense2.cpp#L195-L197

where the 8 bits uchars ( lower and higher bits of the uint16_t) are multiplied by the depth units separately and placed back in the depth image. This would not result in the real depth value being multiplied with the depth units. The resulting image looks choppy with less noise. Is that the aim of this procedure?

matlabbe, 2019-10-16:
Honestly, I am not sure exactly what this code is doing. This function was copy paste from realsense ros wrapper. Well, I just found that they modified that code in this commit: https://github.com/IntelRealSense/realsense-ros/commit/6c1e58c2107afa7795246f4e7ae878a0d4c3442d
I think the align functionality is [now](https://github.com/IntelRealSense/librealsense/issues/1569#issuecomment-384683110) available in the realsense library directly. We would have to compare the approaches.

srinath-iko, 2019-10-21:
I think the new alignment code makes sense as the older one seemed to reference the 16bit depth values as 2- 8bit values and manipulate them(multiply by the depth scale) separately. If there was no overflow when the 8bit depths were multiplied, it produces the same result as multiplying with the 16 bit values, but otherwise leads to many points being incorrectly assigned. This somewhat increases stability of the tracking/odometry when using RGBD but it is incorrect. We should use the 16bit values as mentioned in one of your links.

[https://github.com/IntelRealSense/realsense-ros/commit/6c1e58c2107afa7795246f4e7ae878a0d4c3442d](url)

This should solve the issue.

matlabbe, 2019-10-22:
Thanks for the info, I reopened the issue with label Bug to make and test the fix soon.

srinath-iko, 2019-10-23:
Hey @matlabbe,

You can maybe test this code out. should do the trick with minimal changes.

`void alignFrame(const rs2_intrinsics& from_intrin,
                                   const rs2_intrinsics& other_intrin,
                                   rs2::frame from_image,
                                   uint32_t output_image_bytes_per_pixel,
                                   const rs2_extrinsics& from_to_other,
								   cv::Mat & registeredDepth,
								   float depth_scale_meters)
{
    static const auto meter_to_mm = 0.001f;
    
    uint16_t* p_out_frame = reinterpret_cast<uint16_t*>(registeredDepth.data);
    auto from_vid_frame = from_image.as<rs2::video_frame>();
    auto from_bytes_per_pixel = from_vid_frame.get_bytes_per_pixel();

    static const auto blank_color = 0x00;
    UASSERT(registeredDepth.total()*registeredDepth.channels()*registeredDepth.depth() == other_intrin.height * other_intrin.width * output_image_bytes_per_pixel);
    memset(p_out_frame, blank_color, other_intrin.height * other_intrin.width * output_image_bytes_per_pixel);

    
    auto p_from_frame = reinterpret_cast<const uint16_t*>(from_image.get_data());
    auto from_stream_type = from_image.get_profile().stream_type();
    float depth_units = ((from_stream_type == RS2_STREAM_DEPTH)? depth_scale_meters:1.f);
    UASSERT(from_stream_type == RS2_STREAM_DEPTH);
    UASSERT_MSG(depth_units > 0.0f, uFormat("depth_scale_meters=%f", depth_scale_meters).c_str());
#pragma omp parallel for schedule(dynamic)
    for (int from_y = 0; from_y < from_intrin.height; ++from_y)
    {
        int from_pixel_index = from_y * from_intrin.width;
        for (int from_x = 0; from_x < from_intrin.width; ++from_x, ++from_pixel_index)
        {
            // Skip over depth pixels with the value of zero
            float depth = (from_stream_type == RS2_STREAM_DEPTH)?(depth_units * ((const uint16_t*)p_from_frame)[from_pixel_index]): 1.f;
            if (depth)
            {
                // Map the top-left corner of the depth pixel onto the other image
                float from_pixel[2] = { from_x - 0.5f, from_y - 0.5f }, from_point[3], other_point[3], other_pixel[2];
                rs2_deproject_pixel_to_point(from_point, &from_intrin, from_pixel, depth);
                rs2_transform_point_to_point(other_point, &from_to_other, from_point);
                rs2_project_point_to_pixel(other_pixel, &other_intrin, other_point);
                const int other_x0 = static_cast<int>(other_pixel[0] + 0.5f);
                const int other_y0 = static_cast<int>(other_pixel[1] + 0.5f);

                // Map the bottom-right corner of the depth pixel onto the other image
                from_pixel[0] = from_x + 0.5f; from_pixel[1] = from_y + 0.5f;
                rs2_deproject_pixel_to_point(from_point, &from_intrin, from_pixel, depth);
                rs2_transform_point_to_point(other_point, &from_to_other, from_point);
                rs2_project_point_to_pixel(other_pixel, &other_intrin, other_point);
                const int other_x1 = static_cast<int>(other_pixel[0] + 0.5f);
                const int other_y1 = static_cast<int>(other_pixel[1] + 0.5f);

                if (other_x0 < 0 || other_y0 < 0 || other_x1 >= other_intrin.width || other_y1 >= other_intrin.height)
                    continue;

                for (int y = other_y0; y <= other_y1; ++y)
                {
                    for (int x = other_x0; x <= other_x1; ++x)
                    {
                        int out_pixel_index = y * other_intrin.width + x;
                        //Tranfer n-bit pixel to n-bit pixel

                            const auto out_offset = out_pixel_index * output_image_bytes_per_pixel;
                            const auto from_offset = from_pixel_index * output_image_bytes_per_pixel;
                            p_out_frame[out_offset] = p_from_frame[from_offset] * (depth_units / meter_to_mm);
                        
                    }
                }
            }
        }
    }
}
`

closed 2019-10-23 by matlabbe

matlabbe, 2019-10-23:
To make it more general, I used the [align](https://github.com/IntelRealSense/librealsense/blob/81d469db173dd682d3bada9bd7c7570db0f7cf76/include/librealsense2/hpp/rs_processing.hpp#L645) approach from the realsense2 library directly.

commit 79ad8dc4 (2019-10-23): Fixed #453