introlab/rtabmap #454: Relocalization feature with RGBDMapping example

Keudeugi, 2019-10-17:
Hello, I want to ask about how should I use Relocalization feature on RGBDMapping code example. I am currently implementing a code that gather point clouds from every RGB-D frame sequences (not only points from graph-optimized; this skips points not used for making graph) in localization-only mode.  

So I loaded a pre-built map on top of RGBDMapping example, however I got stucked how should I guarantee the map (along with optimized points) is correctly loaded & the odometry is reflected based on the map.

There was no problem with displaying a map, however the input camera transformation does not originated based on built map. I checked every code about relocalization feature & appearance-based mapping and confirmed that relocalization is automatically done, therefore I guess that I should transform the current QT camera into relocalized one... 

Please correct me if I am wrong or have something missed. Thanks in advance.

matlabbe, 2019-10-21:
If localization happens, [odometryCorrection_](https://github.com/introlab/rtabmap/blob/6435c74bfe3f7b78eb42ba692d1a75dac9fb9c0a/examples/RGBDMapping/MapBuilder.h#L275) won't be Identity. That transform should be applied to odometry pose so that the camera follows the pre-built map. Maybe adding some code example you tried may help to better understand the problem.

cheers,
Mathieu

Keudeugi, 2019-10-23:
Thanks for the quick feedback! I've confirmed that relocalization is actually done automatically and checked that odometryCorrection_ follows pre-built map (not identity) once the pose is relocalized. One more thing I'd like to ask is, is there any way to decrease relocalization inference speed by tuning parameters, or by making more robust pose graph?

matlabbe, 2019-10-23:
For visual localization, the camera should indeed have a similar point of view than a least one frame in the graph. Some features like SURF or SIFT can be more scale/rotation/shear invariant, so more robust to different point of view. We can also increase `Rtabmap/DetectionRate` (default 1 Hz) to add more frames to map, thus having more point of views to compare to. If you want to be able to localize in any orientations/positions in the environment, you would have to map the environment with many point of views (e.g., a corridor should be mapped in two directions so we can localize wen traversing the corridor in one way or the other).

For graph robustness, look at this page: https://github.com/introlab/rtabmap/wiki/Robust-Graph-Optimization

For localization inference speed, do you mean the frame rate at which localization is done? If so, you can set `Rtabmap/DetectionRate` to 0 to do localization with all received images.

Keudeugi, 2019-11-09:
Thanks for the meaningful advices. Most of problems solved, closing the issue for now.

closed 2019-11-09 by Keudeugi