introlab/rtabmap #455: [ERROR] (2019-10-14 17:31:07.831) OdometryF2M.cpp:208::computeTransform() IMU received doesn't have orientation set, it is ignored. If you are using RTAB-Map standalone, enable IMU filtering in Preferences->Source panel. On ROS, use "imu_filter_madgwick" or "imu_complementary_filter" packages to compute the orientation.

2017qiuju, 2019-10-17:
[ERROR] (2019-10-14 17:31:07.831) OdometryF2M.cpp:208::computeTransform() IMU received doesn't have orientation set, it is ignored. If you are using RTAB-Map standalone, enable IMU filtering in Preferences->Source panel. On ROS, use "imu_filter_madgwick" or "imu_complementary_filter" packages to compute the orientation.

need help ！

matlabbe, 2019-10-21:
Are you on ROS or with the standalone? You are feeding IMU data to rtabmap without orientation estimated.

2017qiuju, 2019-10-23:
> Are you on ROS or with the standalone? You are feeding IMU data to rtabmap without orientation estimated.
I used vinsfusion and d435i to generate the binocular map, and then I used rtabmap / bin $. / rtabmap to load it. The above error occurred.

matlabbe, 2019-10-23:
The error above is with Odometry F2M approach:

- In Preferences->Source, scroll down to IMU filtering section, select a filtering approach (Magwick or Complementary), and leave "Publish inter IMU..." unchecked.

If you select VINS-Fusion for Odometry:

- In Preferences->Source, scroll down to IMU filtering section and check "Publish inter IMU...".

cheers,
Mathieu